raspberrypi/firmware #1128: OMX_EmptyThisBuffer takes 33ms to finish

oomek, 2019-04-04:
What might be the reason for OMX_EmptyThisBuffer on the audio_render taking 33ms to finish?
I'm trying to get the media player working with EGLImage drawn with opengl, but I'm having infinite amount of problems, I fix one, another appears.

Since I'm not aware of any synchronization mechanism between OMX_FillThisBuffer and eglSwapBuffers that would eliminate tearing allocated 2 opengl textures and I alternate between them. It fixed tearing, but introduced stuttering.

I fixed stuttering by delaying OMX_FillThisBuffer after the callback has launched until the appropriate eglSwapBuffers occured, but now I get OMX_EmptyThisBuffer on the audio_render taking up to 33ms to finish, which results in audio gaps and slowdowns after 1-2 second of playing a mp4 file.

I've tried increasing the size and amount of buffers on the audio_render, but it has only effect when I decrease it. Increasing does not have any effect.

I would appreciate if someone shared some hints regarding this issue. Is it a bug or just the lack of understanding of the underlying mechanism of how the callbacks work. Does blocking the callback in one thread (video) can affect the OMX_EmptyThisBuffer execution time in the other (audio) ?

popcornmix, 2019-04-04:
> Does blocking the callback in one thread (video) can affect the OMX_EmptyThisBuffer execution time in the other (audio) ?

Never block in a callback. You will stall other users.
Always signal another thread to do any real work or block.

oomek, 2019-04-04:
Thanks, That explains this behaviour. Where should I block in the video decoding thread then to have the egl buffer filled right after the eglswapbuffer finishes waiting?

oomek, 2019-04-04:
I use the following tunneling path:
```
clock ---> video_decode ---> video_scheduler ---> egl_render
       |
        -> audio_render
```

If i cannot stall the callback what is the proper way of postponing the OMX_FillThisBuffer call on the egl buffer in the fill_egl_texture_buffer() callback until a desired vsync has finished ( that will be every 2nd frame for 30fps video ). I've run out of ideas.

6by9, 2019-04-09:
(I thought the image on https://www.raspberrypi.org/forums/viewtopic.php?f=70&t=79929 looked familiar. Please put reference to other threads/github issues if you've already asked the question elsewhere)

You can not block in the callback context. Put the buffer on a queue and pass it to some other thread to process at an appropriate time, allowing the [Empty|Fill]BufferDone thread context to continue servicing other callbacks.
See ilclient's ilclient_empty_buffer_done and ilclient_fill_buffer_done for examples of this. https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/libs/ilclient/ilclient.c#L1653 https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/libs/ilclient/ilclient.c#L1709

There is one broker thread handling all the callbacks, therefore if you block in it you block all callbacks.

oomek, 2019-04-09:
Yes, I used the same infographics, but the Issues are unrelated. Since I've taken the advice and instead of stalling the callback I manipupate the xScale parameter of the video clock based on the delta time of the eglswapbuffers and the callback time to be around 12ms. This way I get no stuttering. 

I'm closing the issue then.

closed 2019-04-09 by oomek