raspberrypi/firmware #1130: h264_omx periodic keyframe

Serveurperso, 2019-04-10:
Hello,

How to add periodic keyframe with h264_omx hardware encoder ? I can do it with raspivid or v4l2 and uv4l... but I can't find a way for ffmpeg h264_omx (the -g option do not work)

Thanks

Serveurperso, 2019-04-10:
When I check NAL unit type I get with h264_omx I have only a SPS and PPS once, on startup :
SPS PPS I P P P P (...) P P P P I P P P P (...) P P P P I P P P P (...) P P P P etc...

On raspivid or /dev/video0 from V4L2/UV4L I get a periodic SPS and PPS NAL unit before every complete I frame like :
SPS PPS I P P P P (...) P P P P SPS PPS I P P P P (...) P P P P SPS PPS I P P P P (...) P P P P etc...

6by9, 2019-04-10:
(a) This sounds like an ffmpeg issue, not a Pi firmware one.

(b) Your question is ambiguous. The H264 will default to a IntraPeriod of 60, but also allows override either via ```SetParameter(OMX_IndexParamVideoAvc)``` field ```nPFrames```, or ```OMX_IndexConfigVideoAVCIntraPeriod```. This means it will insert an I frame every 60 frames by default.
Are you saying that that you never see I-frames, or they're at a different rate to your requested one?
You haven't quoted the raspivid/v4l2/uv4l option that you are comparing with, so that makes it very hard to assist.

NB I-frames do NOT require the header bytes to be resent. That would be raspivid's -ih option, or V4L2's repeat_sequence_header control.

6by9, 2019-04-10:
> When I check NAL unit type I get with h264_omx I have only a SPS and PPS once, on startup :

Not a firmware bug then.
That is the correct behaviour for an H264 stream. There is absolutely no obligation to resend the SPS/PPS before an I-frame, unless they have changed.

If you want those inserted, then you need to set ```OMX_IndexParamBrcmVideoAVCInlineHeaderEnable``` on the video_encode component. (Whilst it may have Param in the index enum, it appears that it can be set in any state, so is actually an OMX Config).
I don't believe there is a standardised IL enum for that function, hence the Brcm designation in the enum. I doubt ffmpeg will accept platform specific enums like that, but I may be wrong.

Serveurperso, 2019-04-10:
My problem is I need "Sequence Headers on every Keyframe", I don't think if I'm in the right place.
It's ffmpeg or h264_omx hardware encoder ?

6by9, 2019-04-10:
FFmpeg, or at least the way you are calling FFmpeg. The default will not be changed within the firmware.

By the looks of it, the option AV_CODEC_FLAG_GLOBAL_HEADER has some handling in omx.c 
```
{"global_header", "place global headers in extradata instead of every keyframe", 0, AV_OPT_TYPE_CONST, {.i64 = AV_CODEC_FLAG_GLOBAL_HEADER }, INT_MIN, INT_MAX, V|A|E, "flags"},
```

https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/options_table.h#L64
https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/omx.c#L685
Whether it is the option you need or not I couldn't say.

Serveurperso, 2019-04-10:
Browser bug sorry I just read your answer after I posted.
For streaming periodic headers it's usefull.
I found here : https://video.stackexchange.com/questions/20820/h264-omx-and-non-existing-pps-0-referenced/21245
I think the "-flags2 +local_header" ffmpeg option do not work for OMX (waiting a test tonight)
Else I must store and copy myself the SPS + PPS before each I-frame, it's easy for me because my client must check NAL unit type before sending to Broadway for decoding (I always start on a I frame)

Serveurperso, 2019-04-10:
Same here https://github.com/raspberrypi/firmware/issues/242

closed 2019-04-10 by Serveurperso

Serveurperso, 2019-12-10:
I successfully patched myself FFMpeg for this :
https://github.com/Serveurperso/FFmpeg/commit/6e410823e63103342e8fc3407ff6698808d4d1ef

dimitry-ishenko, 2022-04-16:
@VigibotDev does your patch work with `h264m2m` or just `h264_omx`?

VigibotDev, 2022-04-16:
I don't patch ffmpeg anymore because I fond the working cmdline not tested with other than h264_omx, but this should work else it's a ffmpeg bug.

dimitry-ishenko, 2022-04-16:
@VigibotDev oh. Which cmdline did you end up using?

VigibotDev, 2022-04-16:
Look inside my https://github.com/vigibot/vigiclient/blob/master/sys.json cmdline javascript array

dimitry-ishenko, 2022-04-16:
@VigibotDev the only relevant things I see are:

>    " -flags:v +global_header",
>   " -bsf:v dump_extra",

but they don't seem to make any difference for me.

VigibotDev, 2022-04-19:
What is your complete ffmpeg cmdline ?

dimitry-ishenko, 2022-04-19:
@VigibotDev here is a simplified version:
```
ffmpeg -hide_banner \
    -f v4l2 -framerate 30 -video_size 1280x720 -input_format mjpeg -i /dev/video0 \
    -f alsa -i hw:1 \
    -keyint_min 90 -g 90 -sc_threshold 0 -c:v h264_v4l2m2m -pix_fmt yuv420p \
    -flags:v +global_header -bsf:v dump_extra \
    -map 0 -s:v:0 1280x720 -b:v:0 5M -maxrate:v:0 5.5M -bufsize:v:0 7.5M \
    -map 1 -c:a:0 aac -b:a:0 256k \
    -f hls -hls_time 3 -hls_list_size 10 -hls_flags delete_segments -master_pl_name master.m3u8 -var_stream_map "v:0,a:0" \
v%v.m3u8
```