raspberrypi/firmware #1134: OMX video_render: error on enable render port

zoff99, 2019-05-01:
here is the testcase we use:

https://github.com/zoff99/ToxBlinkenwall/tree/3a5a0217ba4f2b724acd611e958f3f7ab1e96bf6/toxblinkenwall
```
cd toxblinkenwall

gcc -O0 \
-DDEBUG_OMX_TEST_PROGRAM \
-DRASPBERRY_PI -DOMX_SKIP64BIT -DUSE_VCHIQ_ARM \
-I/opt/vc/include -I/opt/vc/interface/vmcs_host/linux -I/opt/vc/interface/vcos/pthreads \
-fPIC -export-dynamic -o test_omx_02 \
-lm \
test_omx_02.c omx.c \
-I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads \
-I/opt/vc/include/interface/vmcs_host/linux -lbrcmEGL -lbrcmGLESv2 \
-lbcm_host -L/opt/vc/lib \
-std=gnu99 \
-lrt \
-lpthread -lv4lconvert \
-lmmal -lmmal_core -lmmal_vc_client -lmmal_components -lmmal_util \
-L/opt/vc/lib -lbcm_host -lvcos -lopenmaxil -ldl

./test_omx_02
```

zoff99, 2019-05-01:
i think we used the commands in the correct order. other command ordering i have tried give either more errors, or don't work at all.

any help welcome.

6by9, 2019-05-01:
Generally following the way that the ilclient library handles IL is a good plan.

In the case of enabling ports, ```ilclient_enable_port_buffers``` would be the recommended template - https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/libs/ilclient/ilclient.c#L798

I can't recall whether ports are initialised in the enabled or disabled state. I suspect enabled as ilclient_create_component has an option for ILCLIENT_DISABLE_ALL_PORTS - https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/libs/ilclient/ilclient.c#L362. I suspect sending OMX_CommandPortEnable to a port that is already enabled will cause an error to be raised.