raspberrypi/firmware #1135: Pi 3B doesn't detect WiFi. Exact same kernel/firmware works perfect on another Pi with the same revision

dividuum, 2019-05-09:
I have this odd problem and I've run out of things to try to fix this myself. Essentially I have two identical Pi3 running the exact same OS setup. One detects the WiFi adapter, while the other doesn't. This Pi is an exception as I have at least 50 other Pis successfully using the same kernel and Wifi works for them. Output of `grep Revision /proc/cpuinfo ; vcgencmd version; uname -a` is exactly the same (except for the hostname) on both:

```
Revision        : a02082
Jan 22 2019 16:50:58 
Copyright (c) 2012 Broadcom
version 7bfabcecab2918f85a2a217b389e256eac696962 (clean) (release) (start)
Linux info-beamer-764521935 4.14.94-v7+ #1193 SMP Tue Jan 22 15:34:30 GMT 2019 armv7l GNU/Linux
```

Firmware log via `vcdbg log msg 2>&1|grep -v HDMI`: Except for timestamps it's the same on both, except for last two lines. On the working Pi:

```
001202.918: brfs: File read: /mfs/sd/config.txt
001204.474: brfs: File read: 2346 bytes
001205.147: Failed to open include file 'config/device-settings.txt'
001205.671: brfs: File read: /mfs/sd/config/screen-settings.txt
001205.819: brfs: File read: 26 bytes
001206.067: Failed to open include file 'config/serial-settings.txt'
001206.342: Failed to open include file 'config/rtc-settings.txt'
001206.616: Failed to open include file 'config/poe-settings.txt'
001206.936: Failed to open include file 'config/userconfig.txt'
001206.991: brfs: File read: /mfs/sd/config/userconfig.txt
001263.708: brfs: File read: /mfs/sd/config.txt
001264.832: brfs: File read: 2346 bytes
001265.079: Failed to open include file 'config/device-settings.txt'
001265.609: brfs: File read: /mfs/sd/config/screen-settings.txt
001265.761: brfs: File read: 26 bytes
001266.008: Failed to open include file 'config/serial-settings.txt'
001266.287: Failed to open include file 'config/rtc-settings.txt'
001266.566: Failed to open include file 'config/poe-settings.txt'
001266.880: Failed to open include file 'config/userconfig.txt'
001266.941: brfs: File read: /mfs/sd/config/userconfig.txt
001267.407: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001464.934: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
001466.215: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001568.794: *** Restart logging
001571.888: brfs: File read: /mfs/sd/cmdline.txt
001571.992: Read command line from file 'cmdline.txt':
001572.058: 'dwc_otg.lpm_enable=0 console=tty0 console=tty1 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet plymouth.ignore-serial-consoles loglevel=1 quiet consoleblank=0'
001572.380: brfs: File read: 176 bytes
001907.494: brfs: File read: /mfs/sd/kernel7.img
001907.558: Loading 'kernel7.img' to 0x8000 size 0x4b69f8
001907.633: brfs: File read: 4942328 bytes
002214.431: brfs: File read: /mfs/sd/initramfs.gz
002214.496: Loading 'initramfs.gz' to 0x4be9f8 size 0x45b901
002754.716: No kernel trailer - assuming DT-capable
002754.800: brfs: File read: 4569345 bytes
002760.869: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
002760.936: Loading 'bcm2710-rpi-3-b.dtb' to 0x4be9f8 size 0x62df
002900.990: brfs: File read: 25311 bytes
002904.918: brfs: File read: /mfs/sd/config.txt
002905.206: dtparam: audio=on
002924.985: brfs: File read: 2346 bytes
002925.234: Failed to open include file 'config/device-settings.txt'
002927.783: brfs: File read: /mfs/sd/config/screen-settings.txt
002927.919: brfs: File read: 26 bytes
002928.165: Failed to open include file 'config/serial-settings.txt'
002928.444: Failed to open include file 'config/rtc-settings.txt'
002928.722: Failed to open include file 'config/poe-settings.txt'
002929.034: Failed to open include file 'config/userconfig.txt'
002929.095: brfs: File read: /mfs/sd/config/userconfig.txt
004015.700: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
004139.250: Device tree loaded to 0x2eb9d800 (size 0x6762)
004141.307: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
005770.132: vchiq_core: vchiq_init_state: slot_zero = 0xef980000, is_master = 1
005780.690: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000052
015872.111: *** No KHAN handle found for pid 303
```

[... 125 lines cut ...]
framebuffer_width=1280
gpu_mem_1024=256
gpu_mem_512=128
hdmi_drive=2
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4
overscan_scale=1
sdtv_aspect=3
```

Non-working (on that one Pi):
```
initramfs initramfs.gz followkernel
disable_overscan=1
hdmi_force_hotplug=1
sdtv_mode=2
gpu_mem_512=256
gpu_mem_256=128
dtparam=audio=on
```

Except for the initramfs difference, that looks pretty harmless to me.

Any idea what I might try or what might cause this? I found https://github.com/raspberrypi/linux/issues/2132, but I'm not sure that's relevant as the exact same kernel/firmware works for one configuration and doesn't for another and I fail to see what the difference might be.

JamesH65, 2019-05-09:
Given the same SD card work on other identical Pi's I'd say its was simply broken. The SOC doesn't appear to be able to detect/talk to the wireless chip, so perhaps a broken track or similar?

pelwell, 2019-05-09:
Given that a different OS works on the same Pi, I think there's something more to this than a simple hardware fault.

What does a stock Raspbian image do on the flaky Pi?

dividuum, 2019-05-09:
Thanks a lot of the quick responses. I'll see if we can manage to install a recent stock Raspbian on there. Might take a moment. Any output I might look for?

pelwell, 2019-05-09:
I'm hoping it will just work, but if not, your `dmesg | egrep "mmc|brcm"` would be a good starting point.

JamesH65, 2019-05-09:
Ah, hadn't read that correctly, so the wifi does work with a different OS.
Apologies.

dividuum, 2019-05-09:
Are there any other logs I might not be aware of where I can compare the identically set up Pis? Any way to increase the verbosity on any relevant logs?

pelwell, 2019-05-09:
The kernel log is all that I really care about - dmesg will grab it all. You could compare the device tree on working and non-working systems, just in case the images are actually different or the firmware is manipulating it in different ways:
```
$ dtc -I fs /proc/device-tree > mumble.dts
```

dividuum, 2019-05-09:
The images are bitwise identical. Otherwise my system would automatically repair itself to make sure they are. I'll have a look at the device screen to see if there's a difference. Thanks.

dividuum, 2019-05-12:
Only difference is the Pi's serial number, the resulting MAC and (I guess) the [kaslr-seed](https://github.com/raspberrypi/firmware/issues/694). 
```
--- kaputt.dts  2019-05-12 17:00:50.913390147 +0200
+++ working.dts 2019-05-12 17:00:50.913390147 +0200
@@ -2,7 +2,7 @@

 / {
    compatible = "raspberrypi,3-model-b", "brcm,bcm2837";
-   serial-number = "000000002d91adcf";
+   serial-number = "00000000f2c383ee";
    model = "Raspberry Pi 3 Model B Rev 1.2";
    memreserve = <0x30000000 0x10000000>;
    interrupt-parent = <0x1>;
@@ -74,7 +74,7 @@
    };

    system {
-       linux,serial = <0x0 0x2d91adcf>;
+       linux,serial = <0x0 0xf2c383ee>;
        linux,revision = <0xa02082>;
    };

@@ -1069,7 +1069,7 @@

                usbether@1 {
                    compatible = "usb424,ec00";
-                   local-mac-address = [b8 27 eb 91 ad cf];
+                   local-mac-address = [b8 27 eb c3 83 ee];
                    phandle = <0x6e>;
                    reg = <0x1>;
                };
@@ -1163,7 +1163,7 @@
        bootargs = "8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty0 console=tty1 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet plymouth.ignore-serial-consoles loglevel=1 quiet consoleblank=0";
        linux,initrd-start = <0x2eba4000>;
        linux,initrd-end = <0x2efff901>;
-       kaslr-seed = <0x98c64b19 0x5ac708cf>;
+       kaslr-seed = <0x9df514bf 0xc0e67dfa>;
    };

    thermal-zones {
```
I still wonder if this can be a hardware problem. Another Pi 3 worked perfect.

pelwell, 2019-05-12:
It is sounding that way. Did you try the Raspbian image?

dividuum, 2019-05-13:
No WiFi indicator icon on the latest Raspbian Desktop. So I guess it's an hardware issue and not just some driver/firmware problem we somehow missed. I guess I'll close this ticket as there's nothing left to investigate?

closed 2019-05-13 by dividuum