raspberrypi/firmware #1136: Conditional filter to set HDMI as the default display - Feature support

paulenuta, 2019-05-15:
Please add a new condition in `config.txt` to allow disabling the RPILCD when HDMI display is connected at boot.
```
# general config.txt settings
...
[HDMI]
# settings here are applied only if a HDMI monitor is connected
display_default_lcd=0
disable_touchscreen=1
hdmi_group=2
hdmi_mode=16
...
```
I have a Raspberry Pi DSI LCD as default/primary display, but if a HDMI display is attached I need to have this one as default/primary display for the system.
Currently, if a Raspberry Pi DSI LCD is detected it will be used as the default display, but if we connect an HDMI display we need
`display_default_lcd=0`
or
`ignore_lcd=1`
in `config.txt` to make HDMI the default display.
[https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=240402&p=1467658#p1467323](url)

popcornmix, 2019-05-15:
Can you try using the ``[edid=...]`` filter?
https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

paulenuta, 2019-05-15:
> Can you try using the `[edid=...]` filter?
> https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

I will try, but this way I'll be bound only to a specific HDMI monitor.

paulenuta, 2019-05-15:
Just tested [EDID=] filter and I can confirm that it works, at least with the monitor I have now.
This is the content:
` /home/boot/config.txt
[EDID=LEN-LEN_L22e-20]
display_default_lcd=0
#display_lcd_rotate=0
#display_hdmi_rotate=0
#hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=9
`

paulenuta, 2019-05-16:
@popcornmix since the [EDID=...] worked, does also mean that it will be easier to implement an [HDMI] condition?

JamesH65, 2019-05-16:
It's probably not hugely difficult, its just finding the time to do it, in an already busy schedule, and it won't use [HDMI], but a simpler config item.

paulenuta, 2019-05-16:
If it is open source I could try a solution in a PR.

JamesH65, 2019-05-16:
It's in the firmware so closed source I am afraid.

paulenuta, 2019-06-13:
No happy news?

JamesH65, 2019-06-13:
Not had time and this area of the code is under a lot of churn right now due to work on the KMS driver. I'd prefer to wait until that is stable before digging in.

paulenuta, 2019-06-24:
Since Pi 4 is out, any chance to get this option sooner @JamesH65, @pelwell ?

paulenuta, 2019-08-02:
Hi there,
any update on this issue?

JamesH65, 2019-08-02:
No, not as yet.

paulenuta, 2019-09-20:
Hi @JamesH65 hope you'll get some time for this.
Thanks!

JamesH65, 2019-09-20:
We've just merged a change to the firmware that enables [edid=*] to mean any connected HDMI monitor. So you can use this to define settings to set only if an HDMI device is attached. 

Should be in the next rpi-update, in the next few days. Once it's in I'll close this issue.

JamesH65, 2019-09-23:
@paulenuta Did this change fix your issue?

paulenuta, 2019-09-23:
> @paulenuta Did this change fix your issue?

Yes. I'll be back with confirmation after testing.

Thanks a lot for your support.

closed 2019-09-23 by paulenuta

paulenuta, 2019-09-30:
Just tested [EDID=*] and it works with "ignore_lcd=1" but not with "display_default_lcd=0" contrary to what I was [saying](https://github.com/raspberrypi/firmware/issues/1136#issuecomment-492650079) previously.
I'll search for that testing card and monitor to verify my comment on [EDID=LEN-LEN_L22e-20].

owlyowl, 2024-09-15:
Sorry where do you get the Eid for the HDMI input? from I'm new to all this

6by9, 2024-09-16:
https://en.wikipedia.org/wiki/Extended_display_identification_data
It's provided by the monitor (or other HDMI sink) to tell the source what it is and what modes it supports.

`tvservice` is now deprecated as we've moved to the kernel display stack. You probably want `edid-decode` to decode the EDID, but you don't say what you're actually wanting to do with it.

owlyowl, 2024-09-16:
Sorry I'm trying to achieve similar where the LCD is classed as the primary
display but if HDMI is plugged in it takes priority for debugging and setup
kind of purposes.

6by9, 2024-09-16:
Open a new issue or forum post with the details of what you're trying to do.

Most of the content on this thread is largely out of date due to the changes in display drivers.
The EDID conditional filters will still work, but commands like `ignore_lcd=1`, `display_default_lcd=0`, `disable_touchscreen=0`, and `hdmi_*` are all deprecated.