raspberrypi/firmware #1137: VCHIQ mailbox channel is undocumented

ptesarik, 2019-05-16:
There is a wiki page about the mailbox channels (https://github.com/raspberrypi/firmware/wiki/Mailboxes), but no explanation of the VCHIQ interface. The Linux driver has been in staging for years, and more information on the interface is needed to get it promoted as a 'real' kernel driver.

JamesH65, 2019-05-16:
If this https://github.com/raspberrypi/documentation/tree/JamesH65-mailbox_docs/configuration/mailboxes doesn't help, I don't think there is much that can be done in the near future.

What specific information is needed to make a 'real' kernel driver ? Although I suspect our definition of real may differ, and I also doubt that any extra docs will help get it out of staging.

ptesarik, 2019-05-16:
Well, the current (upstream) driver may cause data corruption (according to the TODO). I suspect this can happen when the VPU (firmware) writes to a buffer that has already been freed by the ARM core.

ptesarik, 2019-05-16:
> If this https://github.com/raspberrypi/documentation/tree/JamesH65-mailbox_docs/configuration/mailboxes doesn't help, I don't think there is much that can be done in the near future.

If nothing else, tag 0x00048010 (RPI_FIRMWARE_VCHIQ_INIT) is missing from this documentation.

ptesarik, 2019-05-17:
Please review  raspberrypi/documentation#1150 for starters. However, some documentation of the VCHIQ mechanism is also desirable (message format, doorbell usage, etc.), so I'd like to leave this issue open even if you merge my PR.

lategoodbye, 2019-05-21:
Please also refer to this comment https://github.com/raspberrypi/linux/issues/2730#issuecomment-442599071