raspberrypi/firmware #1138: Create firmware based interface to I2C0

vianpl, 2019-05-17:
As most of you know the I2C0 bus is 'owned' by the firmware and it's encouraged not to use it. If we had a firmware interface to the I2C0 bus, that took care internally of managing firmware and OS requests, we could create a I2C0 firmware based driver. With that we could get rid of a whole set of drivers that exist today in the kernel tree and provide some extra flexibility to users. Off the top of my head we could delete the touchscreen, back-light and exp-gpio drivers which probably have their custom I2C device driver counterpart.

This would both simplify maintenance for everyone involved with the RPi kernel and firmware as everything would be centralized in one place.

The downside would be users connecting slow or broken I2C devices on the bus, which would potentially break the devices the firmware is responsible of. That said, as far as I know, they are not critical, so it  doesn't seem too bad, and it's an issue that already exists.

I could help with the kernel part of the feature.

6by9, 2019-05-17:
The gpio expander isn't on i2c0.

If you're not using the camera or display from the firmware then i2c0 can be used from the ARM.

(You can get the firmware to initialise the display bridge chip and then not poll the display touchscreen using ```disable_touchscreen=1``` in config.txt)

vianpl, 2019-05-17:
> The gpio expander isn't on i2c0.

Sorry, my bad

ptesarik, 2019-08-29:
Correct me if I'm wrong, but looking at dt-blob.dts, it seems to me that at least RPi2 and RPi3 boards also route i2c0 to the SMPS chip. Your answer suggests that the firmware does not use SMPS_SDA and SMPS_SCL while an ARM operating systems is running and that it's guaranteed to be that way (even by future firmware versions).

As an aside, how do you talk to the FXL6408 GPIO expander? Does the firmware implement bit banging on a dedicated pair of GPIOs?

ptesarik, 2019-08-29:
FWIW it all boils down to the fact that J3-13 is short-circuited to J4-11 and J3-14 to J4-12, so the camera I²C cannot be de-coupled from the display I²C, whatever the pin configuration. In effect, both of them must be controlled by the firmware; you can't let the firmware control one of them (eg. camera) and Linux the other one (eg. display).