raspberrypi/firmware #1142: Under-voltage detected on Raspberry Pi 3

rohitg76, 2019-05-20:
Recently I noticed the following in the startup log of my RPi3. It has coincided with rpi-update on it - hence, I am unsure as to what may have been the root cause of WiFi stopping to work, and under voltage warnings being reported, and SD card coming up at only 25Mhz.

(with following params in config.txt)
total_mem=1024
gpu_mem=64
arm_freq=1350
gpu_freq=500
core_freq=500
sdram_freq=500
sdram_schmoo=0x02000020
over_voltage=4
sdram_over_voltage=2
dtparam=sd_overclock=100

**dmesg output lists the following**

[  149.911657] Under-voltage detected! (0x00050005)
[  160.311690] Voltage normalised (0x00000000)
[  166.551633] Under-voltage detected! (0x00050005)
[  170.711658] Voltage normalised (0x00000000)
[  310.071465] rpi_firmware_get_throttled: 6 callbacks suppressed
[  310.071472] Under-voltage detected! (0x00050005)
[  328.791509] Under-voltage detected! (0x00050005)
[  343.351410] Under-voltage detected! (0x00050005)
[  428.631334] rpi_firmware_get_throttled: 14 callbacks suppressed
[  428.631341] Voltage normalised (0x00000000)
[  457.751265] Voltage normalised (0x00000000)
[  482.711234] Voltage normalised (0x00000000)
[  611.671143] rpi_firmware_get_throttled: 16 callbacks suppressed
[  611.671150] Under-voltage detected! (0x00050005)
[  619.991053] Under-voltage detected! (0x00050005)
[  628.311152] Under-voltage detected! (0x00050005)
[  730.230915] rpi_firmware_get_throttled: 14 callbacks suppressed
[  730.230921] Voltage normalised (0x00000000)

**But more importantly, the WiFi proc is seeming to have trouble. Hostapd is failing to start.**

[    4.675790] usbcore: registered new interface driver uas
[    4.725432] scsi 0:0:0:0: Direct-Access     USB TO I DE/SATA Device   0016 PQ: 0 ANSI: 4
[    4.727282] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[    4.727299] sd 0:0:0:0: [sda] 0-byte physical blocks
[    4.728139] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[    4.728941] sd 0:0:0:0: [sda] Asking for cache data failed
[    4.728957] sd 0:0:0:0: [sda] Assuming drive cache: write through
[    4.736961] sd 0:0:0:0: [sda] Attached SCSI disk
[    4.889297] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    4.913910] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    4.965735] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.979833] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    4.979864] cfg80211: failed to load regulatory.db
[    5.052221] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    5.057618] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.057926] usbcore: registered new interface driver brcmfmac
[    5.245972] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    5.245985] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    5.249556] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    5.249568] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    5.252283] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.252385] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    5.253142] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    5.253842] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    5.253850] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    5.593679] usb 1-1.5: new high-speed USB device number 5 using dwc_otg

Could setting never_over_voltage=1 may have caused an irreversible change of this nature?

Any pointers appreciated. Thx.

Rohit

P33M, 2019-05-21:
It's far more likely that you have a marginal power supply - one that has either drifted in output voltage or a slight increase in current consumption on the Pi has caused the output voltage to droop.

rohitg76, 2019-05-21:
I am using the Stontronics psu for this. It is a 2.5A 5.1V psu and the same
is able to power another RPi3 with a different SD card at the throttled
voltages. No errors or warnings appearing there. I have tried removing all
deviations from standard yet the same behavior with this board + sd
combination. This happened after an rpi-update and I used
never_over_voltage=1 when I was trying something out. Unable to see why
this should brick the wifi proc.

Thanks for your help.

popcornmix, 2019-05-21:
If you think this coincided with an rpi-update, then can you revert to an earlier version of firmware and see if issue disappears? See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this issue.

rohitg76, 2019-05-21:
Thanks for the suggestion @popcornmix. I reverted back to FW level corresponding to kernel 4.9.35 and as I suspected, the voltage warnings are gone and WIFI is back working. The SD (16GB SDHC) continues to be set to 25Mhz, despite the revised parameters and old FW, which was earlier able to operate at 100Mhz after throttling, prior to rpi-update. Will incrementally up the FW level and share the FW that breaks.
[image]

rohitg76, 2019-05-21:
That said, rpi-update to the latest FW on another RPi3 running with same psu has not introduced the under-voltage warnings or impacted WiFi proc the way it did on this board. And not sure how and why a hitherto working mSD at 100Mhz would downgrade to 25Mhz - not even run at the base HS of 50Mhz (default).

Strange mentions in dmesg which were not reported earlier (prior to rpi-update, but now even after reverting to old FW, still being reported)

[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.772204] mmc-bcm2835 3f300000.mmc: could not get clk, deferring probe
[    0.774671] sdhost-bcm2835 3f202000.mmc: could not get clk, deferring probe
[    0.833792] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    0.836322] mmc-bcm2835 3f300000.mmc: DMA channel allocated
[    0.930992] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    0.934989] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    0.938847] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    0.943810] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    1.041429] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    1.049065] Waiting for root device /dev/mmcblk0p2...
[    1.108725] mmc0: host does not support reading read-only switch, assuming write-enable
**[    1.114743] mmc0: Problem switching card into high-speed mode!**
**[    1.117866] mmc0: new SDHC card at address 0001** _(this should have also said - new high speed SDHC...)_
[    1.120647] mmcblk0: mmc0:0001 SRTR 14.9 GiB
[    1.124871]  mmcblk0: p1 p2
**[    1.140673] mmc1: new high speed SDIO card at address 0001**
[    1.151036] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    4.747745] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)

Will report more as I find more.

popcornmix, 2019-05-21:
If you can identify the first firmware commit with the issue it may shed some light.
4.9.35 is nearly two years old. I have a recollection that under-voltage detection wasn't working a couple of years back, so you may have just stumbled upon that build.

rohitg76, 2019-05-22:
OK - this is the result of a quick and dirty job that took me two RPi3 and about 8 hours or repetitive rpi-updates. It can be more thorough if I can get the time again. Btw, I did try to look at fixes that are indicated to have gone in, but that was inconclusive to draw any intelligence on which one may have impacted.

There are three problems
(1) WIFI not working with hostapd
(2) Under-voltage errors
(3) mSD card fails to start in high-speed mode (50Mhz) instead started at Basic rate of 25Mhz
NOTE - 16GB SDHC (so far only one card, Strontium UHS1) and 64GB SDXC (consistently for all brand new cards too, tried with two Strontium UHS1 cards) reporting. Tested from 16GB to 256GB cards. Rest are OK.
"[    2.183115] mmc0: Problem switching card into high-speed mode!"

I have not been able to solve #3 even after turning off all throttled tunables. The one 16GB card and consistently all 64GB cards continue to start at 25Mhz only.

For #1 and #2, I have not tested all FWs from 4.9.35 to 4.19.42, but binary folded and narrowed down to following. I was paying more attention to fixing #1 as it is more hampering, #2 is a little erratic in showing up and did not seem to have an impact on noticeable functionality.

**For #1 - WiFi not working with hostapd**
This is the warning/error that I see from dmesg (note in red about 43430) and symptom is hostapd service while shows active, the hostapd process fails. Hence, no WiFi SSID can be seen. This probably could be a case that may be exposing a bug in hostapd due to something fixed in 

In 4.14.22
[image]

In 4.19.23
[image]

Interestingly, this is fluctuating - it showed up in 4.14.22 and around 4.14.70 or so it got fixed but then again showed up in 4.19.23. The last recent FW that does NOT show signs of this error is **4.14.98**.

**For #2 - Under-voltage errors**
Started showing in 4.14.24. Can be noted in dmesg, but not sure if exposes any symptoms. From the things I tried, it did not seem to interfere except for the red indicators.
[image]
For this too, the last recent FW that does NOT show signs of this error is **4.14.98**.

Hope this helps narrowing down to some extent. Let me know if any more tests are needed. I am unable to spend time over next week. Can resume post that. Thanks.

rohitg76, 2019-05-22:
@popcornmix - As a side note, can you advise if upgrading from 4.9.35 to 4.14.98 will yield any improved performance in terms of network throughput or compute or memory utilization? This information will help. Thanks.

rohitg76, 2019-05-22:
Given there are over two years of FW upgrades available from our current level of 4.9.35, I binary folded and tried. Looks like the WIFI issue cropped from 4.19.23 onwards. The last recent stable version that I could run with WIFI not failing is 4.14.98.

For the under-voltage issue, the under-voltage alerts in dmesg are not wrong - it appears only when I switch to 1A supply, goes away when I use 2.5A. So, this is a non-issue.

For the WIFI issue, I initially thought the WIFI issue was related to an alert in red for brc43430 in dmesg, but that seems incorrect - there does not seem to be a correlation since the alert is present in 4.14.98 too but WIFI seems to be working, unlike the 4.19.23 where it did not work. I am finding it difficult to believe that no one else has hit this so far - no one upgraded to 4.19.23 (Feb 2019) and tried using WiFi on the RPi3 - that's very unlikely.

For the mSD card issue - I'm still getting "[    2.183115] mmc0: Problem switching card into high-speed mode!" after downgrading the FW for one 16GB SDHC card (Strontium UHS1) and consistently for all, even brand new, 64GB SDXC cards (Strontium UHS1) which was previously being recognized as high-speed and able to throttle to 100Mhz. Now, even after downgrading FW back to 4.9.35, it throws up this note in dmesg and able to start at basic speed and frequency of 25Mhz. Surprisingly, other 16/32GB SDHC cards and 128/256GB SDXC cards are able to throttle to 100Mhz and 90Mhz respectively. I can understand that something is gone in the one 16GB card unable to get throttled - may have just coincided with rpi-update, but all the brand new 64GB cards also not being recognized as high-speed on this one box which was FW upgraded is a cause of worry.

For now, keeping all our boxes at 4.9.35 - as that has been stable for a long time.

bedaniels, 2020-07-07:
Guys, I'm also having what looks like low power issues.  Updated everything, checked the cables, USB power supply output, new SD card but all I see is a sea of red.  Primary issue is failure to recognise the USB ports even prior to booting the OS with screen error for all four USB ports.

usb 1-1: device descriptor read/64, error -71

Finally managed to SSH and VNC into it as a head end.  I think I need to revert back, but need some guidance there if this is the case.  This is the mess I'm trying to decipher:

pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.50-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1324 SMP Wed Jul 1 17:00:22 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 1896 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49792 r8192 d23936 u81920
[    0.000000] pcpu-alloc: s49792 r8192 d23936 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240792
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=2486647e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 882472K/970752K available (8192K kernel code, 680K rwdata, 2532K rodata, 1024K init, 824K bss, 22744K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 27957 entries in 55 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x328/0x4e8 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000019] Switching to timer-based delay loop, resolution 52ns
[    0.000282] Console: colour dummy device 80x30
[    0.000308] printk: console [tty1] enabled
[    0.000362] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000381] pid_max: default: 32768 minimum: 301
[    0.000710] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000731] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001894] CPU: Testing write buffer coherency: ok
[    0.002447] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003380] Setting up static identity map for 0x100000 - 0x10003c
[    0.003574] rcu: Hierarchical SRCU implementation.
[    0.004249] smp: Bringing up secondary CPUs ...
[    0.005356] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006609] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007756] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007908] smp: Brought up 1 node, 4 CPUs
[    0.007923] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007933] CPU: All CPU(s) started in HYP mode.
[    0.007942] CPU: Virtualization extensions available.
[    0.008875] devtmpfs: initialized
[    0.025456] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.025727] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.025754] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.028769] pinctrl core: initialized pinctrl subsystem
[    0.029903] NET: Registered protocol family 16
[    0.034307] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.040676] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.040688] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.040932] Serial: AMBA PL011 UART driver
[    0.042860] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.060099] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-07-02 15:01, variant start
[... cut ...]
[   11.065983] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   12.787375] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   18.899411] fuse: init (API version 7.31)
[   31.915911] Voltage normalised (0x00000000)
[   40.235938] Under-voltage detected! (0x00050005)
[   48.353686] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead

Any advice appreciated or you think the HW is just screwed?

Cheers.

Ben.

popcornmix, 2020-07-07:
Closing this issue are original bug was due to an insufficient power supply.
@bedaniels you are best off posting on the forum, as this is very unlikely to be a firmware issue. You almost certainly have voltage drop across your cables and the solution will be power supply or power supply USB cable related.

closed 2020-07-07 by popcornmix