raspberrypi/firmware #1143: 1-Wire in Parasite Power configuration (1-Wire using 2 wires) does not work in 4.19.42

brabl2, 2019-05-22:
Hi,
I had working setup of two 1-Wire devices (DS18B20) in parasite mode.
/boot/config.txt: dtoverlay=w1-gpio,gpiopin=4,pullup=on
It stopped working right after the kernel upgrade to the latest version
Linux raspberrypi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
Both sensors are correctly detected by the system, however reporting 85000, which is value after sensor restart. From the symptoms, it looks that the strong pull-up functionality necessary for the parasite mode is not working.
```
pi@raspberrypi ~ $ ls /sys/bus/w1/devices/
w1_bus_master1  28-000000429998  28-000002f1e200

pi@raspberrypi ~ $ cat /sys/devices/w1_bus_master1/28*/w1_slave
50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
50 05 4b 46 7f ff 0c 10 1c t=85000
50 05 4b 46 7f ff 0c 10 1c : crc=1c YES
50 05 4b 46 7f ff 0c 10 1c t=85000
```
Can anybody confirm that 1-Wire devices in parasite mode are working in the latest kernel?
Note: @iz8mbw had similar problem in #348 some time ago.

iz8mbw, 2019-05-22:
Actually I'm running kernel 4.14.87 (#1178 Tue Dec 11 14:32:54 GMT 2018) and 1-Wire works.
Not tested on 4.19.42+ #1219.

adrianhewitt, 2019-05-28:
yup updated my pi zero friday to find my 1 wire DS18B20 reporting in some wildly wrong values (should be closer to 9000. reverted to a backup disk image to double check the sensor wasn't broken, then updated to find the broken values returned)

Linux raspberrypi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux

```
c1 04 4b 46 7f ff 0c 10 a4 : crc=a4 YES
c1 04 4b 46 7f ff 0c 10 a4 t=76062
```

nebman, 2019-06-07:
same for me, had to revert to 4.14 kernel to get correct DS18B20 readings again...  
`sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b` will install 4.14.98

pelwell, 2019-06-11:
I believe I have a fix for this now, but its waiting on the delivery of some new DS18B20s. (Of the two I have, one is on a module with an LED draining the power, and the other appears to be dead.)

pelwell, 2019-06-12:
As in previous w1 failures, the downstream support for Device Tree and module parameters had been dropped due to a merge conflict. The conflict is understandable - most of the work for supporting strong pullup is now standard in the upstream kernel, but the w1-gpio driver had a bug: by failing to ensure the data pin was an output, the attempt to drive the line high was failing. This bug should now be fixed in the source repo.

N.B. The parasitic power feature is now enabled by default, meaning the `pullup` parameter now does nothing; the overlays and documentation have been updated accordingly.

N.B.2. Support for module parameters has been removed pending an outcry.

jdenver13, 2020-04-19:
Just come across this, Im having similar problems and not sure how to solve this:

I have a DS18B20 in parasite power mode on GPIO 4, the pi responds with the correct temperature in the /sys/bus/w1/devices/28.xxxxxxxxxxxxxx/w1_slave.

But in /mnt/1wire/28.xxxxxxxxxxxxx/temperature it shows as fixed at 85.

Im using owserver to report to home assistant and so I get the 85 deg's as the reading.

If I replace the DS18B20 for a non parasite mode, all works correctly.

Any ideas please?

pelwell, 2020-04-20:
How does owserver access the w1 hardware attached to a Pi? Is it using /sys/bus/w1?

jdenver13, 2020-04-20:
I'm not sure, how would I check?
I'm using server: w1 in the owfs.conf if that's what your asking.

pelwell, 2020-04-21:
You're going to have to ask people familiar with owfs - although the Pi isn't the best platform for w1 (without external hardware), the fact that the /sys interface is showing the correct data makes it sound like a configuration issue.

jdenver13, 2020-04-21:
OK, thanks, I've ordered a 1-wire interface board for the pi so will see how that goes

iz8mbw, 2023-03-10:
Hi all @brabl2 @popcornmix 
I'm in progress to migrate my sensors from a Raspberry Pi 1 (Model B) to a Raspberry Pi 3 (Model B+).
On the Raspberry Pi 3 I installed Raspberry Pi OS (Debian11.6) and I'm still able to read my sensors using 1-Wire in Parasite Power configuration.

Some info here:

On file `/boot/config.txt` I have:
```dtoverlay=w1-gpio,gpiopin=4,pullup=on```

`cat /etc/debian_version`
```11.6```

`uname -a`
```Linux rpi3 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux```

and here the correct temperature from the 1-Wire sensor in Parasite Powe:
```
root@rpi3:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
4c 01 ff ff f0 ff ff ff c3 : crc=c3 YES
4c 01 ff ff f0 ff ff ff c3 t=20750
root@rpi3:/sys/devices/w1_bus_master1/3b-00000014fe14#
```

pelwell, 2023-03-10:
That sounds like success. Do you think the issue can be closed?

iz8mbw, 2023-03-10:
I didn't open this issue.
For me yes, 1-Wire in Parasite Power works.

pelwell, 2023-03-10:
> I didn't open this issue.

Understood, but your feedback is very welcome.

closed 2023-03-10 by pelwell

iz8mbw, 2023-03-10:
@pelwell always available to contribute.
Ciao!