raspberrypi/firmware #1144: Bluetooth controllers do not work in 1.20190517-1

adangert, 2019-05-23:
I am the main contributor behind https://github.com/adangert/JoustMania which requires playstation move controllers to pair to the pi. This is done using https://github.com/thp/psmoveapi

In kernel version 1.20190401-1 the controllers correctly paired to the pi, once upgraded to 1.20190517-1 the controllers stop pairing.

The output from btmon is as follows:
[failing_btmon.txt](https://github.com/raspberrypi/firmware/files/3214335/failing_btmon.txt)

The bluetooth info for one of these devices is:

```
[General]
Name=Motion Controller
Class=0x002508
SupportedTechnologies=BR/EDR;
Trusted=true
Blocked=false
Services=00001124-0000-1000-8000-00805f9b34fb;

[DeviceID]
Source=1
Vendor=1356
Product=981
Version=1
```

Likewise the btmon output under 1.20190401-1 looks as follows:
[working_btmon.txt](https://github.com/raspberrypi/firmware/files/3214332/working_btmon.txt)

pelwell, 2019-05-24:
Looking at the logs I can see the working image sends a Write Scan Enable command - the failing image doesn't. The BT firmware shouldn't have changed between those releases, but you help to locate the source of the regression by swapping just the kernel (and VC firmware, but that should be irrelevant) the two images to see if the behaviour is swapped.

To confirm my understanding of the BT firmware versions can you say which platform you are testing on, and report the output of `md5sum /lib/firmware/brcm/BCM434*.hcd` on both images.

We don't recommend using `rpi-update` as a matter of course, but in this situation it is very useful. Releases are identified using git commit IDs as found on [this page](https://github.com/Hexxeh/rpi-firmware/commits/master). I believe the commit corresponding to the kernel in the working image is https://github.com/Hexxeh/rpi-firmware/commit/00bb2894c0a4349464c19ba62f62b8d9e17a1a9e, while the one for the failing kernel is https://github.com/Hexxeh/rpi-firmware/commit/413e9a212ee40a0a2506b403915598a74a90f3ef.

Starting with the failing image, please run `sudo rpi-update 00bb289`, reboot and retest. If that doesn't fix it then it's probably best to stop. If that image starts to work then switch to the working image and run `sudo rpi-update 413e9a2`, reboot and retest. To return to the original state, run the same commands but on the opposite images.

joolswills, 2019-05-24:
From my understanding it's a regression in the kernel 4.19 code with sony-hid

Was brought up here

https://retropie.org.uk/forum/topic/22312/latest-update-and-ps3-sixaxis-controller/

This is the Kernel report - https://bugzilla.kernel.org/show_bug.cgi?id=203643

pelwell, 2019-05-24:
That would have been useful information to put in your original report. The referenced patch is now in the rpi-4.19.y tree (see https://github.com/raspberrypi/linux/commit/00de9e45fe619d9fd3e810d10d99ac328ee5190c).

joolswills, 2019-05-24:
The original report was by someone else.

pelwell, 2019-05-24:
Yes, sorry - thanks for the pointer.

adangert, 2019-05-24:
oh yeah running sudo rpi-update 00bb289 does fix the issue, I will try sudo rpi-update 413e9a2 next.

pelwell, 2019-05-24:
I'm sure it will break it. A fix is in the source tree and will appear in future releases.

adangert, 2019-05-24:
ok thank you.

joolswills, 2019-06-01:
The fix in https://github.com/raspberrypi/linux/commit/00de9e45fe619d9fd3e810d10d99ac328ee5190c is confirmed working thanks.

Would be great if a new kernel/firmware could be packaged up soon with this - would make a lot of RetroPie users happy (We will wait until this is included in Raspbian before we release the upcoming RetroPie 4.5 image).

Thanks very much for backporting the fix.