raspberrypi/firmware #1145: with 4.19.46 dtoverlay=pi3-disable-wifi stopped working

kapitainsky, 2019-05-28:
with upgrade to 4.19.46 I seem not to be able to disable internal wifi card on RPi 3 B+.

Until now dtoverlay=pi3-disable-wifi in /boot/config.txt did the job. With 4.19.46 wifi is on again despite unchanged /boot/config.txt

as I can tell there is no change in Device Tree overlays for wifi (but I can see that overlays were tinkered for other things)

ncguk, 2019-05-29:
Same here. `rpi-update` to `4.19.46-v7+ #1230 SMP Tue May 28 16:18:48 BST 2019 armv7l GNU/Linux`, reboot and do `lsmod`. `brcmfmac`, `brcmutil`, `cfg80211` and `rfkill` modules are loaded despite `dtoverlay=pi3-disable-wifi` being in `config.txt`

pelwell, 2019-05-29:
Yes, I can see how that got broken with the recent DTS/overlay refactor. A fix is on the way.

popcornmix, 2019-06-03:
Latest rpi-update should have the fix.