raspberrypi/firmware #1146: Audio frequency rate error when playing music (slow effect).

nagualcode, 2019-05-28:
I just noticed that with the most recent firmware (archlinux-arm), 
music played with mpd was slowed down, i guess some frequency rate error.
Editing config.txt to:
audio_pwm_mode=0
Solved the issue.

popcornmix, 2019-05-29:
It was okay with previous firmware? What version was previous firmware?

nagualcode, 2019-05-29:
Last version I  was running before update (when pwm_mode 0, 1 and 2 where ok):
Linux moods 4.14.62-1-ARCH #1 SMP Fri Aug 17 05:43:55 UTC 2018 armv7l GNU/Linux
Aug 16 2018 17:30:30 
Copyright (c) 2012 Broadcom
version 31e0613622dc2f2463bf3dd74e6c897d91201a4d (clean) (release)

Version I am running now, where pwm_mode 2 sounds slower (with mpd):

Linux moods 4.19.42-1-ARCH #1 SMP PREEMPT Wed May 15 00:39:47 UTC 2019 armv7l GNU/Linux
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

popcornmix, 2019-05-29:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.