raspberrypi/firmware #1148: Requested feature, consistently provide HDMI connection status

dachshund-digital, 2019-06-01:
Requested feature, consistently provide HDMI connection status.  CEC is not consistent across all environments or devices.  The tvservice command 'remembers' the last device connected, so using -s or -n parameters with tvservice is not applicable.  Using GPIO 46 is not consistent across all models of Pi.

There should be some standard method across all Pi devices to report if or if not cable connected to HDMI port.  Maybe some where under /sys/proc would be the place to show such status?

I discovered this issue testing both tvservice and cec-client.  Moreover cec-client fails to report consistently the slave devices on the HDMI (bus), it will consistently report the source (master), i.e. the Pi, but sometimes the TV or other CEC devices is not reported when a CEC scan is done.  Even though cec-client will blast CEC commands to the missing device, it basically assumes some device is out there.

popcornmix, 2019-06-03:
Try running:
``tvservice -s``
and in the number after state, check for the ``HDMI_ATTACHED`` bit (2).
That will show if HDMI is current attached (i.e. hotplug is asserted).

dachshund-digital, 2019-06-10:
Where is the status bit map documented?

popcornmix, 2019-06-10:
https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_hdmi.h#L471

dachshund-digital, 2019-06-10:
Thank you very much!

closed 2019-06-10 by dachshund-digital