raspberrypi/firmware #1150: Bluetooth: hci0: Frame reassembly failed (-84)

risa2000, 2019-06-05:
After the last system upgrade of Raspbian to:
```
pi@hass:~ $ uname -a
Linux hass 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
```
I am observing a problem with bluetooth, which, after some time, stops working. The kernel log shows repeating message

> Bluetooth: hci0: Frame reassembly failed (-84)

or sometimes also a stack dump (I do not have one right now).

The RPi is 3B the dmesg log is here: [dmesg.log](https://github.com/raspberrypi/firmware/files/3256994/dmesg.log)

The HCI configuration is:
```
pi@hass:~ $ hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:A1:A8:DE  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:1733551 acl:20475 sco:1 events:67235 errors:0
        TX bytes:387061 acl:5214 sco:0 commands:27972 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'hass'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 4.1 (0x7)  Revision: 0x168
        LMP Version: 4.1 (0x7)  Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)
```
Now my BT usage is a bit specific. The main role for this RPi is running homeassistant, which means (among others) having: USB flash stick (with `f2fs` for logging), Z-Wave USB stick for home automation, Zigbee USB stick for home automation, and also using built-in BT for tracking BT hygrothermo devices.

Apart from that, I also (ab)use the built-in BT for controlling Valve's lighthouses. Both the hygrothermo devices and the lighthouses use BTLE protocol. HT devices are read-out every 2 minutes, the lighthouses (when running) are talked to every 20 seconds.

When I do not run the lighthouses there is no communication with them and the error seems far less likely to happen. When I run a VR session, and run the lighthouse control, after some time, the BT becomes unresponsive, the errors are logged in the kernel log and the only resolution is a reboot.

Before the last system (and I assume also the firmware) upgrade, the system worked, in the exactly same configuration, fine, for several months.

pelwell, 2019-06-05:
The 3B doesn't have the flow control signals hooked up to the BT modem - we were short of pins - so it is possible for characters to get lost under load (especially interrupt load). What sort of data rate is typical for your application? If the rate is low, you may find that reducing the baud rate by editing /usr/bin/btuart gets a more stable connection.

risa2000, 2019-06-09:
@pelwell I believe I can go with lower rate, my use is light and also nothing happens if one packet is lost. 

What however bothers me (and I am not sure it is actually evident from the message as I am not familiar with BT) is not the intermittent corruption of the data, but the fact that BT becomes completely unresponsive.

This is what started to happen (I believe) with the latest update. I had some missed comms before, but then, the comm recovered the next time the connection was retried and it never led to the state where I need to reboot the RPi to get BT working again.

J-Luc16, 2019-07-02:
>built-in BT for tracking BT hygrothermo devices.
Same sort of application (temp+hygro for a plant dryer), same issue since the last kernel update (may ?) on my PI3,

I have put an USB dongle and it works, but I am not happy with this solution, the integrated BTLE of PI3 **was stable for months before**, now it crashes within one day or 2.

GorkaMM, 2019-07-04:
It stopped working for me too on my Raspberry Pi 3 after the update.

codyc1515, 2019-07-06:
I am having this issue as well and it's just continuous of the following.

[98613.228402] Bluetooth: hci0: SCO packet for unknown connection handle 23967
[98659.629015] Bluetooth: hci0: Frame reassembly failed (-84)
[98659.629051] Bluetooth: hci0: hardware error 0xc5

My version is Linux hoobs 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux.

codyc1515, 2019-07-10:
This worked for me:

sudo rpi-update c50c65e52bad69e8fa7d6576d9e2399eade7faf6

J-Luc16, 2019-07-10:
Cool, is this a regression to the previous version of the firmware ?

pelwell, 2019-07-10:
There may be a related issue - https://github.com/raspberrypi/firmware/issues/1017. I've found a driver/hardware bug with a simple workaround which I'm prepared to merge if the real fix doesn't appear soon.

pelwell, 2019-07-11:
A real fix has been found and pushed to rpi-4.19.y - see https://github.com/raspberrypi/linux/commit/9bf5cd2dc5bf4680ec14b372fae5b457c6ccb497

Hopefully a firmware with the fix will be released in the next few days.

milou161185, 2019-08-02:
hi, i have the same problem and the fix "sudo rpi-update c50c65e52bad69e8fa7d6576d9e2399eade7faf6" don't worked for me. Is the new firmware corrected available? If no, is there any other workaround?

bieniu, 2019-08-02:
@milou161185 I had to downgrade kernel to version 4.14.98 to fix this issue.

pelwell, 2019-08-02:
I don't know where you got that commit hash, but it's wrong. Try `sudo rpi-update 43fa687c`, which at least includes the patch.

milou161185, 2019-08-02:
thanks @pelwell but i try "sudo rpi-update 43fa687c" and i have the same error in the log

pelwell, 2019-08-02:
1. Which platform are you on?

2. Which distribution are you running?

3. What are you doing when the error appears?

4. How long does the error take to occur?

milou161185, 2019-08-02:
1 : Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
2 : Raspbian GNU/Linux 9 \n \l
3 : nothing, just after booting
4 : these errors appeared after an update

pelwell, 2019-08-02:
5. Which model of Raspberry Pi?

6. And they happen after every reboot?

Please paste the output of dmesg after booting.

milou161185, 2019-08-05:
5 : Pi model 3B
6 : yes after every reboot

dmesg :

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.58-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1245 SMP Fri Jul 12 17:25:51 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] On node 0 totalpages: 225280
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 870068K/901120K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 22860K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
                   lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26351 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000286] Console: colour dummy device 80x30
[    0.000906] console [tty1] enabled
[    0.000962] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001008] pid_max: default: 32768 minimum: 301
[    0.001365] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001401] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002343] CPU: Testing write buffer coherency: ok
[    0.002826] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003500] Setting up static identity map for 0x100000 - 0x10003c
[    0.003667] rcu: Hierarchical SRCU implementation.
[    0.004484] smp: Bringing up secondary CPUs ...
[    0.005345] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006269] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007131] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007251] smp: Brought up 1 node, 4 CPUs
[    0.007334] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007358] CPU: All CPU(s) started in HYP mode.
[    0.007378] CPU: Virtualization extensions available.
[    0.008361] devtmpfs: initialized
[    0.021351] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.021626] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.021674] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.022283] pinctrl core: initialized pinctrl subsystem
[    0.023158] NET: Registered protocol family 16
[    0.026120] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.031782] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 202 lines cut ...]
[    6.526241] brcmfmac: power management disabled
[    7.244336] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.244613] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.333443] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    8.391995] Under-voltage detected! (0x00050005)
[   10.410110] Bluetooth: Core ver 2.22
[   10.410164] NET: Registered protocol family 31
[   10.410168] Bluetooth: HCI device and connection manager initialized
[   10.410181] Bluetooth: HCI socket layer initialized
[   10.410188] Bluetooth: L2CAP socket layer initialized
[   10.410212] Bluetooth: SCO socket layer initialized
[... cut ...]
[   11.124447] Bluetooth: BNEP filters: protocol multicast
[   11.124477] Bluetooth: BNEP socket layer initialized
[   11.257837] Bluetooth: RFCOMM TTY layer initialized
[   11.257876] Bluetooth: RFCOMM socket layer initialized
[   11.257913] Bluetooth: RFCOMM ver 1.11
[   18.066574] fuse init (API version 7.27)
[   23.541260] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[   27.111915] Voltage normalised (0x00000000)
[   31.272096] Under-voltage detected! (0x00050005)

pelwell, 2019-08-05:
When your Pi has a marginal power supply (or too much load) as yours seems to, all sorts of errors can happen that wouldn't normally.

milou161185, 2019-08-05:
a screen is connected to the pi, I will try by unplugging it to see if the messages are disappearing

milou161185, 2019-08-05:
the error " Under-voltage detected! " is gone but I still have error messages "Bluetooth: hci0: Frame reassembly failed (-84)" with no screen connected

pelwell, 2019-08-05:
Going back to your answer to question 3, an error like the one you are seeing is unlikely on an idle system. Which Bluetooth devices are connected?

milou161185, 2019-08-05:
I use the bluetooth of pi as antenna for jeedom, so there is 2 nut connected

milou161185, 2019-08-05:
by stopping the service making the connection between the pi and jeedom, no error ... I will therefore get closer to the forum jeedom, thank you for your help!

pelwell, 2019-08-05:
What is a "nut"? Do you mean the Nut Mini device tracker?

How large and CPU intensive is Jeedom? Unlike the later Bluetooth Pis, Pi 3B doesn't have flow control signals to the Bluetooth modem which means it can lose data under load. One option is to follow the suggestion I gave earlier to reduce the baud rate to the modem - edit /usr/bin/btuart and change the 921600 to 460800 (or 230400, 115200 etc.). Unless you are streaming audio it shouldn't be too restricting.

milou161185, 2019-08-05:
thanks, i try this

milou161185, 2019-08-05:
same problem :(

pelwell, 2019-08-05:
I don't think this is an acceptable solution, but to help me understand the problem please try with 115200 if you haven't already.

furiaceka87, 2019-08-06:
What is 115200?
Is it a Git Hash for a new firmware?

popcornmix, 2019-08-06:
Baud rate of BT uart.

pelwell, 2019-08-06:
> >  edit /usr/bin/btuart and change the 921600 to 460800 (or 230400, 115200 etc.). 

> > I don't think this is an acceptable solution, but to help me understand the problem please try with 115200 if you haven't already.

> What is 115200?

It's a very low baud rate to put in /usr/bin/btuart.

furiaceka87, 2019-08-06:
At the moment I had:

>                $HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
>         else
>                 $HCIATTACH /dev/serial1 bcm43xx 921600 noflow - $BDADDR
>         fi
> else

        $HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR

Can I divide everything by a two factor?

pelwell, 2019-08-06:
After following my instructions earlier you should have had a file like this:
```
...

if [ "$uart0" = "$serial1" ] ; then
	uart0_pins="`wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' '`"
	if [ "$uart0_pins" = "16" ] ; then
		# This line is necessary to sort out the flow control pins
		stty -F /dev/serial1 115200 raw -echo
		$HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
	else
		$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
	fi
else
	$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
fi
```
where the first `460800` used to say `921600`. If you haven't tried that yet, please do so. If you have tried that, please change that first 460800 (or 921600 if you've changed it back) into `115200` and try again. It should look like this:
```
...

if [ "$uart0" = "$serial1" ] ; then
	uart0_pins="`wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' '`"
	if [ "$uart0_pins" = "16" ] ; then
		# This line is necessary to sort out the flow control pins
		stty -F /dev/serial1 115200 raw -echo
		$HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
	else
		$HCIATTACH /dev/serial1 bcm43xx 115200 noflow - $BDADDR
	fi
else
	$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
fi
```

furiaceka87, 2019-08-06:
Thanks, I try first with `460800` and later with `115200` and I let you know the status after few hours.
Meanwhile I've updated in my message above some information about my configuration

milou161185, 2019-08-06:
I made the change as requested and I have no more errors. can this change have other impacts?

pelwell, 2019-08-06:
It reduces the maximum data rate you can send over Bluetooth, but I don't think your application requires a high bandwidth. It will have no effect on anything else.

It seems that a change between the 4.14 and 4.19 kernel has affected scheduling in a way that makes it more likely that some UART data is dropped because the ARM can't empty the UART FIFO in time. By halving the baud rate we are doubling the amount of time available to respond, making data loss much less likely.

milou161185, 2019-08-06:
ok, thank you so much for your help and your reactivity !!

pelwell, 2019-08-06:
For applications where Bluetooth reliability is important I would recommend a Zero W, 3B+ or 4B, but I understand why changing now might not appeal.

milou161185, 2019-08-06:
I will watch the price of 4 or at least 3+ right now!

pelwell, 2019-08-06:
Some people use external USB dongles - just search for "usb bluetooth dongle for raspberry pi".

J-Luc16, 2019-08-07:
> 		$HCIATTACH /dev/serial1 bcm43xx 3000000 flow - $BDADDR
> 	else
> 		$HCIATTACH /dev/serial1 bcm43xx 115200 noflow - $BDADDR
> 	fi
> ```

115200 is perfect for me too (telemetry) and seems to work like a charm. I have cancelled the kernel hack and put back the last kernel, all is working perfectly until now.

Thanks a lot !

PS : the external dongle solved, obviously, the issue but as my Pi is a PI3, the goal was to use it "naked".

pymumu, 2019-08-29:
same problem

Pickel, 2019-09-02:
pelwell could you clarify the gain in going from a 3B to a 3B+ in this regard? 
I cannot see the difference in architecture on the bluetooth side or is the clock difference the point here?

popcornmix, 2019-09-02:
> pelwell could you clarify the gain in going from a 3B to a 3B+ in this regard?
> I cannot see the difference in architecture on the bluetooth side or is the clock difference the point here?

From earlier in this issue:

> Unlike the later Bluetooth Pis, Pi 3B doesn't have flow control signals to the Bluetooth modem which means it can lose data under load

pymumu, 2019-09-15:
This issue seems to occur in system that upgraded from stretch.

Eistee82, 2019-10-03:
Is this the solution? [https://forum.fhem.de/index.php/topic,75559.msg980043.html#msg980043](url)

pelwell, 2019-10-03:
That (German) forum post contains my suggestion from a few comments back (https://github.com/raspberrypi/firmware/issues/1150#issuecomment-518648032) that on a 3B you may benefit from reducing the baud rate to the BT modem. This is only going to be acceptable if your application doesn't require a high data throughput.

risa2000, 2019-12-02:
Just a heads up for anyone facing this issue.

I did the mod of `/usr/bin/btuart` suggested by @pelwell and it was fine until some recent system update, which apparently overwrote the file. I am not sure if it was a full distro upgrade (Stretch -> Buster) or just a regular `apt-get dist-upgrade`, but I had to rewrite it again.

Unfortunately I noticed that only after I saw some BT failures, which did not recover.

akomelj, 2020-01-10:
> This issue seems to only occur in system that upgraded from stretch.

Not really. Fresh Buster install here...

lucagiove, 2020-01-26:
> 		$HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
Did the change on HassOS with Raspberry 3 and seems to work

lucagiove, 2020-01-27:
@pelwell there are lots of complains on bluetooth problems with Raspberry 3 for many different sensors in home assistant other with raspbian.
Do you think it's kernel or firmware related? Don'y you think would be better to submit a patch to btuart in order to reduce the baudrate?

pelwell, 2020-01-27:
Just this evening I found a possible explanation for the main UART dropping bytes, even with flow control enabled. Assuming the fix works, I'd expect it resolve at least some of the issues people are seeing.

lucagiove, 2020-01-27:
> Just this evening I found a possible explanation for the main UART dropping bytes, even with flow control enabled. Assuming the fix works, I'd expect it resolve at least some of the issues people are seeing.

cool,

pelwell, 2020-01-29:
A patch has just gone into rpi-4.19.y: https://github.com/raspberrypi/linux/commit/65aa6ec0faaa012508489886ac357cbb86cdb9a4
It shouldn't break anything, and I think there's a good chance the data loss is fixed (although there may be a better implementation - this is more of a workaround).

lucagiove, 2020-01-29:
Cool thanks @pelwell I can test it!
What the fastest way to run this kernel on hassos? Or easier with raspbian?

pelwell, 2020-01-29:
There's likely to be a firmware build by the end of the day that will be available via rpi-update. Or you can [build it yourself](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

popcornmix, 2020-01-30:
rpi-update contains the potential fix. Can you test it?

lucagiove, 2020-01-31:
Unfortunately I'm abroad until Monday and as I said the environment where I can easily riproduce the bug is with home assistant distro hassos that does not have rpi-update I've to figure out how to update the firmware there. If someone have ideas..

talondnb, 2020-02-03:
> > ```
> >   $HCIATTACH /dev/serial1 bcm43xx 460800 noflow - $BDADDR
> > ```
> 
> Did the change on HassOS with Raspberry 3 and seems to work

Can you please explain how this was done?

Xitro01, 2020-02-03:
Having this exact same issues, update from 4.19.86-v7+ to 4.19.97-v7+ (fdb5c37e330e7cb3027ac4fcc5b1cd5f244b351f).
The Frame reassembly failed issue is still there, it keeps spamming my syslog with these messages, but I'll check if it stops crashing now.

Xitro01, 2020-02-04:
This problem is not solved yet with the latest firmware! Same issues occur, my bluetooth hardware crashed 8 hours ago.

[80 more replies]