raspberrypi/firmware #1151: h264_omx encoder: Occasional low bitrate video

jasaw, 2019-06-07:
**Description:**
When it's working correctly, h264_omx (via ffmpeg) output video has bitrate of a few Mbps, but very occasionally the output video has bitrate of a few hundred kbps.
Quite a lot of [motionEyeOS](https://github.com/ccrisan/motioneyeos) users are seeing occasional low bitrate video from the h264_omx encoder. I'm seeing this problem about once a week, but don't know how to reproduce it reliably. Many users have reported the issue here: https://github.com/ccrisan/motioneyeos/issues/1900

**System setup:**
MotionEyeOS is a security camera software that runs [motion](https://github.com/Motion-Project/motion) software. Motion software calls ffmpeg API to encode video using h264_omx encoder. I have been seeing this issue since July 2017 with GPU firmware version 3202f1b16896029f9da1b074b0912177e8960b52. We have been upgrading the GPU firmware regularly and the problem persists.

I have checked motion source code, but can't see anything obviously wrong with it. Motion also records videos correctly when other encoders are used, e.g. libx264. At this stage, I'm inclined to say the issue is either in ffmpeg's h264_omx code, or the GPU firmware.

Any thoughts on what might be the cause?

6by9, 2019-06-07:
CBR or VBR?

I haven't looked at it recently, but my recollection is that bitrate control is based on the xFramerate set on the OUTPUT port. If that is 0, then it drops back to using the timestamp on the individual frames to estimate the framerate. Bits are then allocated based on the previous few frames and the number of bits it has left in the bitrate average.

A quick check would imply that FFmpeg is setting xFramerate on both input and output ports, but it might be worth checking that. Logging the timestamps on the OMX buffers would be sensible too - they should be in microseconds for OMX.

I'm assuming you're still patching ffmpeg's omx encoder due to it not handling fragmented buffers sensibly as you contributed to the issue previously - https://github.com/raspberrypi/firmware/issues/1087

jasaw, 2019-06-12:
I'm not sure if it's CBR or VBR. Motion software sets the `bit_rate` field in the AVCodecContext struct, which ends up in `out_port_params.format.video.nBitrate` in ffmpeg OMX.

Regarding the `xFramerate`, it looks like it's set to a non-zero value.
Motion sets the `time_base` like this:
```c
    ffmpeg->ctx_codec->time_base.num = 1;
    ffmpeg->ctx_codec->time_base.den = ffmpeg->fps;
```
ffmpeg OMX eventually applies `time_base` to the xFramerate on OUTPUT port.
```c
    if (avctx->framerate.den > 0 && avctx->framerate.num > 0)
        in_port_params.format.video.xFramerate = (1 << 16) * avctx->framerate.num / avctx->framerate.den;
    else
        in_port_params.format.video.xFramerate = (1 << 16) * avctx->time_base.den / avctx->time_base.num;

    out_port_params.format.video.xFramerate    = in_port_params.format.video.xFramerate;
```

I do have the OMX fragmented buffer patch applied, but this low bitrate issue happens regardless of this patch.

Looks like logging the OMX timestamp is the only way to debug this issue? You're talking about `nTimeStamp` in `OMX_BUFFERHEADERTYPE` ? How can I tell that it's not working properly from this timestamp?
To be clear, you're talking about this timestamp in ffmpeg OMX code right?
```c
        // Convert the timestamps to microseconds; some encoders can ignore
        // the framerate and do VFR bit allocation based on timestamps.
        buffer->nTimeStamp = to_omx_ticks(av_rescale_q(frame->pts, avctx->time_base, AV_TIME_BASE_Q));
```

6by9, 2019-06-12:
It'd be worth adding logging to confirm that ```out_port_params.format.video.xFramerate``` is actually set to a sane value. Plumbing can easily go wrong.

Yes, log ```buffer->nTimeStamp```. Or add a ```static OMX_TICKS last_timestamp``` and do a check that the difference between the new and last timestamp is sane (eg ~33000000 if 30fps).

One query is whether the timestamps end up wrapping around at any level.
```OMX_TICKS``` is a 64bit microsecond counter so shouldn't overflow within a sensible lifetime.
There is a slight inconsistency in ```OMX_TICKS``` though. With ```OMX_SKIP64BIT``` not defined it is ```OMX_S64```, but with it defined both bits are unsigned
```
typedef struct OMX_TICKS
{
   OMX_U32 nLowPart;
   OMX_U32 nHightPart;
} OMX_TICKS;
```
FFmpeg uses int64_t, therefore passing negative timestamps could give odd results due to sign extension.

Other than for rate control in the absence of a framerate, the codec itself doesn't use the timestamps.

jasaw, 2019-08-29:
I'm pretty sure this is caused by integer overflow in motion software which has been fixed, so yes it's plumbing issue. Sorry about the noise. I'll close this issue for now.

closed 2019-08-29 by jasaw