raspberrypi/firmware #1152: HASSIO Bluetooth error

bruvv, 2019-06-08:
Running the latest update:
```
Linux raspberrypi 4.19.46-v7+ #1231 SMP Mon Jun 3 17:42:48 BST 2019 armv7l GNU/Linux
```

and latest hassio and getting an error with bluetooth:
```
Error looking up Bluetooth device
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 31, in discover_devices
    lookup_class=lookup_class, device_id=device_id)
_bluetooth.error: (110, 'Operation timed out')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 102, in update_bluetooth_once
    for dev in discover_devices():
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 57, in discover_devices
    lookup_class=False, device_id=device_id)
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 34, in discover_devices
    raise BluetoothError ("error communicating with local "
bluetooth.btcommon.BluetoothError: error communicating with local bluetooth adapter
```

bruvv, 2019-06-08:
cross posted: https://github.com/home-assistant/home-assistant/issues/24206

peetereczek, 2020-02-03:
Some update, with HA 104.3 and HASS.IO 195 on Raspberry Pi 4B error still occurs:

```txt
Mon Feb 03 2020 17:14:11 GMT+0100 (Central European Standard Time)
Error looking up Bluetooth device
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 31, in discover_devices
    lookup_class=lookup_class, device_id=device_id)
_bluetooth.error: (4, 'Interrupted system call')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 140, in perform_bluetooth_update
    devices = await hass.async_add_executor_job(discover_devices, device_id)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 62, in discover_devices
    device_id=device_id,
  File "/usr/local/lib/python3.7/site-packages/bluetooth/bluez.py", line 34, in discover_devices
    raise BluetoothError ("error communicating with local "
bluetooth.btcommon.BluetoothError: error communicating with local bluetooth adapter
```

closed 2020-04-12 by bruvv