raspberrypi/firmware #1153: Intermittent mmal_vll_load issue on Pi-Zero-W using raspivid

budius, 2019-06-17:
I'm using `raspistill` in a script that loop records in 10 minutes blocks of videos, but I keep getting random failures. When the fails happens `raspistill` still takes 10 minutes to complete, but the h264 file is only 29-bytes in size.

Below some system info and logs. I'll be happy to provide any more information as needed

- in `raspi-config` is selected 256M to the GPU
- raspistill command is: `raspivid -t 600000 -b 5000000 -fps 15 -w 1640 -h 1232 -o /home/picam/remote/video_"$(date +%Y_%h_%d_%Hh%Mm%Ss)".h264`
- `uname -a`: Linux picam 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
- `sudo vcdbg log msg`:
```
001828.333: dtparam: i2c_arm=off
001839.848: dtparam: spi=off
001849.652: dtparam: audio=on
002791.621: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
002814.648: Device tree loaded to 0xffea000 (size 0x5f32)
002816.232: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
004660.907: vchiq_core: vchiq_init_state: slot_zero = 0x4f580000, is_master = 1
004666.394: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
013425.849: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
013726.629: camsubs: Camera not found
013726.714: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
014027.493: camsubs: Camera not found
014027.576: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
014029.252: camsubs: Camera found OK
038482.216: brfs: File read: 1611 bytes
038482.282: mmal: mmal_vll_load: could not load VLL 'videnc.vll': 

```
- `sudo vcdbg reloc`:
```
Relocatable heap version 4 found at 0x10000000
total space allocated is 236M, with 232M relocatable, 4.5M legacy and 0 offline
2 legacy blocks of size 2359296

free list at 0x1d14e1e0
192M free memory in 10 free block(s)
largest free block is 191M bytes

0x10000000: legacy block 2.3M
0x10240000: legacy block 2.3M
0x10480000: free 191M
[  51] 0x1c38e0c0: used  64K (refcount 1 lock count 0, size    65536, align   32, data 0x1c38e0e0, d1rual) 'ril mem ril.video_encod-2'
[  54] 0x1c39e100: used 3.0M (refcount 1 lock count 0, size  3168256, align 4096, data 0x1c39f000, d1ruAl) 'venc_recon'
[  55] 0x1c6a4920: used 3.0M (refcount 1 lock count 0, size  3168256, align 4096, data 0x1c6a5000, d1ruAl) 'venc_recon'
[  56] 0x1c9ab140: used 1.5M (refcount 1 lock count 0, size  1544199, align 4096, data 0x1c9ac000, d3rual) 'venc_mem_alloc'
[  57] 0x1cb25180: used 1.1M (refcount 1 lock count 0, size  1118208, align 4096, data 0x1cb26000, d1ruAl) 'video_encodeRIL: subsample pool'
[  58] 0x1cc371a0: used 1.1M (refcount 1 lock count 0, size  1118208, align 4096, data 0x1cc38000, d1ruAl) 'video_encodeRIL: subsample pool'
[  59] 0x1cd491c0: used 1.1M (refcount 1 lock count 0, size  1118208, align 4096, data 0x1cd4a000, d1ruAl) 'video_encodeRIL: subsample pool'
[  60] 0x1ce5b1e0: used 1.1M (refcount 1 lock count 0, size  1118208, align 4096, data 0x1ce5c000, d1ruAl) 'video_encodeRIL: subsample pool'
[  43] 0x1cf6d200: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x1cf6d220, d0rual) 'stabilise current image'
[  45] 0x1cf71240: used  48K (refcount 0 lock count 0, size    49152, align   32, data 0x1cf71260, d0rual) 'stabilise ref'
[  47] 0x1cf7d280: used  48K (refcount 1 lock count 0, size    49152, align   32, data 0x1cf7d2a0, d0rual) 'stabilise ref'
0x1cf892c0: free 436K
[  37] 0x1cff6180: used  52K (refcount 1 lock count 0, size    53504, align    4, data 0x1cff61a0, d1rual) 'AWB State (makesum_direct)'
0x1d0032c0: free 55K
[  52] 0x1d010e80: used 7.1K (refcount 1 lock count 0, size     7200, align    4, data 0x1d010ea0, d0rual) 'ilfifo header'
[  61] 0x1d012ae0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012b00, d1rual) 'ril.video_encode:in:0(OPQV)'
[  62] 0x1d012ba0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012bc0, d1rual) 'ril.video_encode:in:0(OPQV)'
[  67] 0x1d012c60: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012c80, d1rual) 'ril.video_encode:in:0(OPQV)'
[  63] 0x1d012d20: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012d40, d1rual) 'ril.video_encode:in:0(OPQV)'
[  64] 0x1d012de0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012e00, d1rual) 'ril.video_encode:in:0(OPQV)'
[  65] 0x1d012ea0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012ec0, d1rual) 'ril.video_encode:in:0(OPQV)'
[  66] 0x1d012f60: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d012f80, d1rual) 'ril.video_encode:in:0(OPQV)'
[  19] 0x1d013020: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d013040, d1rual) 'ril.video_encode:in:0(OPQV)'
[  41] 0x1d0130e0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d013100, d1rual) 'ril.video_encode:in:0(OPQV)'
[  42] 0x1d0131a0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d0131c0, d1rual) 'ril.video_encode:in:0(OPQV)'
0x1d013260: free 137K
[  53] 0x1d035680: used  160 (refcount 1 lock count 0, size       96, align    4, data 0x1d0356a0, d0rual) 'ilfifo state'
0x1d035720: free 70K
[  68] 0x1d0470a0: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d0470c0, d1rual) 'ril.video_render:in:0(OPQV)'
[  69] 0x1d047160: used  192 (refcount 1 lock count 0, size      128, align   32, data 0x1d047180, d1rual) 'ril.video_render:in:0(OPQV)'
[... 9 lines cut ...]
0x1d047820: corrupt trailer (space 4448 != 4449)
[  29] 0x1d048980: used 135K (refcount 0 lock count 0, size   138240, align    4, data 0x1d0489a0, d0rual) 'isp sw stage buffers'
0x1d048980: corrupt trailer (space 142752 != 138305)
[  87] 0x1d06a5c0: used 271K (refcount 0 lock count 0, size   277512, align    4, data 0x1d06a5e0, d0rual) 'concurrent mdata'
0x1d0ae220: free 640K
[  16] 0x1d14e200: used 9.8M (refcount 1 lock count 0, size 10289152, align 4096, data 0x1d14f000, d1ruAl) 'cameraRIL:input pool'
[  15] 0x1db1f220: used 1.1M (refcount 1 lock count 0, size  1164480, align 4096, data 0x1db20000, d1ruAl) 'cameraRIL: lowres pool'
[  14] 0x1dc3c700: used 1.1M (refcount 1 lock count 0, size  1164480, align 4096, data 0x1dc3d000, d1ruAl) 'cameraRIL: lowres pool'
[  13] 0x1dd59be0: used 1.1M (refcount 1 lock count 0, size  1164480, align 4096, data 0x1dd5a000, d1ruAl) 'cameraRIL: lowres pool'
[  12] 0x1de770c0: used 3.0M (refcount 1 lock count 0, size  3121344, align 4096, data 0x1de78000, d1ruAl) 'cameraRIL: hires stills pool'
[  11] 0x1e1721a0: used 3.1M (refcount 1 lock count 0, size  3214528, align 4096, data 0x1e173000, d1ruAl) 'cameraRIL: hires vid pool'
[  10] 0x1e483e80: used 3.1M (refcount 1 lock count 0, size  3214528, align 4096, data 0x1e484000, d1ruAl) 'cameraRIL: hires vid pool'
[   9] 0x1e795b60: used 3.1M (refcount 1 lock count 0, size  3214528, align 4096, data 0x1e796000, d1ruAl) 'cameraRIL: hires vid pool'
[   5] 0x1eaa7840: used  832 (refcount 1 lock count 0, size      740, align   32, data 0x1eaa7860, d1rual) 'ril mem ril.camera-0'
[   6] 0x1eaa7b80: used  832 (refcount 1 lock count 0, size      740, align   32, data 0x1eaa7ba0, d1rual) 'ril mem ril.camera-0'
[   7] 0x1eaa7ec0: used  832 (refcount 1 lock count 0, size      740, align   32, data 0x1eaa7ee0, d1rual) 'ril mem ril.camera-0'
[   8] 0x1eaa8200: used  832 (refcount 1 lock count 0, size      740, align   32, data 0x1eaa8220, d1rual) 'ril mem ril.camera-0'
[   4] 0x1eaa8540: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1eaa8560, d0rual) 'ILCS VC buffer pool'
[   3] 0x1eaa8780: used 1.3M (refcount 1 lock count 8, size  1382400, align 4096, data 0x1eaa9000, d3rual) 'ARM FB'
[   2] 0x1ebfafa0: used  16K (refcount 1 lock count 0, size    16384, align   32, data 0x1ebfafc0, d0ruAl) 'audioplus_tmp_buf'
[   1] 0x1ebfefe0: used 4.0K (refcount 1 lock count 0, size        0, align 4096, data 0x1ebff000, d1rual) 'camera fast alloc arena'
heap corruption detected
small allocs not requested

```

6by9, 2019-06-17:
Please don't try to interpret data.
```mmal: mmal_vll_load: could not load VLL 'videnc.vll'``` will be present every time you run ```raspivid``` (NOT ```raspistill```). The firmware tries videnc.vll as the generic encoder, and then drops back to the more specific h264enc.vll which is present.

29 bytes would be the H264 header bytes describing the stream. Nothing in the provided logs provides any information as to why no video data has been encoded.
Please add ```start_debug=1``` to /boot/config.txt, reboot, and try to reproduce. If you can reproduce it, then run ```sudo vcdbg log assert``` to ensure that the firmware hasn't raised any complaints. Please post the complete output if anything is logged.

Your resolution of 1640x1232 makes me assume you have a V2.1 imx219 camera module, but that command would equally work on a V1 ov5647 camera module. Use of an extended camera flexi cable can cause signal integrity issues - it is not recommended to go longer than the standard ~10cm ones.

budius, 2019-06-17:
@6by9 thanks for the input.
I'll run it with `start_debug=1`, it's pretty reproducible.
Yes, I got the newer (V2.1) camera and trying to grab the full-frame of it.
I'm using the pi/camera/box [directly from the official reseller here in Germany](https://buyzero.de/products/pi-zero-w-kamera-kit?variant=31239421650). Pretty short cable.
The box is just closed and I'm not moving or touching cables anywhere, reason I was under the impression it could be software issue, but with your comments, it makes sound might be a hardware issue instead.

Anyway, I'll test and report back.

budius, 2019-06-18:
@6by9 I've ran with debug as requested and here's the output from `sudo vcdbg log assert`
```
001278.566: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115 rev 2e98b31
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
014201.197: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1137 rev 2e98b31
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------

I don't know what to make out of it this log, but just for the sake of testing I already purchased a few new cables that should arrive tomorrow. I'll report back when I test them.

budius, 2019-08-06:
This can be closed @6by9 
At the end, it was hardware issue (bad cable). The reason I originally dismissed hardware issue is because it's all inside the enclosing and it was not moving or being touched.

But new cable, it's been running for a weeks and no more 26-byte files.
Thanks.

closed 2019-08-06 by budius