raspberrypi/firmware #1156: hdmi_enable_4k is ignored

kapetanos, 2019-06-24:
It seems that hdmi_enable_4k=1 in config.txt is ignored. With or without the option, my TV still gets a 4K resolution.

pi@raspberrypi:~ $ tvservice -s
state 0xa [HDMI CUSTOM RGB lim 16:9], 3840x2160 @ 30.00Hz, progressive

pi@raspberrypi:~ $ vcgencmd get_config hdmi_enable_4k
hdmi_enable_4k=0

Firmware version:
Jun 20 2019 16:04:31 
Copyright (c) 2012 Broadcom
version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d (clean) (release) (start)

popcornmix, 2019-06-24:
hdmi_enable_4k=1 enables 4kp60 (which requires higher clocks and scrambling).
4kp30 is available without this setting if display's edid reports support..

kapetanos, 2019-06-24:
Thanks. It seems that the documentation at https://github.com/raspberrypi/documentation/blob/master/configuration/hdmi-config.md is not clear about this.

closed 2019-06-24 by kapetanos

popcornmix, 2019-06-24:
@JamesH65 I agree - documentation isn't clear on this option.

JamesH65, 2019-06-24:
Indeed, the docs were written before the change to the functionality - originally that option enabled all variants of 4k. It should really have changed name to hdmi_enable_4kp60 but the change was too close to release. I'll do the docs tomorrow.

JamesH65, 2019-06-25:
Docs have been updated.