raspberrypi/firmware #1157: Loading U-boot on Raspberry Pi Zero W failed

KozikR, 2019-06-24:
Updating start.elf and bootcode.bin to "firmware: Updates for Pi4" (commit Id: 64b5649a41b69d09bfe0ed05448d28a66be3edfd) caused "CACHE: Misaligned operation" warning printed in infinite loop during U-boot starting.
Issue not reproduced on firmware from previous commit "kernel: Bump to 4.19.50" (07937c7d48bcd44cc1015c6257ae2cfa5da51298).

U-boot version: v2019.04.

jonathangray, 2019-06-25:
After updating to firmware with the release tag of 1.20190620 rpi3b also gets stuck in a loop with U-Boot.

MMC:   mmc@7e202000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... OK
mbox: Header response code invalid
bcm2835: Could not configure display
"Synchronous Abort" handler, esr 0x8a000000
elr: b2e262ab67cc3a9b lr : b2e262ab67cc3a9b (reloc)
elr: b2e262aba2fa2a9b lr : b2e262aba2fa2a9b
x0 : 000000003b000000 x1 : 000000003b000000
x2 : 0000000000000040 x3 : 000000000000003f
x4 : 0000000000000008 x5 : 0000000000000000
x6 : 0000000000003500 x7 : 0000000000000015
x8 : 000000003ae00000 x9 : 0000000000000008
x10: 0000000000000000 x11: 0000000000200000
x12: 0000000000000002 x13: 000000003ae00000
x14: 0000000000000000 x15: 00000000ffffffff
x16: 0000000000004110 x17: deec4042bc705879
x18: 000000003af5ade8 x19: 0000000000000002
x20: 000000003af6a200 x21: 0000000000000002
x22: 8226bb2ab0cfad09 x23: aa8eafaaeeae8eab
x24: 8eaeab8f2a2a3ab7 x25: 0000000000000002
x26: 000000003af6a200 x27: 0000000000000002
x28: 8226bb2ab0cfad09 x29: aaa8aaaaeaa286aa

Resetting CPU ...

resetting ...

pelwell, 2019-06-25:
Although we try not to break existing behaviour, we don't officially support U-boot, so you'll need to provide a set of minimal instructions allowing us to reproduce these boot failures.

jonathangray, 2019-06-25:
U-Boot will be required as long as hardware doesn't ship with UEFI.

To reproduce an OpenBSD/arm64 network install disk image can be used
on rpi3b.  dd the following to a uSD card:

https://cdn.openbsd.org/pub/OpenBSD/6.5/arm64/miniroot65.fs

This includes U-Boot 2019.01, u-boot.bin is created from rpi_3_defconfig with
CONFIG_OF_EMBED=y removed and CONFIG_OF_BOARD=y added.

It includes raspberrypi-firmware-1.20190215

SHA1 (bootcode.bin) = c43d24c2dbed7cad569e82395bfc29461fd3e2db
SHA1 (fixup.dat) = cf7ee5379de97479947a3cccf96111c20f483685
SHA1 (start.elf) = 9ae62dd2723f29b20842035c7e8e1f9eedf4dc82

and uses the following config.txt:
arm_64bit=1
enable_uart=1
device_tree_address=0x02600000
kernel=u-boot.bin

With serial attached this disk image boots to an OpenBSD install prompt on rpi3b.

Replacing bootcode.bin fixup.dat and start.elf with the 1.20190620 versions gets
stuck in a loop.
SHA1 (bootcode.bin) = e895819e68e97eec73ad0aa5ecda8a92745dcae5
SHA1 (fixup.dat) = 28e2b354e11f1bde66f963200f068527552319df
SHA1 (start.elf) = 6ba6098877a5b200c5cd10784cad9fae12fa734f

The latest U-Boot master at the time of writing (5eea874b5e989e62519824ad140aa086432d01ee) still has the same problem with the new firmware.

Googulator, 2019-06-25:
Does this also affect the RaspberryPiPkg UEFI bootloader?

satmandu, 2019-06-25:
I'm seeing a similar issue with current ubuntu server arm64 builds and this new bootcode.bin, which also use u-boot, on a 3b+.

You should be able to write the image at http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/eoan-preinstalled-server-arm64+raspi3.img.xz to a sd card, and replace the bootcode.bin, .dat, and .elf files on the fat boot partition to see a similar boot issue.

pelwell, 2019-06-25:
Thanks, that helps. I am aware of a possible problem for some platforms in situations where a Device Tree file is not available, but I doubt that would be the case here.

ggardet, 2019-06-27:
I confirm the problem on RPi3B+

Vogtinator, 2019-06-28:
I hit this issue as well while trying to get u-boot to work on the Pi4. Possible workaround:

```
diff --git a/drivers/video/bcm2835.c b/drivers/video/bcm2835.c
index bc41090aed..4c7962cad8 100644
--- a/drivers/video/bcm2835.c
+++ b/drivers/video/bcm2835.c
@@ -19,13 +19,15 @@ static int bcm2835_video_probe(struct udevice *dev)
 
        debug("bcm2835: Query resolution...\n");
        ret = bcm2835_get_video_size(&w, &h);
-       if (ret)
+       if (ret || w == 0 || h == 0)
                return -EIO;
 
        debug("bcm2835: Setting up display for %d x %d\n", w, h);
        ret = bcm2835_set_video_params(&w, &h, 32, BCM2835_MBOX_PIXEL_ORDER_RGB,
                                       BCM2835_MBOX_ALPHA_MODE_IGNORED,
                                       &fb_base, &fb_size, &pitch);
+       if(ret)
+               return -EIO;
 
        debug("bcm2835: Final resolution is %d x %d\n", w, h);

```

nullr0ute, 2019-07-01:
So I tested this on Fedora because we saw something similar. It's last and I only had a RPi3A+ to hand. If necessary I can test across RPi2/3B/3B+ too.

Testing with firmware "2019-06-21 18:44" with both a ARMv7 and aarch64 image both using U-Boot 2019.04 I got the same result but two slightly different outputs. If I have a HDMI display attached they both boot fine, if I don't I have a crash, the 32 bit is the "CACHE: Misaligned operation", on 64 bit I get the ""Synchronous Abort" handler, esr 0x86000004"

RPi3A+ 64 bit, serial console (no display) U-Boot 2019.04:

<pre>
MMC:   mmc@7e202000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

mbox: Header response code invalid
bcm2835: Could not configure display
"Synchronous Abort" handler, esr 0x86000004
elr: 80000003e2141004 lr : 80000003e2141004 (reloc)
elr: 8000000400000004 lr : 8000000400000004
x0 : 000000001dc00000 x1 : 000000001dc00000
x2 : 0000000000000040 x3 : 000000000000003f
x4 : 0000000000000008 x5 : 000000001da00701
x6 : ffffffff80800000 x7 : 0000000000000002
x8 : 000000001da00000 x9 : 0000000000000008
x10: 0000000000000000 x11: 0000000000200000
x12: 0000000000000002 x13: 000000001da00000
x14: 0000000000000000 x15: 0000000000000020
x16: 000000001df66844 x17: 0000000000000000
x18: 000000001db3ade8 x19: 0000000000000000
x20: 000000001df671fc x21: 8000000000000020
x22: 0000000800028001 x23: 0000000080000008
x24: 0000000000000001 x25: 000000001db467f0
x26: 000000001db46a30 x27: 0000000000000002
x28: 000000001db46a38 x29: 0004000800000000

Resetting CPU ...

resetting ...
</pre>

RPi3A+ 64 bit, serial console (HDMI display) U-Boot 2019.04:

<pre>
boots as expected
</pre>

RPi3A+ 64 bit, serial console (no display) U-Boot 2019.04:

<pre>
MMC:   mmc@7e202000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

mbox: Header response code invalid
bcm2835: Could not configure display
In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
CACHE: Misaligned operation at range [1dfbabe8, 1dfbac00]
<pre>

RPi3A+ 32 bit, serial console (HDMI display) U-Boot 2019.04:

<pre>
boots as expected
</pre>

nullr0ute, 2019-07-01:
To reproduce it's probably enough just to put the U-Boot binary in the VFAT partition and adjust the config.txt. The relevant bits for the Fedora config is basically:

64 bit:
<pre>
[pi3]
kernel=rpi3-u-boot.bin
# Put the RPi3 into 64 bit mode
arm_control=0x200
[all]
# Enable UART
enable_uart=1

32 bit:
<pre>
[pi2]
kernel=rpi2-u-boot.bin
[pi3]
kernel=rpi3_32-u-boot.bin
[all]
# Enable UART
enable_uart=1
</pre>

GoofySpeed, 2019-07-02:
I'm also having the same issue on a Raspberry Pi 3 running Arch Linux Arm aarch64, the following is the serial output during the loop.  Just like @nullr0ute I was able to get it to boot by hooking the HDMI to a monitor, then just remove the monitor after a successful boot.  I was talking with an individual on Reddit about this problem, but they were not able to get it to boot with a monitor attached, [https://www.reddit.com/r/archlinux/comments/c6pzv3/anyone_elses_raspberry_pi_dead/](url).

`U-Boot 2018.07-1 (Jul 18 2018 - 23:27:57 +0000) Arch Linux ARM

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

Failed (-5)
mbox: Header response code invalid
bcm2835: Could not configure display
"Synchronous Abort" handler, esr 0x86000004
elr: 80000003c4d3b004 lr : 80000003c4d3b004 (reloc)
elr: 8000000400000004 lr : 8000000400000004
x0 : 000000003b000000 x1 : 000000003b000000
x2 : 0000000000000040 x3 : 000000000000003f
x4 : 0000000000000008 x5 : 0000000000000000
x6 : ffffffff80800000 x7 : 0000000000000002
x8 : 000000003ae00000 x9 : 0000000000000008
x10: 0000000000200000 x11: 0000000000000000
x12: 0000000000000002 x13: 000000003ae00000
x14: 0000000000000000 x15: 00000000ffffffff
x16: 0000000000004110 x17: fe4fff49ed2df6ec
x18: 000000003af40df0 x19: 0000000000000000
x20: 000000003b36afbc x21: 000000003af44ad8
x22: 0000000000000002 x23: 0004000800000000
x24: 8000000400000004 x25: 000000003af44ad0
x26: 000000003af44890 x27: 000000003af44ad8
x28: 0000000000000002 x29: 0004000800000000

Resetting CPU ...

resetting ...`

Vogtinator, 2019-07-02:
@nullr0ute @GoofySpeed: Can you try the diff I posted?

lategoodbye, 2019-07-02:
The video driver in u-boot relies on the mbox interface. I think the problem is the introduction of multi framebuffer support. This possibly resulted in change in the mbox property 0x00040003 (Get physical display width / height) behavior (https://github.com/raspberrypi/linux/commit/4600e91bb6dd75140bbafb271f30370bc2e33d60#diff-7434d1db58b58fb45a55de66c12bdc1a).

First the documentation of the mbox should updated and then the video driver in u-boot.

6by9, 2019-07-02:
I thought @jamesh had retained the old API too.

JamesH65, 2019-07-02:
I don't remember changing that API, and did put in a lot of effort to retain backwards compatibility. I'll check the mbox call.

JamesH65, 2019-07-02:
Cannot see any changes to that specific property. There are additional properties - 

      SET_DISPLAY_NUM            = 0x00048013,
      GET_NUM_DISPLAYS           = 0x00040013,
      GET_DISPLAY_SETTINGS       = 0x00040014,

which are used for multiframebuffer support. But things *should* work without using any of those, but perhaps not. I guess will need to debug at the firmware level but not time to set everything up and understand how uboot works to look at that right now.

nullr0ute, 2019-07-02:
It seems to work, at least for me, when a display is attached to HDMI, it's when there's no display that things fall apart.

Vogtinator, 2019-07-02:
Let me go into more detail what the diff I posted works around.

u-boot first queries the "GET_PHYSICAL_W_H" property, but with the newer firmware that fails (response is not `0x80000000`).

u-boot prints an error message but ignores the error. This leads to u-boot using uninitialized values for the width and height, causing the reported cache warnings and aborts.

lategoodbye, 2019-07-02:
@Vogtinator Could you please clarify the influence of "HDMI attached or not"?

JamesH65, 2019-07-02:
Hmm, so maybe when no device is attached, there is no default values for width and height and somehow that affects the mbox call. I'll have a quick look.

satmandu, 2019-07-03:
@Vogtinator I can confirm that your patch allows current firmware to be used to boot a current arm64 build of ubuntu using u-boot on a 3b+.

GoofySpeed, 2019-07-03:
@Vogtinator Yes, the workaround worked my Arch Linux Arm aarch64.  Sorry, I don't have my serial cable right now so I can't post a new boot console.

ggardet, 2019-07-10:
I cannot reproduce on my Pi3B+ with https://github.com/raspberrypi/firmware/commit/025759b8634c2f8df35203be02c19a0633f1cec1

mbgg, 2019-07-11:
I treid

> I cannot reproduce on my Pi3B+ with [025759b](https://github.com/raspberrypi/firmware/commit/025759b8634c2f8df35203be02c19a0633f1cec1)
> 
> Same for you guys?

I tried with RPi3B and it worked. I suppose that got fixed in the new FW.
@JamesH65 can you confirm please.

JamesH65, 2019-07-11:
There was a change to ensure composite started up correctly if nothing else did which may have fixed this issue - it would mean if nothing was connected, a composite display would be assumed, which in turn would mean a default framebuffer would be created.

jonathangray, 2019-07-12:
The 1.20190709 tagged firmware works with U-Boot on rpi3b.

nullr0ute, 2019-08-04:
When testing RPi4 support with this patch set for U-Boot with the 2019.07 release https://lists.denx.de/pipermail/u-boot/2019-July/378340.html I'm seeing the same crash on a RPi4 4Gb Edition when the display isn't attached using firmware cba4be2 (Jul 15):

<pre>
MMC:   emmc2@7e340000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

mbox: Header response code invalid
bcm2835: Could not configure display
"Synchronous Abort" handler, esr 0x86000004
elr: 80000003c2116004 lr : 80000003c2116004 (reloc)
elr: 8000000400000004 lr : 8000000400000004
x0 : 000000003dc00000 x1 : 000000003dc00000
x2 : 0000000000000040 x3 : 000000000000003f
x4 : 0000000000000008 x5 : 000000003da00701
x6 : ffffffff80800000 x7 : 0000000000000002
x8 : 000000003da00000 x9 : 0000000000000008
x10: 0000000000000000 x11: 0000000000200000
x12: 0000000000000002 x13: 000000003da00000
x14: 0000000000000000 x15: 0000000000000020
x16: 000000003df8a8f0 x17: 4293900008300680
x18: 000000003db65de8 x19: 0000000000000000
x20: 000000003df8cad4 x21: 8000000000000020
x22: 0000000800028001 x23: 0000000080000008
x24: 0000000000000001 x25: 000000003db73b50
x26: 000000003db73d10 x27: 0000000000000002
x28: 000000003db73d18 x29: 0004000800000000

Resetting CPU ...

resetting ...
</pre>

mbgg, 2019-08-04:
Please try to cherry-pick:
970baf16d1 ("video: arm: rpi: Bail out early if querying video information fails")

From the master branch.

nullr0ute, 2019-08-04:
It had been fixed in firmware for the rpi3, will take a look

SamPovilus, 2019-12-01:
I'm trying this on a rpi4 2gbyte, I have head and 970baf16d1 has been merged, I still get this error. Anybody got any other thoughts?

ggardet, 2019-12-02:
It looks like a different issue, since this is after `Starting kernel...` message.

vivsundar, 2020-11-11:
@nullr0ute @SamPovilus I to observed this issue with the latest master branch of u-boot.  But with Tag v2020.07 this issue does not occur

rzr, 2021-06-02:
I am tracking related issues on latest release observed on RPi4

https://github.com/raspberrypi/firmware/search?q=%22Synchronous+Abort%22+handler%2C+esr&type=issues

https://github.com/raspberrypi/firmware/issues/1165

https://github.com/raspberrypi/firmware/issues/1157