raspberrypi/firmware #1158: Buster kernel built numbers screwed up

MrEngman, 2019-06-25:


pelwell, 2019-06-25:
Can you spare a few words to explain the title?

MrEngman, 2019-06-25:
What has happened to the kernel build numbers for the kernel updates since Buster appeared? "#896, #902 and #906".

Kernel build numbers for Buster are using values previously used for kernel 4.4.xx, Jessie.

Someone screwed the up the kernel compiles?

MrEngman

pelwell, 2019-06-25:
Has something led you to expect a continuity between build number for different releases?

MrEngman, 2019-06-25:
Well, it has been like this since kernel 3.2.27+ #6, Aug 18 2012, apart from an odd error, 3.18.6-v7+ #749 and 3.18.6+ #1. although a similar issue occured when moving from kernel 3.1.9 to 3.2.27, when the builds for 3.1.9 went from #21-#278 and then kernel 3.2.27 reverted to build #6.

MrEngman

Ferroin, 2019-06-25:
Kernel build numbers aren't guaranteed unique to begin with.  All they represent is that it's the nth kernel built since the last time someone completely cleaned out the build directory, so it takes effort to keep them increasing for a given project. In fact, most distros reset the build number every time the kernel version changes.

pelwell, 2019-06-25:
Thanks - that was my thinking, but a bit of validation is always welcome.

MrEngman, 2019-06-25:
Latest update uses build #1241 so looks like things are back to normal, AFAIK.

Thank you

closed 2019-06-25 by MrEngman