raspberrypi/firmware #1159: No hdmi signal on Pi4 using "hdmi2" after upgrading

richrboo, 2019-06-25:
hdmi2 is lost after upgrading Pi4. Switching to hdmi1 gets the Pi4 booting again, but hdmi2 no longer works. 

I can get hdmi2 working again, by reflashing SD card to 2019-06-20 release and avoid upgrading. So I'm assuming software issue, rather than a faulty Pi4.

pelwell, 2019-06-25:
The connectors are numbered 0 & 1, where 0 is nearest the USB C power socket.

richrboo, 2019-06-25:
Thanks. Sorry for the confusion, I saw that Raspbian uses hdmi1 and 2 in screen config. I have edited the report.

popcornmix, 2019-06-25:
I assume by upgrade you mean using rpi-update?

Can you explain the exact configuration you have when it fails?
e.g. one or two hdmi displays connected?
what resolution/refresh rate?
Do you have hdmi_enable_4k=1 in config.txt?

richrboo, 2019-06-26:
When you first boot 2019-06-20 release, it has a gui checking for an update and then upgrades. After upgrading I then reboot and lose hdmi1. I assume it is doing apt-get update and upgrade behind the gui.

I was only using one hdmi connection, hdmi1. This was plugged in to a TV.
Res refresh on the working hdmi0 is 3840x2160@30Hz. I assume the same for hdmi1, as I just swapped from hdmi1 to hdmi0, using the same TV.

hdmi_enable_4k=1 is not enabled.

richrboo, 2019-06-26:
A colleague tried his Pi4 and his hdmi sockets work. Using his working sdcard in my Pi, and hdmi1 doesn't work ruling out any sdcard issue. So possibly a hardware fault, but it did work with 2019-06-20 release, before the gui did the update.

popcornmix, 2019-06-26:
Does applying a little pressure (keep it light) to the hdmi plug when booting make it work?
Pushing from underside is more likely to help.

If so that would suggest a mechanical failure between PCB/hdmi socket/cable.

richrboo, 2019-06-26:
Nope. There was no play at all, applying pressure made no difference. I've been back to the Pi shop and exchanged it, Tim helped and is holding my Pi to show an engineer.

However now back in the office and I have the exact same issue with the exchanged one.

6by9, 2019-06-26:
It looks like there may be an issue with 4k30 on HDMI1 as I have replicated this. There are slight hardware differences between the two, and HDMI0 is the more capable port.
It looks like the HDMI state machine has stalled as I'm getting flip timeouts on HDMI-1.

Can you try adding 
```
hdmi_mode:1=16
hdmi_group:1=1
hdmi_mode:0=16
hdmi_group:0=1
```
to config.txt to force it to 1080p60 and reboot? At least that will confirm if you do have another dodgy board, or is it "only" software.

richrboo, 2019-06-26:
I went back to the shop. They are seeing issues with dual monitor and using hdmi1 too. I have yet to try the above, but have a 3rd Pi4. This is no better. The shop think it isn't hardware issue, but very strange my colleague Pi4 works as expected on either hdmi. I'll use a single monitor on hdmi0 for now. The Pi store have two boards for investigation at least. Hopefully a software fix can be found.

popcornmix, 2019-06-27:
No point returning more Pi's - this is a software issue that seems to affect some setups and not others.
Can you try adding to config.txt
``core_freq_min=500``
and report if this helps

popcornmix, 2019-06-28:
Did you add ``enable_uart=1`` to config.txt? Does the problem disappear without adding that?

richrboo, 2019-06-30:
I didn't have enable_uart=1 in my config.txt. My config.txt was as per defaults. Adding it prevents display on either hdmi0 or hdmi1.

Using core_freq_min=500 brings hdmi1 back to life, both hdmi0 and hdmi1 work with this setting.

popcornmix, 2019-07-02:
Can you test whether latest rpi-update firmware fixes your issue without adding ``core_freq_min=500``?

giddyhup, 2019-07-02:
I am not the original reporter but I have the same issue: single monitor connected on HDMI1 shows no picture. Result after rpi-update: short flash of boot rainbow and raspberries and then dark again.

giddyhup, 2019-07-02:
dmesg shows several entires related to DRM, if that's related.

popcornmix, 2019-07-02:
@giddyhup I assume monitor works when connected to HDMI0?
What does:
```
vcgencmd version
vcgencmd get_config int
vcgencmd measure_clock core
```
report?
Does ``core_freq_min=500`` in config.txt make HDMI1 work?

giddyhup, 2019-07-02:
I can't use HDMI0, my adapter is too wide. HDMI1 worked when I booted the original buster image (slimmed desktop version).

````
pi@raspberrypi:~ $ vcgencmd version
Jul  2 2019 15:12:05
Copyright (c) 2012 Broadcom
version 440db464adbebc026c8b3f2da2b49ad932e05f64 (clean) (release) (start)
pi@raspberrypi:~ $ vcgencmd get_config int
arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=250
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=500
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
pause_burst_frames=1
program_serial_random=1
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=250000496
```
The core minimum frequency workaround works.

P.S., probably unrelated: reboot got stuck, I had to cut the power; could be related to Pimoroni's fan which was running.

popcornmix, 2019-07-02:
What does ``tvservice -s`` show?

giddyhup, 2019-07-02:
...
(I removed the workaround. Now I have problems booting with a monitor attached.)

giddyhup, 2019-07-02:
Without monitor or when plugging in monitor when the Pi is already running
```
tvservice -s
state 0x120001 [TV is off]
```
With monitor after boot:
```
tvservice -s
state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
```
Remark: it looks like the monitor goes blank at the stage where the greater would appear. The rainbow pixel, the raspberries and the splash screen appear (albeit it is rather fast).

popcornmix, 2019-07-02:
I'm interested in results of booting with display already connected to HDMI1.
Are you saying you can't ssh in at this point?

giddyhup, 2019-07-02:
I can't reproduce the stuck-at-boot situation reliably but I am SSHing all the time, I currently have neither keyboard nor mouse attached.

giddyhup, 2019-07-02:
recent dmesg entries:
```
[   17.751802] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[   27.991793] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[   38.231799] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[   48.471794] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[   58.711796] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[   68.951798] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[   79.192843] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[   89.433388] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[   99.673667] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  109.913815] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  120.153882] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[  130.393919] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[  140.633391] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  150.872767] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  161.112277] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[  171.351902] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[  181.591594] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  191.831480] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  202.071181] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[  212.311063] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[  222.550900] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  234.070831] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  244.310647] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[  254.550515] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  265.430429] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  275.670316] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
[  285.910208] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
[  296.150086] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  297.257509] fuse init (API version 7.27)
[  308.310193] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
[  318.549905] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:28:plane-0] flip_done timed out
```

popcornmix, 2019-07-02:
So "With monitor after boot:" means the monitor was plugged in before boot?
And this is a 1080p display (no 4kp30 support)?

giddyhup, 2019-07-02:
Yes. Monitor was plugged it before boot. [Monitor does not appear to support 4k](https://www.philips.co.uk/c-p/243V7QJABF_00/full-hd-lcd-monitor).

giddyhup, 2019-07-02:
Latest syslog output:

```
Jul  2 17:26:20 raspberrypi kernel: [  908.008941] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out
Jul  2 17:26:20 raspberrypi kernel: [  908.009049] ------------[ cut here ]------------
Jul  2 17:26:20 raspberrypi kernel: [  908.009117] WARNING: CPU: 3 PID: 668 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:869 vc4_crtc_atomic_flush+0x98/0x9c [vc4]
Jul  2 17:26:20 raspberrypi kernel: [  908.009127] Modules linked in: fuse rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc brcmfmac brcmutil vc4 sha256_generic drm_kms_helper v3d cfg80211 gpu_sched rfkill snd_soc_core snd_bcm2835(C) snd_compress drm snd_pcm_dmaengine syscopyarea snd_pcm sysfillrect raspberrypi_hwmon sysimgblt hwmon drm_panel_orientation_quirks fb_sys_fops snd_timer snd bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev argon_mem media uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Jul  2 17:26:20 raspberrypi kernel: [  908.009634] CPU: 3 PID: 668 Comm: Xorg Tainted: G         C        4.19.56-v7l+ #1242
Jul  2 17:26:20 raspberrypi kernel: [  908.009642] Hardware name: BCM2835
Jul  2 17:26:20 raspberrypi kernel: [  908.009673] [<c0212c8c>] (unwind_backtrace) from [<c020d49c>] (show_stack+0x20/0x24)
Jul  2 17:26:20 raspberrypi kernel: [  908.009694] [<c020d49c>] (show_stack) from [<c0979880>] (dump_stack+0xd4/0x118)
Jul  2 17:26:20 raspberrypi kernel: [  908.009714] [<c0979880>] (dump_stack) from [<c0222270>] (__warn+0x104/0x11c)
Jul  2 17:26:20 raspberrypi kernel: [  908.009731] [<c0222270>] (__warn) from [<c02223c0>] (warn_slowpath_null+0x50/0x58)
Jul  2 17:26:20 raspberrypi kernel: [  908.009792] [<c02223c0>] (warn_slowpath_null) from [<bf7dba08>] (vc4_crtc_atomic_flush+0x98/0x9c [vc4])
Jul  2 17:26:20 raspberrypi kernel: [  908.009941] [<bf7dba08>] (vc4_crtc_atomic_flush [vc4]) from [<bf79081c>] (drm_atomic_helper_commit_planes+0x1a4/0x298 [drm_kms_helper])
Jul  2 17:26:20 raspberrypi kernel: [  908.010076] [<bf79081c>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf7dd0e0>] (vc4_atomic_complete_commit+0x60/0x33c [vc4])
Jul  2 17:26:20 raspberrypi kernel: [  908.010171] [<bf7dd0e0>] (vc4_atomic_complete_commit [vc4]) from [<bf7dd49c>] (vc4_atomic_commit+0xe0/0x194 [vc4])
Jul  2 17:26:20 raspberrypi kernel: [  908.010398] [<bf7dd49c>] (vc4_atomic_commit [vc4]) from [<bf375410>] (drm_atomic_commit+0x5c/0x60 [drm])
Jul  2 17:26:20 raspberrypi kernel: [  908.010748] [<bf375410>] (drm_atomic_commit [drm]) from [<bf377508>] (drm_mode_atomic_ioctl+0x98c/0xb78 [drm])
Jul  2 17:26:20 raspberrypi kernel: [  908.011086] [<bf377508>] (drm_mode_atomic_ioctl [drm]) from [<bf362bc4>] (drm_ioctl_kernel+0xb4/0xf0 [drm])
Jul  2 17:26:20 raspberrypi kernel: [  908.011418] [<bf362bc4>] (drm_ioctl_kernel [drm]) from [<bf362ff8>] (drm_ioctl+0x230/0x3cc [drm])
Jul  2 17:26:20 raspberrypi kernel: [  908.011594] [<bf362ff8>] (drm_ioctl [drm]) from [<c03c3edc>] (do_vfs_ioctl+0xbc/0x804)
Jul  2 17:26:20 raspberrypi kernel: [  908.011615] [<c03c3edc>] (do_vfs_ioctl) from [<c03c4668>] (ksys_ioctl+0x44/0x6c)
Jul  2 17:26:20 raspberrypi kernel: [  908.011634] [<c03c4668>] (ksys_ioctl) from [<c03c46a8>] (sys_ioctl+0x18/0x1c)
Jul  2 17:26:20 raspberrypi kernel: [  908.011651] [<c03c46a8>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
Jul  2 17:26:20 raspberrypi kernel: [  908.011660] Exception stack(0xda489fa8 to 0xda489ff0)
Jul  2 17:26:20 raspberrypi kernel: [  908.011672] 9fa0:                   02616e58 bee6ca78 0000000c c03864bc bee6ca78 00000000
Jul  2 17:26:20 raspberrypi kernel: [  908.011685] 9fc0: 02616e58 bee6ca78 c03864bc 00000036 02637bd8 026a4910 025fbc70 0000002d
Jul  2 17:26:20 raspberrypi kernel: [  908.011695] 9fe0: b6d0108c bee6ca44 b6ce8594 b697451c
Jul  2 17:26:20 raspberrypi kernel: [  908.011705] ---[ end trace b01d708e91ab7b28 ]---
Jul  2 17:26:20 raspberrypi systemd[1]: Starting Cleanup of Temporary Directories...
Jul  2 17:26:20 raspberrypi systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Jul  2 17:26:20 raspberrypi systemd[1]: Started Cleanup of Temporary Directories.
Jul  2 17:26:30 raspberrypi kernel: [  918.249088] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
Jul  2 17:26:40 raspberrypi kernel: [  928.489242] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:43:crtc-0] flip_done timed out
Jul  2 17:26:51 raspberrypi kernel: [  938.729374] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:45:HDMI-A-1] flip_done timed out

```

popcornmix, 2019-07-02:
The vc4_crtc_atomic_flush backtrace is known and harmless. See: 
https://github.com/raspberrypi/linux/issues/3020

popcornmix, 2019-07-02:
Okay your issue may be something new. OP was using 3840x2160@30 which needs more than 250MHz pixel clock (and also core_freq) and I was able to reproduce that.

1080p60 only requires 148.5MHz so I'd have thought 250MHz would be fine (and is in my setup).
Stick with core_freq_min=500 for now.
@timg236 (who did most of the clock/hdmi code) should be back in the office tomorrow - I'll see if he has any further ideas.

giddyhup, 2019-07-02:
OK, I will stick with core_freq_min=500.

6by9, 2019-07-02:
Can you report ```vcgencmd version``` just to confirm that the expected firmware actually is running?

giddyhup, 2019-07-02:
```
Jul  2 2019 15:12:05
Copyright (c) 2012 Broadcom
version 440db464adbebc026c8b3f2da2b49ad932e05f64 (clean) (release) (start)
```

popcornmix, 2019-07-04:
@richrboo @giddyhup can you try latest rpi-update firmware (without core_freq_min workaround).
There is another fix that could help your use case.

popcornmix, 2019-07-05:
@giddyhup @richrboo 
can you test latest firmware without the core_freq_min workaround?
Hopefully this will work better.

giddyhup, 2019-07-05:
```
Jul  5 2019 13:09:40
Copyright (c) 2012 Broadcom
version 8d1e59694b5cc50b7ecd7e3f3944db8090523006 (clean) (release) (start)
```

Won't even boot with the screen attached.

popcornmix, 2019-07-05:
Does it is core_freq_min=500?

giddyhup, 2019-07-05:
Yes.

popcornmix, 2019-07-05:
@giddyhup another firmware has been pushed. Is this okay for you without core_freq_min=500?

giddyhup, 2019-07-05:
It works for me! Thanks.

popcornmix, 2019-07-05:
@giddyhup thanks for testing!

@richrboo would be useful if you can test too.

richrboo, 2019-07-06:
I'm using the firmware release `Jul 5 2019 19:02:38` without `core_freq_min=500` and both hdmi0 and hdmi1 work.

Thank you.

Brian-Milnes, 2019-07-07:
I have similar issues on RPi4 with dual displays
Raspberry Pi 4 2Gb
Problem when connecting two HDMI monitors (using two official microHDMI leads)
RPi4 freezes during boot.
Both screens (1x1080p, 1x2160p) display rainbow splash screen, then HDMI 2 goes blank and boot process hangs

After normal boot, attaching second cable and second monitor has no effect
`tvservice -s` returns 
`state 0xa`

`# tvservice -l
1 attached device(s), display ID's are :
Display Number 2, type HDMI 0
`
`tvservice -d ~/edid.dat`
produces what looks like a binary file (unreadable in txt editor)

JamesH65, 2019-07-07:
@Brian-Milnes Just to confirm, you have done an rpi-update to get the latest firmware, and you have the 4k monitor in HDMI port 0?

Brian-Milnes, 2019-07-07:
Hi James
I have tried swapping monitors between HDMI0 and HDMI1 without a different result.
Slightly cautious about RPI-UPDATE with the Beta software, but may as well try it...
I'll report back
Brian

popcornmix, 2019-07-08:
@Brian-Milnes if you want to help test, download:
https://github.com/raspberrypi/firmware/blob/master/boot/start4.elf?raw=true
https://github.com/raspberrypi/firmware/blob/master/boot/fixup4.dat?raw=true
and copy to FAT partition of sdcard (alongside start.elf/fixup.dat).
(start4.elf/fixup4.dat will be used in preference to start.elf/fixup.dat when present)

knuckleheadsmiff, 2019-08-25:
It is August now. I'm just reporting the the same thing is happening to me when I applied the raspberries update right after the initial install on a new pi4 on 22nd. It is clear is an upgrade software/config issue. I am connected to an OLED LG 4K TV which supports 30hz as well as 60hz. With the initial noobs software for the pi4 it works fine with the defaults picked. Runs in 4K 30hz.

jmcerrejon, 2019-09-05:
Official power supply.
Rpi 4 4GB.
usb powered with keyb, mouse.

Same issues with an older Samsung TV when rebooting upgrading the OS. Using ``` hdmi_safe=1 ``` did the trick.

maybar, 2019-09-21:
This worked for mi config_hdmi_boost=4

htpbreewel, 2019-10-23:
Oct 23 2019. Same issue here. I tried LibreELEC-RPi4.arm-9.1.501.img (as OS-1) and 2019-09-26-raspbian-buster (as OS-2). One monitor via HDMI.
1. No internet and fresh install of OS-1 and OS-2 -> No audio on HDMI-1. Normal on HDMI-0.
2. Internet connected and fresh install of OS-1 and OS-2 with updates -> No audio on HDMI-1, normal on HDMI-0
I tried 1. and 2. with monitor connected during first boot to HDMI-0 and to HDMI-1 during first boot, makes no differance.

6by9, 2019-10-23:
@htpbreewel If your issue is solely with audio, then that is NOT this issue.

HDMI audio is routed independently of video.
There was an update to Raspbian recently to overcome the ambiguity that KMS represents a single display always as HDMI-1, so audio routing then needs to know which audio device that really relates to.
If you want independent simultaneous audio output from both HDMI ports, then see https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=249204#p1526690

lloyd61, 2019-11-12:
I'm using the latest version of Libreelec on a Raspberry Pi 4 and I am trying to use it with a 75" Samsung TV. My issue is that if I use all the same hardware and cables and boot it up with Raspbian installed, all works fine and it displays correctly on the TV, but if I swap out the SD card and boot up with Libreelec, I just get the no source error. I know the SD card is fine because if I plug it into a monitor Librelec is running. Libreelec doesn't seem to like the Samsung TV. Is it something to do with resolution and frequency?

mullcom, 2019-11-12:
You can try to change to hfmi-failsefe in config.txt

See if that works.

send from mobile

htpbreewel, 2019-11-12:
> Bug from
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url":

With LibreElec only the  HDMI port closest to the power jack works on 
the Pi 4 (for the time being?). Not the second one.

lloyd61, 2019-11-12:
Thank you for the feedback and suggestions. After manipulating the config.txt file, I was able to get an image from the HDMI socket closest to the board corner, nothing from the inner socket. But that still wasn't the end of it, no matter what I did I could not get any sound. In the end I went back to my faithful Pi 3B+, imaged the same SD card, plugged it in and all worked straight away.

ajgbarnes, 2020-01-01:
@popcornmix I have had the same issue with two different Raspberry Pi 4 boards and two separate televisions (a Panasonic and a LG OLED65C9PLA 4K QLED TV).  Nothing would work initially, I was getting No Signal on both TVs regardless of the hdmi_<whatever> configuration that I would put into the config.txt. I already have an RPi 3 B+ running but wanted to replace it with a 4K capable device.

Note that all tests here are conducted using LibreElec 9.2.0 (Leia) that runs Kodi 18.5 as my home theatre solution.

1. The core_freq_min=500 fixes it on my (relatively) old Panasonic TV HDMI connection - can use HDMI 0 or 1, both work
2. Removing (1) and using the linked start4.elf and fixup4.dat also fixes it on my Panasonic TV on both HDMI 0 and 1 - so no other solution required for that TV.
3. Neither solution (1) nor (2) works on the LG TV

I experimented some more and found that adding hdmi_enable_4kp60=1 to config.txt does allow it to be seen on the LG TV (still using start4.ef and fixup4.dat too). However there is speckling/interference on the screen. This only works on HDMI 1 though (not HDMI 0 surprisingly) - HDMI 0 only shows the Rainbow screen and LibreElec boot screen. HDMI 1 also shows the Kodi boot screen and then full interface, just with interference.

Other notes for the LG screen:
1. hdmi_safe=1 doesn't allow any signal to be detected on HDMI 0 or 1
2. hdmi_group=2 / hdmi_mode=4 doesn't allow any signal to be detected on HDMI 0 or 1
3. config_hdmi_boost=<value> or config_hdmi_boost:1=<value> seems to do nothing at all (where <value> I have tried includes 5, 7, 9 and 11)

The interference is too bad for an RPI 4 to be used with the LG (I have tried different HDMI cables too). Any other suggestions above what is above?  I can ssh to it and run any diagnostics.

@lloyd61 I feel your pain.... :)  Maybe time to hack a Fire TV stick to get 4k if this can't be fixed :(

ajgbarnes, 2020-01-01:
> no matter what I did I could not get any sound

Did you try setting hdmi_drive=2 in the config.txt

blacksocket, 2020-03-09:
Hi, I am having issues with displaying any HDMI output from my new RPI4 (4GB) on TV Panasonic Viera TX-l55DT50E. I am using os.: 2020-02-05-raspbian-buster-lite.img up-to-date including rpi-firmware.

I tried almost every possible important configurations located in /boot/config.txt without success.
After rpi reboot the Panasonic TV is showing nothing (no video available). Other devices works fine on the same TV HDMI input.
But when connect the same rpi via the same hdmi cable to my PC Monitor Philips 276E7QDSW/00 it works fine.

Currently I have this configuration in /boot/config.txt:

hdmi_force_mode=1
hdmi_group=1
hdmi_mode=31
hdmi_drive=2
config_hdmi_boost=9
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
pi@raspberrypi:~ $ /opt/vc/bin/tvservice -m CEA
Group CEA has 1 modes:
(prefer) mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive

System
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.106-v7l+ #1297 SMP Tue Feb 25 13:19:54 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-02-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e577677b623b577f2a0ec7cfaffc3c27da005da3, stage2
pi@raspberrypi:~ $ vcgencmd version
Feb 28 2020 18:44:10
Copyright (c) 2012 Broadcom
version 9520663ed4fa4ca8c2a6923a522bff85fcdedef5 (clean) (release) (start)

Can anyone please help with this issue ?

Misak, 2020-03-26:
@blacksocket Hi, I have the same problem, no output to Panasonic TV, were you able to find solution?

blacksocket, 2020-04-12:
Hi Misak, unfortunately I was not able to fix it yet.. ;-/

Misak, 2020-04-14:
Hi @blacksocket, it was a pain, but I find out, that it was caused by HDMI cable (the one that is working for other devices normally). I've switched to more quality one and now I have hdmi output from RPI 4 to Panasonic TV.

blacksocket, 2020-04-17:
Hi Misak, great! Whet exactly cable are you using now ? Can you share the parameters you used in your configuration in /boot/config.txt ?

[11 more replies]