raspberrypi/firmware #1161: PI4: Peripheral Addresses seems not implemented ?

F5OEO, 2019-06-27:
According to https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md

This should be the prefered method to get peripherals address for compatibilty.

On Pi4 :
bcm_host_get_peripheral_address() return **0** : IS IT THE REAL BEHAVIOR ? 
bcm_host_get_sdram_address() return c0000000 (seems good)

JamesH65, 2019-06-27:
This is currently broken, we are looking in to the fix.

lurch, 2019-07-08:
@JamesH65 Presumably this is a duplicate of https://github.com/raspberrypi/documentation/issues/1185 ?

@popcornmix The commit message for https://github.com/raspberrypi/firmware/commit/9c07a7b29cfd8b79f2bb74d6cf382511f9c4f3d4 includes "firmware: bcm_host: Adjust for 64-bit physical addresses
See: #1161"
Isn't that a userland-level fix though, rather than a firmware-level fix? Should there have been a matching change to https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c ?

tangentsoft, 2019-08-14:
Yes, the two issues are the same. The fix should be here in libraspberypi, not in the documentation, but there are useful ideas over in that other thread.

Basically, `bcm_host_get_peripheral_address()` needs to be updated to have the expanded 12-byte `/proc/device-tree/soc/ranges` twiddling as in RPi-distro/raspi-gpio@80fa7d04e. In addition, the library should probably provide something like that check-in's `is_2711` flag and its `get_gpio_pull()` function. As for all of the stuff about pin naming in that check-in, I don't need that for my own purposes, but maybe that would be useful to someone else.

We'd prefer to continue using generic Raspberry Pi Foundation provided interfaces rather than do things the way the raspi-gpio project does it in [our own code](https://github.com/tangentsoft/pidp8i/blob/91a04d8c60895dca2dc34199bd7e14a38f3d0e2c/src/pidp8i/gpio-common.c.in#L704).

Thanks!

popcornmix, 2019-08-15:
@lurch libbcm_host.so is included in /opt/vc/lib as part of firmware update, so the commit message makes sense in the firmware tree.

Userland repo was updated on July 24, so I think this issue has been resolved. @F5OEO ?

lurch, 2019-08-15:
@popcornmix Ahhh, I was simply confused by the firmware repo getting the fix before the userland repo got the fix.

popcornmix, 2019-08-15:
It might seem like firmware libs are build from userland, but it's actually a longer route.

Internal vc4 tree is used for both firmware (start.elf) userland libs (/opt/vc/lib) and userland source.
The userland source repo needs some manual updating (as vc4 commits often span both the internal firmware tree and the public arm tree) so it sometimes lags behind. If pinged I can check the tree is up to date.

F5OEO, 2019-08-17:
Confirm it is fixed, thanks.

closed 2019-08-17 by F5OEO