raspberrypi/firmware #1162: PI3  cma_alloc: alloc failed, req-size: 2025 pages, ret: -12

ivangotoy, 2019-06-28:
0.125464] bcm2708_fb soc:fb: FB found 1 display(s)
[    0.125618] cma: cma_alloc: alloc failed, req-size: 2025 pages, ret: -12
[    0.157050] Console: switching to colour frame buffer device 240x67

 0.614725] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
[    0.615637] [vc_sm_connected_init]: start
[    0.618324] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.618328] [vc_sm_connected_init]: failed to initialize shared memory service
[    0.618332] [vc_sm_connected_init]: end - returning -1
[    0.618807] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    0.618812] [vc_sm_connected_init]: start
[    0.618870] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    0.618874] [vc_sm_connected_init]: failed to initialize shared memory service

gpu_mem=320

vcgencmd version
Jun 26 2019 17:47:16 
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start)

uname -a
Linux desktop 4.19.56-v7+ #1242 SMP Wed Jun 26 17:31:47 BST 2019 armv7l GNU/Linux

With firmware updates after 4.19.50 this was introduced to a smoothly working pi3 systems - similar behavior is observed on both pi3 and pi3 b+

6by9, 2019-06-28:
```
cma: cma_alloc: alloc failed, req-size: 2025 pages, ret: -12
```
is harmless.
The frame buffer driver tries to allocate the FB from CMA, and falls back to the older use of gpu_mem if that fails. There's no way to tell dma_alloc_coherent that you can cope with failure quite happily, and it spams the log over the failure.
Should be dealt with by https://github.com/raspberrypi/linux/pull/3029

There should be no change with regard vcsm in recent releases, so it's not clear what's going on there.

NB Both of these things are Linux kernel issues, therefore should really be reported at https://github.com/raspberrypi/linux/issues

6by9, 2019-06-28:
Tested with the same kernel and firmware as you on a 3B+. No issue observed over ```vc_sm_connected_init```

ivangotoy, 2019-06-28:
Sorry for posting to the wrong place, great to hear it's been resolved. Is gpu video decoding working smoothly on your end?

tmm1, 2019-08-22:
Can be closed?

ivangotoy, 2019-08-23:
sure, the new kernel interface in /dev is working fine with udev proper
rules

closed 2019-08-23 by popcornmix