raspberrypi/firmware #1164: PI2: Latest start.elf and bootcode.bin breaks u-boot

kallisti5, 2019-06-29:
Latest u-boot with latest start.elf and bootcode.bin from this repo:
```
U-Boot 2019.07-rc4-00179-g77f6e2dd05 (Jun 24 2019 - 14:25:14 -0500)

DRAM: 128 MiB
RPI 2 Model B (0xa21041)
MMC: sdhci@7e300000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

mbox: Header response code invalid
bcm2835: Could not configure display
In: serial
Out: vidconsole
Err: vidconsole
Net: CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
CACHE: Misaligned operation at range [07fdb6e8, 07fdb700]
.
.
.
```

Old (known working) u-boot and latest start.elf and bootcode.bin from this repo:
```
U-Boot 2017.11-00023-gebca2083d3 (Nov 17 2017 - 09:00:41 -0600)

DRAM:  128 MiB
RPI 2 Model B (0xa21041)
MMC:   sdhci@7e300000: 0
*** Warning - bad CRC, using default environment

mbox: Header response code invalid
bcm2835: Could not configure display
In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
CACHE: Misaligned operation at range [07fb3f64, 07fb3f80]
```

Old start.elf and bootcode.bin with latest u-boot:

* https://github.com/raspberrypi/firmware/blob/dc93c7922bb31f20759a94a6e000299da8e331e3/boot/start.elf
* https://github.com/raspberrypi/firmware/blob/dc93c7922bb31f20759a94a6e000299da8e331e3/boot/bootcode.bin

(Working as expected..)
```
U-Boot 2019.07-rc4-00179-g77f6e2dd05 (Jun 24 2019 - 14:25:14 -0500)

DRAM:  128 MiB
RPI 2 Model B (0xa21041)
MMC:   sdhci@7e300000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@7e980000: scanning bus usb@7e980000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
396 bytes read in 12 ms (32.2 KiB/s)
## Executing script at 02400000
17 bytes read in 11 ms (1000 Bytes/s)
699075 bytes read in 104 ms (6.4 MiB/s)
1574709 bytes read in 206 ms (7.3 MiB/s)
8327 bytes read in 25 ms (325.2 KiB/s)
## Booting kernel from Legacy Image at 00080000 ...
   Image Name:   Haiku arm loader
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    699011 Bytes = 682.6 KiB
   Load Address: 01000000
   Entry Point:  01000010
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 02700000 ...
   Image Name:   Haiku arm floppyboot
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    1574645 Bytes = 1.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02600000
   Booting using the fdt blob at 0x2600000
   Loading Kernel Image ... OK
   Using Device Tree in place at 02600000, end 02605086

Starting kernel ...
```

You can find the config.txt and the u-boot.bin we're using here:
https://github.com/haiku/firmware/tree/master/arm/rpi2

This is being discussed via the u-boot mailing list as well:
https://lists.denx.de/pipermail/u-boot/2019-June/374083.html

6by9, 2019-06-29:
Duplicate of #1157 ?

nullr0ute, 2019-06-29:
Looks like it