raspberrypi/firmware #1165: Loading U-boot on Raspberry Pi 3B failed

ilyxa, 2019-06-29:
Updating to new firmware cause boot stuck on RPi3b

Messages:
`U-Boot 2018.07-1 (Jul 18 2018 - 23:27:57 +0000) Arch Linux ARM

DRAM:  948 MiB
RPI 3 Model B (0xa02082)
MMC:   mmc@7e202000: 0, sdhci@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

Failed (-5)
mbox: Header response code invalid
bcm2835: Could not configure display
"Synchronous Abort" handler, esr 0x86000004
elr: 80000003c4d3b004 lr : 80000003c4d3b004 (reloc)
elr: 8000000400000004 lr : 8000000400000004
x0 : 000000003b000000 x1 : 000000003b000000
x2 : 0000000000000040 x3 : 000000000000003f
x4 : 0000000000000008 x5 : 0000000000000000
x6 : ffffffff80800000 x7 : 0000000000000002
x8 : 000000003ae00000 x9 : 0000000000000008
x10: 0000000000200000 x11: 0000000000000000
x12: 0000000000000002 x13: 000000003ae00000
x14: 0000000000000000 x15: 00000000ffffffff
x16: 0000000000004110 x17: 0000020000000000
x18: 000000003af40df0 x19: 0000000000000000
x20: 000000003b36afbc x21: 000000003af44ad8
x22: 0000000000000002 x23: 0004000800000000
x24: 8000000400000004 x25: 000000003af44ad0
x26: 000000003af44890 x27: 000000003af44ad8
x28: 0000000000000002 x29: 0004000800000000

Resetting CPU ...

resetting ...
`

ilyxa, 2019-06-29:
https://github.com/raspberrypi/firmware/issues/1157 same here

kallisti5, 2019-06-30:
#1164 same here :-)

satmandu, 2019-07-06:
This u-boot [patch](https://github.com/satmandu/u-boot/commit/b514f892bc3d6ecbc75f80d0096055a6a8afbf75) fixes u-boot as per https://github.com/raspberrypi/firmware/issues/1157#issuecomment-506706177