raspberrypi/firmware #1166: No composite output on Pi0W after update

mati75, 2019-06-30:
After update to firmware from stable branch composite output on Pi0W isn't work.

My settings:
```
# uncomment for composite PAL
sdtv_mode=2
#sdtv_aspect=3
#sdtv_progressive_scan=1
#sdtv_aspect=4:3
#framebuffer_width=960
#framebuffer_height=468

framebuffer_width=480
framebuffer_height=272

#sdtv_mode=1
sdtv_aspect=3
```

Return to the latest version from May fixes the problem. Kernel 4.19.50 with May firmware working fine.

6by9, 2019-06-30:
Could you check the output of ```vcgencmd get_config enable_tvout```?
If it returns 0 (it shouldn't) then you should be able to enable it by adding ```enable_tvout=1``` to /boot/config.txt. Please then confirm the revision code for you Pi0W from /proc/cpuinfo?

Composite out is disabled by default on the Pi4, but it's just possible that that disable has managed to apply to other models.

klricks, 2019-07-01:
On a RPi3B+
$ vcgencmd get_config enable_tvout
enable_tvout is unknown
pi@raspberrypi:~ $ 

enable_tvout=1 to /boot/config.txt. did not help.

6by9, 2019-07-01:
Apologies - I was forgetting that we currently have two different firmware trees. Pi 0-3 won't have that option.
There is only one commit that touches really display, so I'll have a closer look.

JamesH65, 2019-07-02:
Found the problem, fix has been pushed. WIll be in rpi-update in the near future.

joolswills, 2019-07-04:
We just bumped into this with RetroPie 4.5 release. We had manually updated to buster firmware/kernel packages (since the last Stretch kernel had another issue that broke sony bluetooth devices).

We would be hugely appreciative of a kernel/firmware package update for Stretch that includes this. Happy to send bribes/cake/beer etc. Thanks as always for all your efforts.

ghollingworth, 2019-07-04:
Looks like it's already there...

https://github.com/raspberrypi/firmware/commit/383fcf5be7f1bcf8d63ca8b471ecd9b8da3cef0b

Have you updated your firmware? (rpi-update)

Gordon

joolswills, 2019-07-04:
The fix is there yeah. Was referring to it being packaged. Thanks.

mati75, 2019-07-06:
If no errors appeared, your firmware was successfully updated to 7918aa035dd6529f32d242775dd0e3ca6c5bf3d3

Fixed for me this issue.

closed 2019-07-06 by mati75