raspberrypi/firmware #1167: After upgrade to the latest firmware, images from Pi Noir V2 now purple

anfractuosity, 2019-06-30:
With the latest firmware in buster, the results under similar lighting conditions:

[image]

With the firmware in stretch, the results under similar lighting conditions:

[image]

I've tried using the same version of raspivid as in stretch, on buster, but still got purple results, which leads me to think this is a firmware issue.

Cheers

6by9, 2019-06-30:
As per https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=241633, the AWB algorithm was updated a month ago.
Can you check whether this is an issue with the new algorithm by running ```sudo vcdbg set awb_mode 0``` before running raspivid. AWB is never really expecting the IR component, therefore it could be doing odd things.

anfractuosity, 2019-06-30:
Thanks a lot! So I just ran it now (night time here) with my IR light, the results are now B&W, rather than purple :)

I'll run it again tomorrow daytime, but it looks like that fixed it! Thanks a lot!

6by9, 2019-07-01:
That's a workaround rather than the fix. I've reported it to the guys who wrote the new algorithm.

It's a curious one, and I suspect that even the old algorithm is giving up on it, but it then drops back to the simpler greyworld scheme which is likely to give you more sane mono pictures.

anfractuosity, 2019-07-01:
Gotcha thanks a lot.  During daytime, the images now look like the non-purple version again with the workaround.

Pi-DD, 2019-07-03:
Had the same problem on Pi Zero + Buster + MotionEye.

Running the suggested workaround fixes the problem, but seems to need running after every reboot.

anfractuosity, 2019-07-03:
@Pi-DD I created a systemd service that runs it for me:

```
[Unit]
Description=AWB
After=network.target

[Service]
Type=simple
User=root
WorkingDirectory=/usr/bin
ExecStart=vcdbg set awb_mode 0 
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

popcornmix, 2019-07-08:
Latest rpi-update firmware has a potential fix to the new AWB scheme. Can you test it?

anfractuosity, 2019-07-09:
I did 'sudo raspi-update' and disabled the systemd service I created, but the images still look purple unfortunately:

[image]

popcornmix, 2019-07-09:
@anfractuosity can you report output of ``vcgencmd version``?

anfractuosity, 2019-07-09:
Sure:
```
Jul  8 2019 21:30:17 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start_x)
```

naushir, 2019-07-09:
@anfractuosity, could you provide some more details please?

- Is the scene illuminated with IR light?
- Would you happen to know the lux level of the scene?  Failing that, is it bright or dark?
- Any chance you can attach a capture with the --raw option?

anfractuosity, 2019-07-09:
Sure, github didn't like me attaching the files (15MB each), but I've added a link below instead:

https://www.anfractuosity.com/files/hack.jpg - taken after the 'sudo vcdbg set awb_mode 0' trick
https://www.anfractuosity.com/files/nohack.jpg - taken using the latest firmware without vcdbg hack.

Both images were taken with --raw.  Hope that helps.

Both images where taken minutes after each other.  It's from daytime, under typical British weather ;) (rain and cloud).  It's not IR illuminated at the moment.

naushir, 2019-07-09:
Thank you for the images.

We think we have identified why this may be happening - summary is that the algorithm searches along the colour temperature curve measured for the standard camera and is (correctly) restricted from moving too far off from there.

We will need to discuss how to fix this correctly for the NoIR case.  Until then, could you please run with the `vcdbg set awb_mode 0` workaround.

microadam, 2019-07-10:
I am having the same issue. I assume doing `vcdbg set awb_mode 0` completely disables auto white balance? As the white balance on the camera is all over the place (mostly either very yellow or very red)

microadam, 2019-07-10:
Have realised that commands just changes the way white balance is calculated. Must be something else going wrong with my setup! (running that command does remove the pink glow though). Only just got the PI camera, so had nothing to compare it against

naushir, 2019-07-10:
`vcdbg set awb_mode 0` reverts to the old AWB algorithm.  If that shows any issues, we are unlikely to work on fixing them now.  Instead, our efforts would be better spent getting the new algorithm working correctly for the Noir cases.

2sec, 2019-07-14:
just adding my 2 cents..

- running vcdbg set awb_mode 0 while the camera is capturing can freeze the system, in that case it needs to be physically unplugged.

- I'm using a bunch of NoIR cameras as a surveillance system (using only the Y luminance data in YUV), the new algorithm makes the cameras unusable, when the sun shines through the windows, almost nothing can be seen.

Pi-DD, 2019-07-14:
@2sec - thanks for that - I wondered what had happened when, following kernel updates yesterday I got the “purple image problem” on Stretch remote device which then froze when I tried to fix it. Fortunately I can get to it and restart it and then add the system service suggested above to force the old AWB algorithm.

Pi-DD, 2019-07-15:
@anfractuosity - I would appreciate some guidance on implementing your systemd service. 

Not having done anything like that before I have tried to read up on this and make it work, but am obviously missing something!

I have placed the file in /etc/systemd/system  as other material seemed to guide me but this is what I get...

`pi@DDhomepi5:~ $ sudo systemctl start AWB.service
Failed to start AWB.service: Unit AWB.service is not loaded properly: Invalid argument.
See system logs and 'systemctl status AWB.service' for details.
pi@DDhomepi5:~ $ systemctl status AWB.servicve
Unit AWB.servicve.service could not be found.
pi@DDhomepi5:~ $ systemctl status AWB.service
● AWB.service - AWB
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)
`

Pi-DD, 2019-07-15:
@popcornmix - thank you, I can see you are trying to help, but with my (rather limited!) level of skill and knowledge I have no idea what to do about what you are pointing me to!

anfractuosity, 2019-07-15:
@popcornmix I just tried the latest firmware 

Jul 15 2019 17:36:28 
Copyright (c) 2012 Broadcom
version ccada45e1764783c9876237218ef886d96b24fc7 (clean) (release) (start_x)

And attached two pictures taken just now under IR light:

[image]

[image]

@Pi-DD Try this instead maybe, although the version I listed did work for me:

```
[Unit]
Description=AWB
After=network.target

[Service]
Type=simple
User=root
ExecStart=/usr/bin/vcdbg set awb_mode 0 
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

sudo systemctl enable AWB
sudo systemctl start AWB

Pi-DD, 2019-07-16:
@anfractuosity - many thanks for that, perfect! Worked first time. No idea why the original did not, but now working perfectly from boot to use the old algorithm.

naushir, 2019-07-16:
A new firmware has been released (use rpi-update to get) that provides some updates to the AWB algorithm.  In particular, there is a new "Grey World" mode of operation that should hopefully fix the issues with the NoIR cameras.  This is sort of what the old algorithm was doing to get the "correct" results, although mostly by accident.

To use this new awb mode, you will need to update the raspicam binaries with a patch - see [here](https://github.com/raspberrypi/userland/pull/560.)  Until this is fully merged you will have to build the binaries manually.  Once you have done that, run `raspivid -awb greyworld ...` command line option to get it working.

anfractuosity, 2019-07-16:
Thanks, I didn't know about the greyworld parameter, just compiling to try that now, cheers!

naushir, 2019-07-16:
@anfractuosity , thanks for trying it out and let us know how you get on.   The greyworld parameter is a new addition with the linked PR.

anfractuosity, 2019-07-16:
Woo, looks good! :) 

[image]

For reference I did:

```
git clone https://github.com/raspberrypi/userland.git
cd userland
git fetch origin pull/560/head
git checkout -b pullrequest FETCH_HEAD
./buildme
cd build/bin/
./raspivid -o tcp://0.0.0.0:10000 -l -t 0 -fps 30 -b 8000000 -awb greyworld
```
```
mpv tcp://pi_ip:10000
```

Good work!

naushir, 2019-07-16:
Great, thanks for confirming.

2sec, 2019-07-16:
Is there a way to use this "grey world" mode directly (I'm using picamera in Python) using a system setting such as the previous one?

naushir, 2019-07-16:
> Is there a way to use this "grey world" mode directly (I'm using picamera in Python) using a system setting such as the previous one?

I think this should work the same if you set the [awb mode](https://picamera.readthedocs.io/en/release-1.13/api_camera.html#picamera.PiCamera.awb_mode) to 'greyworld' in PiCamera.

anfractuosity, 2019-07-16:
I just used it in PiCamera

```
import picamera
from picamera import mmal 
import ctypes as ct

class PiCamera2(picamera.PiCamera):
    AWB_MODES = {
        'off':           mmal.MMAL_PARAM_AWBMODE_OFF,
        'auto':          mmal.MMAL_PARAM_AWBMODE_AUTO,
        'sunlight':      mmal.MMAL_PARAM_AWBMODE_SUNLIGHT,
        'cloudy':        mmal.MMAL_PARAM_AWBMODE_CLOUDY,
        'shade':         mmal.MMAL_PARAM_AWBMODE_SHADE,
        'tungsten':      mmal.MMAL_PARAM_AWBMODE_TUNGSTEN,
        'fluorescent':   mmal.MMAL_PARAM_AWBMODE_FLUORESCENT,
        'incandescent':  mmal.MMAL_PARAM_AWBMODE_INCANDESCENT,
        'flash':         mmal.MMAL_PARAM_AWBMODE_FLASH,
        'horizon':       mmal.MMAL_PARAM_AWBMODE_HORIZON,
        'greyworld':     ct.c_uint32(10)
        }

with PiCamera2() as camera:
    camera.awb_mode = 'greyworld'
    camera.resolution = (640, 480)
    camera.start_recording('my_video.h264')
    camera.wait_recording(5)
    camera.stop_recording()
```

naushir, 2019-07-16:
Sorry, I forgot to mention, there will be a patch needed for PiCamera to get this plumbed in :)

2sec, 2019-07-16:
Great, thanks, I'll wait and use in the vcdbg trick in the meantime

Pi-DD, 2019-07-16:
A question - how will the new “greyworld” setting work for IR cut cameras that use a physical filter to swap between daylight/artificial lighting and IR illumination? Will you need to find some way of detecting that event and change the WB setting? 

I’ve not used one, but assume that the previous AWB algorithm accidentally handled this OK because with the filter in place it worked as intended for daylight and with the filter removed it “accidentally” worked  OK as it did for NOIR cameras.

naushir, 2019-07-17:
> A question - how will the new “greyworld” setting work for IR cut cameras that use a physical filter to swap between daylight/artificial lighting and IR illumination? Will you need to find some way of detecting that event and change the WB setting?
> 

In these cases, the block of code that moves the filter in/out could also switch the AWB mode appropriately.  If there is no programmatic way to know if the filter is in or out of position (perhaps the sensor module does this automatically?) then I would suggest that you leave the AWB in greyworld mode.  The AWB results with the IR cut filter + greyworld should be reasonable for most cases.

Pi-DD, 2019-07-17:
Thanks, I'm just considering implementing such a setup. Since there is a sensor to trigger the filter I'm guessing there must be a way of accessing this, but it might well be beyond my limited skills (or a steep learning experience!) 
There are a lot of IR cut cameras out there and I suspect that once the new firmware, even with "greyworld", rolls out there will be some very puzzled people who have been relying on AWB until knowledge of the new setting becomes widely known.

JamesH65, 2019-07-24:
This is fixed, so closing.

SdeGeata, 2019-10-11:
I messed with trying to set up that systemd service to run the `vcdbg set awb_mode 0 `, couldn't for the life of me get it to work. Every reboot made the image purple again. Finally it occurred to me that I am already running a pm2 service to auto-start my magic mirror, so I added `sudo vcdbg set awb_mode 0 ` as my first line of my mm.sh.

That whole script looks like this:

`sudo vcdbg set awb_mode 0
cd ~/MagicMirror
DISPLAY=:0 npm start`

Works like a charm!

JessicaWoods03, 2019-11-24:
wtf- why upgrade anything if its just going to fuck up everything- who makes this crap? my shit was perfect till i hit that damn upgrade button- I have spend more time than its honestly worth to fix a super fuck up by a bunch of fuck ups-

pelwell, 2019-11-24:
@JessicaWoods03 Abuse will not be tolerated. Your message has deleted. Further offences will result in blocking.

closed 2019-11-24 by pelwell

JessicaWoods03, 2019-11-25:
no problem, my tower melted down from the upgrade, so did I, I found a way to fix it, my apologies-

JessicaWoods03, 2020-01-07:
So I did the fix- and it fixed- but somehow all of these are temp fixes- I did the awb_mode- IDK if its another upgrade messup- for 2020? Anyone else's camera took a crap this year? Need some help on this-

JessicaWoods03, 2020-01-07:
I have a Pi 3B+ I had a hard time finding software support for the pi4- so I went with my old pi-

naushir, 2020-01-08:
@JessicaWoods03, could you post a detailed explanation of your problem to the Raspberry Pi camera forums please?  You can find it at https://www.raspberrypi.org/forums/viewforum.php?f=43.

As far as we know, there are no known issues with awb_mode=greyworld on Pi4 and this githhub issues will remain closed.

Pi-DD, 2020-03-08:
Has any recent kernel update removed the ability to run the vcdbg “fix” to go back to the old AWB version? 

I ask because everything indicates that by AWB.service runs successfully, but I am now getting the purple effect again.

JessicaWoods03, 2020-03-09:
Yeah mine too- I am rerunning the vcdbg "fix" once before last year and it fixed it- and it went back to purple- Just this year recently with the latest updates-

JamesH65, 2020-03-09:
You should not be using the vcdbg command to fix this, we introduced a new AWB mode parameter, greyworld, that should fix it correctly. 

For example, 

raspistill --awb greyworld -o fred.fpg

The MMAL parameter is MMAL_PARAM_AWBMODE_GREYWORLD.

JamesH65, 2020-03-09:
Note, this should only be necessary for the NoIR camera.

Pi-DD, 2020-03-09:
James - in my case the problem is that I don’t feel competent to install Motion from a tar file and the setup I am using needs the latest level of Motion (4.3.0) that supports “greyworld” to pass the the parameter to the camera. It’s not yet available as a deb file. 

But just to be entirely clear - am I correct that the “fix” no longer works since grey world was introduced?

JamesH65, 2020-03-09:
Quick look at the code and I cannot see any immediate reason why it would not work, would need to spend longer trying to figure it out and don't have time right now, especially since the vcdbg was very much a nasty hack for testing purposes whilst we implemented the correct fix, so should not be relied upon.

JessicaWoods03, 2020-03-09:
James your awesome!! Thanks for helping us out :)

naushir, 2020-03-09:
@JessicaWoods03, could you please post a log captured with:

`sudo vcdb log msg`

After running and stopping your camera application please?  As @JamesH65 mentioned, this bit of code has not been touched in a while, so I am also a bit surprised that it has now broken.

JessicaWoods03, 2020-03-09:
Yeah I can...give me a second...its a bit of a mess right now- I unplugged it...just called it a loss. This was a new issue I am sure no one expected to deal with-

saumyakswain, 2020-03-23:
> You should not be using the vcdbg command to fix this, we introduced a new AWB mode parameter, greyworld, that should fix it correctly.
> 
> For example,
> 
> raspistill --awb greyworld -o fred.fpg
> 
> The MMAL parameter is MMAL_PARAM_AWBMODE_GREYWORLD.

Thanks a lot. I used my Pi after 1year and found everything was pink. At first, Thought my cam had got some hardware issues. This answer solved my issue.

6by9, 2020-03-25:
FYI I've merged a further firmware change for those who can't update their userspace apps. Once it is released, if you add ```awb_auto_is_greyworld=1``` to config.txt then it'll do as it says and selecting auto will choose greyworld.
The preferred solution is to fix the app, but this is the easier fallback than the vcdbg line.

johnb8, 2020-03-29:
Is the auto white balance greyworld preset supposed to work with the v4l2 module as well? I've tried setting it in qv4l2 but I still get purple images.

6by9, 2020-03-30:
> Is the auto white balance greyworld preset supposed to work with the v4l2 module as well? I've tried setting it in qv4l2 but I still get purple images.

Yes it should be available through V4L2 as long as you have updated the firmware appropriately.

johnb8, 2020-03-31:
I've flashed a new SD card with the latest version of raspian but I still get purple images from my noir camera when using V4L2, even though I've set the white balance to greyworld. Images from raspistill look correct with the greyworld awb though. I'm using qv4l2 as my capture application. qv4l2 and `v4l2-ctl --all` both show the white balance preset as greyworld.

dougputt, 2020-06-18:
Is there a logical fix in chronological order on how to fix the purple camera issue on Noir?  There is rag tag information everywhere but ...
running PI4, noir v2, motioneye

Thanks

6by9, 2020-06-22:
If you're really stuck, then add ```awb_auto_is_greyworld=1``` to /boot/config.txt. On firmware since the end of March 2020 it means "auto" setting for awb  mode uses the simple greyworld algorithm.

Flo0, 2020-06-22:
My pics turned pink just yesterday. Im currently iterating over awb_gains(red, blue) in steps of range 0.05 in python to find out what works best. Every awb_mode setting is horribly pink. Is the greyworld setting accessible via the python api?

Also running Pi4 noir v2

[3 more replies]