raspberrypi/firmware #1169: [Question] What are the RPi4 OTP Values?

ptesarik, 2019-07-01:
This is somewhat related to issue #974, but not quite the same. Apparently, the Raspberry Pi 4 has changed the use of OTP. For one thing, I have noticed that the MAC address is now pre-programmed in registers 64 and 65.

Registers 17 & 18 have probably changed layout. RPi3 definitions for register 66 (advanced boot) are no longer meaningful, so this one is either gone, or has a different meaning.

At a minimum, can you clarify at https://www.raspberrypi.org/documentation/hardware/raspberrypi/otpbits.md which definitions are not valid for RPi4, please?

JamesH65, 2019-07-01:
Not at this stage.

ptesarik, 2019-07-02:
That's a pity.

Would you mind warning users that the various `program_*` options in `config.txt` may not work as intended on a RPi4? Please, consider also adding a disclaimer to the [official documentation](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/host.md) (and subpages).

maxnet, 2019-07-06:
>I have noticed that the MAC address is now pre-programmed in registers 64 and 65.

Out of curiosity, why is OTP still used at all, now you have an EEPROM that could theoretically also have settings like that?

ghollingworth, 2019-07-07:
Some of the OTP numbers are there because they're needed before the SPI EEPROM is read (Boot selection modes, some reset selection)
Some are needed there because we want to make sure they don't ever get changed (serial number, mac address, revision number).

timg236, 2019-07-07:
Once other bootmodes are available the configuration will be via the EEPROM in a minimal config.txt file, probably with some Python tool to make this easy.
It may also be possible to create an alternative to the 'customer OTP' rows as EEPROM settings which can be read via a vcgencmd.
So, user-modifiable stuff can all be in the EEPROM.

ptesarik, 2019-07-09:
> Not at this stage.

So, I had to do my own research… Most OTP bits related to early boot have indeed changed. No guarantees for the information below.

OTP register 18 (bootmode) is now:
* bit 1: oscillator frequency (same as RPi3: 0 = 54.0 MHz, 1 = 19.2 MHz)
  **Do not touch.**
* bits 4-6: pins used to access the SPI boot EEPROM:
  * 0: none
  * 1: GPIO 7-11 Alt Fn 0
  * 2: GPIO 35-39 Alt Fn 0
  * 3: GPIO 40-43 Alt Fn 4 (default)
  * 4: GPIO 46-49 Alt Fn 2
  Pre-programmed and **cannot be changed**.
* bit 7: enable USB device mode boot (over the Type-C connector)
  This will be tried only if the boot EEPROM fails (e.g. bad checksum).
  Pre-programmed and **cannot be changed**.
* bits 11-13: eMMC boot pins (SDIO1):
  * 0: none
  * 1: ???
  * 2: GPIO 22-27 Alt Fn 3
  * 3: GPIO 34-39 Alt Fn 3
  * 4: GPIO 48-53 Alt Fn 3 (wired to the SD card slot)

In OTP registers 66 and 67, eight-bit fields are used to define each GPIO pin:
* bits 0-5: GPIO number (Broadcom numbering)
* bit 6: polarity (0 = active HIGH, 1 = active LOW)
* bit 7: enable

OTP register 66 is used to control the boot process:
* bits 0-7: pin which enables eMMC boot
* bits 8-15: pin which forces USB device-mode recovery boot
* bits 24-31: offset in the boot EEPROM (address bits 8-23)

OTP register 67:
* bits 0-8: pin to be activated before starting boot

ptesarik, 2019-07-09:
To sum it up, the factory settings for the bootmode register (17/18) are the only sensible values. One would have to modify the board itself to boot with any other values.

Registers 66 and 67 look interesting. I'm definitely going to try them out… once I find out how to write values to the OTP.

JamesH65, 2019-07-24:
This is really a docs issue, so if you want to progress that, please open an issue or provide  PR on the docs repo.

closed 2019-07-24 by JamesH65