raspberrypi/firmware #1170: Pi4 - No HDMI output when booting from USB key

bartekrutkowski, 2019-07-01:
I've tested two units of Pi4 4GB model and I can't get any HDMI output on any of the HDMI ports when booting from Samsung 64GB USB pendrive. When booting from SD card I have no issues with HDMI signal. 

So far I've tested the following:
* tested 2 different Pi4 4GB units
* tested 3 different HDMI cables
* tested Samsung 64GB USB 3.1 and Patriot 16GB USB 2.0 pendrives
* tested both HDMI ports
* tested `hdmi_force_hotplug=1` config.txt option
* tested `config_hdmi_boost=4` and `config_hdmi_boost=9` config.txt options

ghollingworth, 2019-07-01:
How are you booting from USB pendrive, USB boot is currently not supported on Pi 4, so you'll need to use an SD card with cmdline.txt to point at the pendrive for the root filesystem.

bartekrutkowski, 2019-07-01:
How come? I was under impression, that since Pi3+ supports it natively without any requirements (and older Pi's can also do it after programming USB boot mode), Pi4 would retain the same capabilities? Is there any information around that for Pi4?

ghollingworth, 2019-07-01:
Pi 4 is architecturally very different to Pi 3 when it comes to booting, the USB goes through PCIe and the Ethernet is direct rather than through USB.  Because of this, we decided to remove the support from the bootrom and instead implement the functionality in the SPI EEPROM.

The bad news is that we haven't quite finished this code yet (but are working on it), the good news in that now we've got the ability to fix booting problems by updating the EEPROM!

A first version of Ethernet boot shouldn't take too long...  MSD boot will be following afterwards

Gordon

bartekrutkowski, 2019-07-01:
Okay, thanks, that explains everything. Closing the issue.

closed 2019-07-01 by bartekrutkowski