raspberrypi/firmware #1172: RPi3B+: program_gpio_bootmode=1 bricked board

ptkatch, 2019-07-02:
Hi,
On 3B we used to run program_usb_boot_mode=1 and program_gpio_bootmode=1 in /boot/config.txt to select USB or SD as a boot media.

On 3B+, program_gpio_bootmode=1 makes board unbootable neither from USB, nor from SD.

How to recreate the issue:
- add program_gpio_bootmode=1 to /boot/config.txt 
- Boot 3B+ board 
- check OTP: # vcgencmd otp_dump | grep 17
17:3028000a
- poweroff
- boot: board will not boot anymore

Info (from 3B+ with OTP intact):
# vcgencmd version
Jun 29 2018 13:12:24 
Copyright (c) 2012 Broadcom
version 97765855d9ae607d0e678171cf9d83b218f237dd (clean) (release)

# vcgencmd otp_dump | grep 17
17:3020000a

# uname -a

inux RaspbianTest-WS140-v07-SD 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

# cat /proc/cpuinfo

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 0000000059999fea

pelwell, 2019-07-03:
_**WARNING: Don't use program_gpio_bootmode unless you are certain that you understand exactly what it does and have a plan for how it will work in your application.**_

3B+ (2837B0) supports two additional boot modes (USB mass storage and USB network) and uses a different pin assignment. This behaviour is part of the ROM of each processor - it cannot be changed - and the bits that enable the feature can only be set, not cleared.

There is a documentation page sitting in the queue for copy editing, but the all-important tables are:

## GPIO boot mode pin assignments
### Raspberry Pi 3B and Compute Module 3

 |Bank 1|Bank 2|boot type|
|:----:|:---:|:--------:|
|22    |39   |SD0       |
|23    |40   |SD1       |
|24    |41   |NAND (no Linux support at present)    |
|25    |42   |SPI (no Linux support at present)    |
|26    |43   |USB       |

### Raspberry Pi 3A+, 3B+ and Compute Module 3+

 |Bank 1|Bank 2|boot type|
|:----:|:---:|:--------:|
|20    |37   |SD0       |
|21    |38   |SD1       |
|22    |39   |NAND (no Linux support at present)    |
|23    |40   |SPI (no Linux support at present)    |
|24    |41   |USB device      |
|25    |42   |USB host - mass storage device |
|26    |43   |USB host - ethernet |

 ## Boot order

The various boot modes are attempted in the numerical order of the GPIO lines, i.e. SD0, then SD1, then NAND and so on.

ptkatch, 2019-07-03:
Thank you, Phil, didn't know pins changed. Yes, I am aware of the danger. We make terminals with 
button to select boot device. Works well - if SD gets corrupted, just plug USB in, and re-image SD without opening casing. 

I will re-wire button to new pins, and give it a try.

ptkatch, 2019-07-04:
Update, I did some testing, and confirm:
- /boot/config.txt: program_gpio_bootmode=1 DOES NOT "brick" 3B+ board, but changes OTP correctly
- pull-up GPIO20/Pin38 boots 3B+ with modified OTP from SD card
- pull-up GPIO25/Pin22 boots 3B+ with modified OTP from USB stick

Note, GPIO to pull-up for USB boot is 25 and not 24.

Phil, I have two questions:
Would you please guys, keep GPIO selection boot support in 4B boards (and, hopefully, without pin 
assignment change)? It is very helpful for us as it allows remote re-imaging SD card without opening 
case with board inside.

Second question, is it OK if I post solution with your info into my original post https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=243894&p=1488088#p1488088?

Thank you..

pelwell, 2019-07-05:
We will support GPIO bootmode on 4B once there is more than one bootmode, but we won't be using OTP to enable it. Instead there will be a user-modifiable setting in the EEPROM to enable it.

The public documentation page has now been updated: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/gpio.md
Feel free to add a link to it in your forum thread.

ptkatch, 2019-07-06:
Excellent! Thank you. Should I close the issue?

closed 2019-07-06 by pelwell