raspberrypi/firmware #1173: Overscan applied to RPI official display

paul-1, 2019-07-04:
Not sure if it's intentional, but over-scan is being applied with the RPI 7" display connected to a PI4
Adding disable_overscan=1 does correct the problem.

```[    0.050051] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-06-24 20:54, variant start_cd
[    0.060251] raspberrypi-firmware soc:firmware: Firmware hash is 39b2ae90606289b2b5d8ffba51f238892711eb4e
[    0.070666] clocksource: Switched to clocksource arch_sys_counter
[    0.097558] VFS: Disk quotas dquot_6.6.0
[    0.097596] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.097679] FS-Cache: Loaded
[    0.097777] CacheFiles: Loaded
[    0.098054] simple-framebuffer 3ef66000.framebuffer: framebuffer at 0x3ef66000, 0x95800 bytes, mapped to 0x(ptrval)
[    0.098060] simple-framebuffer 3ef66000.framebuffer: format=r5g6b5, mode=736x416x16, linelength=1472
[    0.098662] Console: switching to colour frame buffer device 92x26
[    0.099153] simple-framebuffer 3ef66000.framebuffer: fb0: simplefb registered!
```

popcornmix, 2019-07-04:
@JamesH65 this was lost in the multi framebuffer pr.
Used to do
```
if (info.display_num == DISPMANX_ID_MAIN_LCD)
      disable_overscan = 1;
```
but doesn't currently.

JamesH65, 2019-07-04:
Hmm, I wonder how that got dropped. Will check it out and fix tomorrow.

JamesH65, 2019-07-05:
I've just tried this with V3D driver on Pi4 and overscan is correct - is this is legacy graphics mode? Where I can definitely see it happening.

paul-1, 2019-07-05:
This is with dtoverlay=vc4-fkms-v3d set.   If I don’t have that overlay, the screen scaling is way off as well as the overscan issue.

JamesH65, 2019-07-05:
On a Pi4, using that driver, I see no overscan applied to the DSI display. During startup it does seem to have it applied, but then the FKMS DRM driver takes over and the overscan goes away before the desktop starts up.

paul-1, 2019-07-05:
This is application is a straight framebuffer application, not running the full desktop.

JamesH65, 2019-07-05:
OK, that would explain it.

popcornmix, 2019-07-05:
Latest rpi-update firmware should have a fix for this.

paul-1, 2019-07-05:
Looks good.

closed 2019-07-05 by paul-1