raspberrypi/firmware #1174: Documentation update needed for pi3-act-led dtoverlay

thagrol, 2019-07-04:
Current device tree overlay documentation does not mention anywhere the overlay to use to change the pin used for the ACT LED on a Pi 4.

Experimental evidence is that the pi3-act-led overlay does this.

Should the documentation be update and/or the overlay name changes?

pelwell, 2019-07-04:
This isn't the only overlay in this category - pi3-disable-bt, pi3-miniuart-bt and pi3-disable-wifi work on all Pis with onboard WiFi. The documentation lives in the Linux repo here: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/README

If you've got any suggested wording then feel free to contribute.

thagrol, 2019-07-04:
Fair enough. I'll give it some thought.

However my preferred solution would be to remove the "pi3-" prefix in cases like pi3-disable-bt, pi3-miniuart-bt and pi3-disable-wifi.

I realise there will be back compatibility issues with doing this but similar things have been done before e.g. the overlays relating to RTC modules.

pelwell, 2019-07-04:
The pi3- overlays are potentially widely used (I use pi3-disable-bt a lot), so I'd rather retain the originals names as clones. The new incarnations will still need documentation changest though.

pelwell, 2019-07-05:
Take a look at https://github.com/raspberrypi/linux/pull/3052 - is that what you had in mind?

thagrol, 2019-07-05:
Close but there's no mention of the 3A+ and CM3xx series in the updated documentation.

I have no access to these devices but am assuming they are the same as the 3B/3B+ they are derived from.

pelwell, 2019-07-05:
The 3A+ has WiFi, Bluetooth and the ACT LED on the GPIO expander, so it should be mentioned in the docs - good catch.

CM3 and CM3+ have no WiFi or Bluetooth, and although there is a GPIO expander there is no ACT LED on the CM itself and only 2 of the extra GPIOs are used - HPD and EMMC_EN - so none of the overlays apply to any CMs.

A followup commit adds the necessary 3A+ references.

thagrol, 2019-07-06:
Fair enough. Incidentally, at the time of writing my comments about 3A+ and CM3 series applied to several of the other pi3- overlays: disable-bt, disable-wifi, miniuart-bt, ...

Would it help if I opeend seperate issues for those and the CM3 series?

pelwell, 2019-07-06:
You'll have to explain what is needed for the CM3s, but I think I've already done the others.

thagrol, 2019-07-07:
I'm not the best person to do that as I have no access to them. That said...

It's my understabnd that CM3x and CM3+x use the same WiFi and Bluetooth hardware as the 3B/3B+ so should probably mentioned in the documentaion for pi3-disable-bt, pi3-disable-wifi, and pi3-miniuart-bt.

As that would make the list of models somewhat long (and doesn't future proof it) might it be better to replace "Pi 3B, 3B+, 3A+ and 4B" with "all Pi models with onboard WiFi and Bluetooth"

pelwell, 2019-07-07:
Show me where it says that CM3 or CM3+ has WiFi/BT and I'll fix the overlay documentation.

closed 2019-07-09 by pelwell