raspberrypi/firmware #1175: Can't boot non device-tree capable kernel on RPi4

9pi, 2019-07-05:
The Plan 9 kernel uses ATAGs, not device tree, and successfully boots on RPi3 using the latest `start.elf`.  On RPi4 with `start4.elf`, it boots and R2 is pointing to a flattened device tree.  According to documentation, I should be able to specify `device_tree=` in config.txt, or add a trailer with DTOK=n or DTOK tag missing, to boot with ATAGs instead.  But when I do either or both of these, the kernel does not boot at all.

Does `start4.elf` still support non device-tree capable kernels?

9pi, 2019-07-05:
It seems also that `start4.elf` won't boot a kernel binary file whose size isn't a multiple of 4 bytes, when the same file boots fine on RPi3.  Luckily that's easy to work around!

pelwell, 2019-07-05:
The very latest build (available as a download from this repo and via `rpi-update`) should work for non-Device Tree uses again. The problem was caused by the stubs exceeding 256 bytes, which is where the ATAGs are created - apparently some old/bare-metal projects expect them to be there.

I wasn't aware of the multiple-of-4 restriction, and off the top of my head I can't think of a reason why Pi4 would be different from Pi3 in this regard.

9pi, 2019-07-05:
Yes, it now works again.  Thank you!

closed 2019-07-05 by 9pi

pelwell, 2019-07-05:
Cool. If you have a non-multiple-of-4 "kernel" around to try with the new firmware I'd be grateful for the data point.

9pi, 2019-07-05:
> If you have a non-multiple-of-4 "kernel" around

Easily done: `echo >>/boot/kernel7l.img`  
Result won't boot, even with the new firmware.

pelwell, 2019-07-05:
Cheers - it shouldn't be hard to fix.

popcornmix, 2019-07-08:
Latest rpi-update firmware should have a fix for mon aligned "kernel" sizes. Please test.