raspberrypi/firmware #1178: RPI4 : PWM0 & PWM1 Alternate pins

ajefr, 2019-07-07:
Hello,
It looks like switching gpio 18 and gpio 19 to alternate function number 5 is no more working on raspberry pi 4.
On pi3 and below it allowed us to redirect sound pwm to gpio pins.
From now on, if you tried to switch pin 18 & 19 to pwm 0 & pwm 1, no signals are outputed to this pins.
I checked with wiring pi if alternate fuction is correctly selected and it is... 
Hope it's only a software bug.
Thanks

lategoodbye, 2019-07-07:
Please describe your actual steps and the used kernel version. It don't this has something to do with the firmware.

ajefr, 2019-07-07:
Steps are quite simple.
I have a working OS with buster installed and very last firmware.
On this OS, I'm using an overlay to swith pins to the correct alternate function upon boot.
If I try the OS with a pi 3b+, it's working.
if i try with the pi 4, it's not.

pelwell, 2019-07-07:
Don't bother. It's an intentional hardware change, not a software bug, but it's something we should be able to work around in the firmware so that the aud remap overlay works on Pi4 as intended.

ajefr, 2019-07-07:
Thanks, it's reassures me if it's a known "issue".
Hope that a solution will come. Maybe an option in the config.txt will be great.

pelwell, 2019-07-23:
BCM2711 has two PWM controllers, allowing one to be used for general PWM duties while the other is used for audio. The new controller - PWM1 - is only available on GPIOs 40 & 41, while PWM0 is on all the other PWM pins - 12, 13, 18, 19, 45, 52 & 53. Pi 4B drives the headphone output from GPIOs 40 & 41, meaning that the firmware has to use PWM1 for sound. This is what broke the audremap overlay for you.

A change has now been merged into our internal firmware software tree that detects PWM pin remapping and switches the audio output back to the original PWM block. This will appear in future firmware releases. As a bonus, the rpi-4.19.y Linux tree includes an enhanced `audremap` overlay that adds a `pins_18_19` parameter to select the other pair on the 40-pin header. Note that the firmware change should work with any overlay that modifies the `audio_pins` node - you aren't forced to use `audremap`.

popcornmix, 2019-07-23:
Latest rpi-update kernel includes this.

vince87, 2019-09-05:
hello, i'm trying the audremap overlay on pi4, but this happens
https://imgur.com/1QWK00h
similar happens also with pins 12/13

pelwell, 2019-09-06:
It works for me using either 12/13 or 18/19.

1. What distribution are you running?
2. `uname -a`
3. `vcgencmd version`
4. Without the overlay, `raspi-gpio get 12-19,40-45`
5. With the overlay in either mode, `raspi-gpio get 12-19,40-45`
6. `grep -E "^[^#]" /boot/config.txt`

pelwell, 2019-09-06:
7. `dmesg | grep Model`

vince87, 2019-09-06:
Thanks for the reply:

uname -a
Linux raspberrypi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux

vcgencmd version
Aug 15 2019 12:03:51
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)

raspi-gpio get 12-19,40-45 (without overlay)
GPIO 12: level=0 fsel=6 alt=2 func=DPI_D8 pull=NONE
GPIO 13: level=0 fsel=6 alt=2 func=DPI_D9 pull=NONE
GPIO 14: level=0 fsel=6 alt=2 func=DPI_D10 pull=NONE
GPIO 15: level=0 fsel=6 alt=2 func=DPI_D11 pull=NONE
GPIO 16: level=0 fsel=6 alt=2 func=DPI_D12 pull=NONE
GPIO 17: level=0 fsel=6 alt=2 func=DPI_D13 pull=NONE
GPIO 18: level=0 fsel=2 alt=5 func=PWM0_0 pull=DOWN
GPIO 19: level=0 fsel=2 alt=5 func=PWM0_1 pull=DOWN
GPIO 40: level=0 fsel=4 alt=0 func=PWM1_0 pull=NONE
GPIO 41: level=0 fsel=4 alt=0 func=PWM1_1 pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=0 func=INPUT pull=UP
GPIO 45: level=1 fsel=0 func=INPUT pull=UP

raspi-gpio get 12-19,40-45 (with overlay)
GPIO 12: level=0 fsel=6 alt=2 func=DPI_D8 pull=NONE
GPIO 13: level=0 fsel=6 alt=2 func=DPI_D9 pull=NONE
GPIO 14: level=0 fsel=6 alt=2 func=DPI_D10 pull=NONE
GPIO 15: level=1 fsel=6 alt=2 func=DPI_D11 pull=NONE
GPIO 16: level=0 fsel=6 alt=2 func=DPI_D12 pull=NONE
GPIO 17: level=0 fsel=6 alt=2 func=DPI_D13 pull=NONE
GPIO 18: level=0 fsel=2 alt=5 func=PWM0_0 pull=DOWN
GPIO 19: level=0 fsel=2 alt=5 func=PWM0_1 pull=DOWN
GPIO 40: level=1 fsel=0 func=INPUT pull=NONE
GPIO 41: level=1 fsel=0 func=INPUT pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=0 func=INPUT pull=UP
GPIO 45: level=1 fsel=0 func=INPUT pull=UP

grep -E "^[^#]" /boot/config.txt
hdmi_force_hotplug=1
dtparam=audio=on
[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
[all]
dtparam=i2c_vc=on
audio_pwm_mode=2
dtoverlay=audremap,pins_18_19
dtoverlay=pwm-2chan,pin=18,func=2,pin2=19,func2=2
disable_audio_dither=1
dtoverlay=vga666-6
enable_dpi_lcd=1
display_default_lcd=1
dpi_output_format=6
dpi_group=2
dpi_mode=87
hdmi_timings=480 1 14 45 56 300 1 10 5 5 0 0 0 60 0 9600000 1
dtoverlay=pi3-act-led,gpio=27
gpio=26=op,dl

dmesg | grep Model
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1

pelwell, 2019-09-06:
```
dtoverlay=audremap,pins_18_19
dtoverlay=pwm-2chan,pin=18,func=2,pin2=19,func2=2
```
You have two overlays both trying to use the same pins, which is never going to work. Precisely what are you trying to achieve? Is this about audio or other kinds of PWM?

pelwell, 2019-09-06:
You do mention sound in your original post, so try removing the pwm-2chan overlay and see if that solves your problem.

vince87, 2019-09-06:
Thank you so much for the tip!!!! now it works, I also tried to disable the pwm2chain overlay, but the result didn't change!
Then it occurred to me that my vga666-6 overlay also incorporates audio!!!
By disabling that, everything started to work!!!

closed 2019-09-06 by pelwell