raspberrypi/firmware #1179: Clock ID (mailbox property interface) of EMMC2

lategoodbye, 2019-07-07:
Is the the clock of the EMMC2 interface on the Pi 4 accessible via mailbox property interface (at least "Get clock rate" 0x00030002)?
Which ID does the clock have?

popcornmix, 2019-07-08:
Looks like it doesn't currently. Let me see if it can be added.

popcornmix, 2019-07-08:
Latest rpi-update firmware has support for this. Use clock id=12 for emmc2.

lategoodbye, 2019-07-09:
@vianpl @agherzan I think this is a necessary puzzle piece for the u-boot sdhci-bcm2835 driver, which get the current clock rate via mailbox interface.

agherzan, 2019-07-09:
That is definitely a good point @lategoodbye . Also in my case the dts was incorrectly using the old address for the SD card interface.

agherzan, 2019-07-09:
@lategoodbye there is also another bit that might be missing. The kernel uses a 1.8 regulator for the emmc2 interface.

lategoodbye, 2019-07-09:
I'm not sure what you meant with old address of the SD card interface. The emmc2 is an additional interface (3rd), so we need to extend the driver with an additional compatible for the emmc2 interface. Yes, the 1.8 regulator new but not essential for booting.

vianpl, 2019-07-09:
@mbgg might be interested

agherzan, 2019-07-09:
@lategoodbye that's basically what I meant. I'll make the changes, test and push to the same branch.

agherzan, 2019-07-09:
@popcornmix , can you also confirm the power device id?

popcornmix, 2019-07-09:
There is no separate power domain for emmc2. It will be powered as long as arm is running.

agherzan, 2019-07-09:
Thanks @popcornmix . I asked because I thought my current issue is somehow related to that. What happens is that the arm hangs as soon as I reset sdhci (0x2F). I've checked the ioaddr and it is `fe340000` and also the clock if used as being 12. Any brilliant ideas?

lurch, 2019-07-12:
I've just added EMMC2 to https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#clocks

P.S. Hello @agherzan :wave:

agherzan, 2019-07-12:
@lurch Nice. Thanks.

agherzan, 2019-07-12:
With some help from @mbgg https://github.com/agherzan/u-boot/tree/ag/rpi4 now contains support for both 32 and 64bit Raspberry Pi 4.

pelwell, 2019-07-12:
Nice work - congratulations!

JamesH65, 2019-07-24:
Closing this issue as questions answered/issue resolved.

closed 2019-07-24 by JamesH65