raspberrypi/firmware #1181: add support for 1440p

indritix, 2019-07-08:
I'm trying to configure the rpi4 to work with my BenQ_EW2770QZ in native resolution (2560x1440), so far no luck.
I've managed to get full HD (1080p) but can't go higher.
With the 43" 4k (2160p) TV my rpi4 works though.

tvservice -s outputs:
state 0xa [HDMI DMT (87) RGB full 16:9], 2560x1440 @ 60.00Hz, progressive

here's my config:
```
HDMI:EDID guessing the format to be 2560x1440p @60 Hz
#HDMI:EDID found unknown detail timing format: 2560x1440p hfp:47 hs:32 hbp:81 vfp:3 vs:5 vbp:33 pixel clock:241 MHz max pixel clock is 270 MHz

hdmi_ignore_edid=0xa5000080
hdmi_cvt=2560 1440 60 3
hdmi_drive=2
hdmi_group=2
hdmi_mode=87
#hdmi_mode=82
#the pixel clock is calculated from HxVxRR, but with adding all porch/blanking pixels into H & V
#@50Hz: hdmi_pixel_freq_limit=201416000
@60Hz: hdmi_pixel_freq_limit=241699200
#timing of the display
hdmi_timings=2560 0 47 32 81 1440 0 3 5 33 0 0 0 60 0 241699200 3
```

6by9, 2019-07-08:
Why are you ignoring the EDID?
The Pi4 is now using the vc4-fkms-v3d overlay by default, so it fully parses the EDID and will allow selection of all modes listed therein.

indritix, 2019-07-09:
the edid doesn't list a mode for 1440p, the highest I get is 1080p(which works when I select it)

timg236, 2019-07-09:
Please can you try removing hdmi_pixel_freq_limit=241699200 the default limit is 300000000 ? It might also be worth trying the rpi-update firmware/kernel if you have nothing important on the sd-card and are willing to try the bleeding edge code.

Do you have any other sources which output HDMI at 2560x1440 ? It's just possible that the HDMI supports only a subset of the display port resolutions. It's quite common for 4Kp60 to not be supported over HDMI1.4 inputs but admittedly I've not seen that limitation for QHD.

6by9, 2019-07-09:
- Use a stock Raspbian image.
- Connect your monitor to HDMI0.
- Remove all changes from config.txt.
- Reboot.
- In X run the Screen Configuration application
- Right click on the rectangle labelled HDMI-1.
- Does it include the other resolutions? If so click on one of the higher resolutions, and click the green tick. If not, then run ```tvservice -d edid.dat``` and attach the resulting file here.

The Pi4 is using the DRM/KMS stack for setting up the screen. hdmi_timings etc still have some use with legacy mode, but you then have no 3D acceleration.
As @timg236 has said, there are a couple of fixes relating to HDMI setup in the latest rpi-update release, so those would be worth trying.

indritix, 2019-07-09:
I'll try once back home with the default limit.
I already did a rpi-update yesterday, no luck.. sorry that's the only 1440p I have at the moment

indritix, 2019-07-09:
> * In X run the Screen Configuration application
> 
> * Right click on the rectangle labelled HDMI-1.

I don't see a Screen Config App, is there a cmd for that?

`17:22:48 [@raspberrypi:~] % edidparser edid.dat
Enabling fuzzy format match...
Parsing edid.dat...
HDMI:EDID version 1.3, 1 extensions, screen size 16x9 cm
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 24-75 Hz, horizontal is 26-81 kHz, max pixel clock is 230 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is SyncMaster
HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
HDMI:EDID established timing I/II bytes are BF EF 80
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x714F 8100 8140 8180 9500 950F A940 B300 
HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 0
HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 1
HDMI:EDID found DMT format: code 32, 1280x960p @ 60 Hz (4:3) in standard timing 2
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
HDMI:EDID found DMT format: code 48, 1440x900p @ 75 Hz (16:10) in standard timing 5
HDMI:EDID found DMT format: code 51, 1600x1200p @ 60 Hz (4:3) in standard timing 6
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 7
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 60 Hz (5)
HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz (native)
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz 
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz 
HDMI:EDID found CEA format: code 33, 1920x1080p @ 25Hz 
HDMI:EDID found CEA format: code 34, 1920x1080p @ 30Hz 
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
HDMI:EDID found Video Capability DB length 2
HDMI:EDID video capability: CE:3 IT:3 PT:0 QS:0
HDMI:EDID found HDMI VSDB length 7
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:yes, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:yes 30-bit:yes DC_yuv444:yes
HDMI:EDID HDMI VSDB max TMDS clock 225 MHz
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID adding mandatory support for CEA (1) 640x480p @ 60Hz
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
[... 12 lines cut ...]
HDMI:EDID best score mode is now CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz (score 5398248)
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 47185
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 2566472
HDMI:EDID DMT mode (17) 1024x768p @ 70 Hz with pixel clock 75 MHz has a score of 13762
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID DMT mode (18) 1024x768p @ 75 Hz with pixel clock 78 MHz has a score of 14745
HDMI:EDID CEA mode (19) 1280x720p @ 50 Hz with pixel clock 74 MHz has a score of 4663240
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 4232360
HDMI:EDID DMT mode (21) 1152x864p @ 75 Hz with pixel clock 108 MHz has a score of 43662
HDMI:EDID DMT mode (28) 1280x800p @ 60 Hz with pixel clock 83 MHz has a score of 86440
[... cut ...]
HDMI:EDID DMT mode (47) 1440x900p @ 60 Hz with pixel clock 106 MHz has a score of 102760
HDMI:EDID DMT mode (48) 1440x900p @ 75 Hz with pixel clock 136 MHz has a score of 49300
HDMI:EDID DMT mode (51) 1600x1200p @ 60 Hz with pixel clock 162 MHz has a score of 140200
HDMI:EDID DMT mode (58) 1680x1050p @ 60 Hz with pixel clock 146 MHz has a score of 130840
HDMI0:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0
`

6by9, 2019-07-09:
> attach the **resulting file** here.

The parsed thing is next to useless. With the raw file I can emulate your setup on my Pi4.

Raspbian Buster has "Screen Configuration" under the Preferences menu. Otherwise it is ```arandr```.

indritix, 2019-07-09:
oh sorry bout that, here the file.
[edid.zip](https://github.com/raspberrypi/firmware/files/3373787/edid.zip)
I don't have "Screen Configuration" under the Preferences menu, tried with randr but couldn't really select anything other than 1080p

6by9, 2019-07-09:
> I don't have "Screen Configuration" under the Preferences menu, tried with randr but couldn't really select anything other than 1080p

That comes back to my first step of using a vanilla Raspbian Buster download.
Please confirm that you are doing that with no changes to config.txt.

The quickest way to provide all the info to try and debug this is to run the ```raspinfo``` script from https://raw.githubusercontent.com/raspberrypi/utils/master/raspinfo/raspinfo and post that.

6by9, 2019-07-09:
Using your EDID file on my Pi4, I get the following as my list of supported modes:
```
pi@raspberrypi:~ $ DISPLAY=:0 xrandr
Screen 0: minimum 320 x 200, current 1440 x 900, maximum 7680 x 7680
HDMI-1 connected primary 1440x900+0+0 (normal left inverted right x axis y axis) 160mm x 90mm
   1920x1080     60.00 +  60.00    50.00    59.94    30.00    25.00    24.00    29.97    23.98  
   1600x1200     60.00  
   1680x1050     59.88  
   1280x1024     75.02    60.02  
   1440x900      74.98    59.90* 
   1280x960      60.00  
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00    60.00    50.00    50.00    59.94  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   720x576       50.00    50.00  
   720x480       60.00    60.00    59.94    59.94  
   640x480       75.00    72.81    66.67    60.00    59.94  
   720x400       70.08  
```
Using "Screen Configuration" (arandr) to set up the screen I get all those modes listed in the drop down menu, and my monitor seems quite happy with 1440x900@59.9 and 74.98Hz.

There have been some updates in the recent days to the firmware to address some HDMI clocking issues. Normal warnings over rpi-update apply, but it'd be worth updating to see if they are the cause of your issues.

indritix, 2019-07-09:
> my monitor seems quite happy with 1440x900@59.9 and 74.98Hz

by 1440p I mean 2560x1440@60

indritix, 2019-07-09:
`18:40:29 [@raspberrypi:~] % ./raspinfo                        
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2

Linux raspberrypi 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux
Hardware	: BCM2835
Revision	: c03111
Serial		: 10000000f368b1a2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul  9 2019 14:37:58 
Copyright (c) 2012 Broadcom
version d2b1b7fb01475cb3914b2086299e32d724e832f1 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30400236 3666740  25466124  13% /
devtmpfs         1867796       0   1867796   0% /dev
tmpfs            1999892       0   1999892   0% /dev/shm
tmpfs            1999892    8760   1991132   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1999892       0   1999892   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096   42675    215421  17% /boot
tmpfs             399976       8    399968   1% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20190708
raspberrypi-sys-mods:
  Installed: 20190429
openbox:
  Installed: 3.6.1-8+rpt2
lxpanel:
  Installed: 0.10.0-2+rpt4
pcmanfm:
  Installed: 1.3.1-1+rpt7
rpd-plym-splash:
  Installed: 0.19

Networking Information
----------------------

./raspinfo: line 67: ifconfig: command not found

USB Information
---------------

/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
[... 500 lines cut ...]
[   19.408484] [<c02223c0>] (warn_slowpath_null) from [<bf5a520c>] (drm_vblank_put+0xf0/0x100 [drm])
[   19.408618] [<bf5a520c>] (drm_vblank_put [drm]) from [<bf5a5240>] (drm_crtc_vblank_put+0x24/0x28 [drm])
[   19.408704] [<bf5a5240>] (drm_crtc_vblank_put [drm]) from [<bf92a654>] (vc4_crtc_handle_page_flip+0x50/0x60 [vc4])
[   19.408744] [<bf92a654>] (vc4_crtc_handle_page_flip [vc4]) from [<bf92bf48>] (vc4_crtc_irq_handler+0xc8/0x144 [vc4])
[   19.408767] [<bf92bf48>] (vc4_crtc_irq_handler [vc4]) from [<c0281594>] (__handle_irq_event_percpu+0xc0/0x22c)
[   19.408774] [<c0281594>] (__handle_irq_event_percpu) from [<c028173c>] (handle_irq_event_percpu+0x3c/0x8c)
[   19.408778] [<c028173c>] (handle_irq_event_percpu) from [<c02817e0>] (handle_irq_event+0x54/0x78)
[   19.408784] [<c02817e0>] (handle_irq_event) from [<c0285638>] (handle_fasteoi_irq+0xb4/0x194)
[   19.408790] [<c0285638>] (handle_fasteoi_irq) from [<c0280604>] (generic_handle_irq+0x34/0x44)
[   19.408797] [<c0280604>] (generic_handle_irq) from [<c0280d40>] (__handle_domain_irq+0x6c/0xc4)
[   19.408803] [<c0280d40>] (__handle_domain_irq) from [<c0202244>] (gic_handle_irq+0x4c/0x88)
[   19.408808] [<c0202244>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
[   19.408811] Exception stack(0xc1001ed8 to 0xc1001f20)
[   19.408815] 1ec0:                                                       c0209a84 00000000
[   19.408819] 1ee0: 40000093 40000093 c1004dbc c1000000 c1004e04 00000001 c10957ca c10a4a00
[   19.408823] 1f00: c0e67a38 c1001f34 c10051c4 c1001f28 00000000 c0209a88 40000013 ffffffff
[   19.408829] [<c02019bc>] (__irq_svc) from [<c0209a88>] (arch_cpu_idle+0x34/0x4c)
[   19.408837] [<c0209a88>] (arch_cpu_idle) from [<c09933e4>] (default_idle_call+0x34/0x48)
[   19.408844] [<c09933e4>] (default_idle_call) from [<c0254228>] (do_idle+0xec/0x17c)
[   19.408851] [<c0254228>] (do_idle) from [<c0254578>] (cpu_startup_entry+0x28/0x2c)
[   19.408856] [<c0254578>] (cpu_startup_entry) from [<c098cac0>] (rest_init+0xbc/0xc0)
[   19.408863] [<c098cac0>] (rest_init) from [<c0e00fe4>] (start_kernel+0x4b8/0x4e8)
[   19.408867] ---[ end trace d981ff50db96d3e0 ]---
[   22.383086] fuse init (API version 7.27)
18:40:36 [@raspberrypi:~] %`

6by9, 2019-07-09:
Sorry, saw the 1440 and my brain went the wrong way.

You really ought to complain to your monitor manufacturer then, as it doesn't advertise the mode at all in the EDID
```
pi@raspberrypi:~/tmp $ edid-decode edid.dat 
EDID version: 1.3
Manufacturer: SAM Model 5b0 Serial Number 0
Made in week 13 of 2009
Digital display
Maximum image size: 16 cm x 9 cm
Gamma: 2.20
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6396, 0.3300
  Green: 0.2998, 0.5996
  Blue:  0.1503, 0.0595
  White: 0.3125, 0.3291
Established timings supported:
  720x400@70Hz 9:5 HorFreq: 31469 Hz Clock: 28.320 MHz
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  640x480@67Hz 4:3 HorFreq: 35000 Hz Clock: 30.240 MHz
  640x480@72Hz 4:3 HorFreq: 37900 Hz Clock: 31.500 MHz
  640x480@75Hz 4:3 HorFreq: 37500 Hz Clock: 31.500 MHz
  800x600@56Hz 4:3 HorFreq: 35200 Hz Clock: 36.000 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
  800x600@72Hz 4:3 HorFreq: 48100 Hz Clock: 50.000 MHz
  800x600@75Hz 4:3 HorFreq: 46900 Hz Clock: 49.500 MHz
  832x624@75Hz 4:3 HorFreq: 49726 Hz Clock: 57.284 MHz
  1024x768@60Hz 4:3 HorFreq: 48400 Hz Clock: 65.000 MHz
  1024x768@70Hz 4:3 HorFreq: 56500 Hz Clock: 75.000 MHz
  1024x768@75Hz 4:3 HorFreq: 60000 Hz Clock: 78.750 MHz
  1280x1024@75Hz 5:4 HorFreq: 80000 Hz Clock: 135.000 MHz
  1152x870@75Hz 192:145 HorFreq: 67500 Hz Clock: 108.000 MHz
Standard timings supported:
  1152x864@75Hz 4:3 HorFreq: 67500 Hz Clock: 108.000 MHz
  1280x800@60Hz 16:10
  1280x960@60Hz 4:3 HorFreq: 60000 Hz Clock: 108.000 MHz
  1280x1024@60Hz 5:4 HorFreq: 64000 Hz Clock: 108.000 MHz
  1440x900@60Hz 16:10 HorFreq: 55500 Hz Clock: 88.750 MHz
  1440x900@75Hz 16:10 HorFreq: 82300 Hz Clock: 156.000 MHz
  1600x1200@60Hz 4:3 HorFreq: 75000 Hz Clock: 162.000 MHz
  1680x1050@60Hz 16:10 HorFreq: 64700 Hz Clock: 119.000 MHz
Detailed mode: Clock 148.500 MHz, 160 mm x 90 mm
               1920 2008 2052 2200 hborder 0
               1080 1084 1089 1125 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 67500 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1280 1720 1760 1980 hborder 0
                720  725  730  750 vborder 0
               +hsync +vsync 
               VertFreq: 50 Hz, HorFreq: 37500 Hz
Monitor ranges (GTF): 24-75Hz V, 26-81kHz H, max dotclock 230MHz
Monitor name: SyncMaster
Has 1 extension blocks
Checksum: 0xd1 (valid)

CTA extension block
Extension version: 3
31 bytes of CTA data
  Video data block
    VIC  19 1280x720@50Hz 16:9 (native) HorFreq: 37500 Hz Clock: 74.250 MHz
    VIC   4 1280x720@60Hz 16:9  HorFreq: 45000 Hz Clock: 74.250 MHz
    VIC   5 1920x1080i@60Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
    VIC  20 1920x1080i@50Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC   3 720x480@60Hz 16:9  HorFreq: 31469 Hz Clock: 27.000 MHz
    VIC  18 720x576@50Hz 16:9  HorFreq: 31250 Hz Clock: 27.000 MHz
    VIC  16 1920x1080@60Hz 16:9  HorFreq: 67500 Hz Clock: 148.500 MHz
    VIC  31 1920x1080@50Hz 16:9  HorFreq: 56250 Hz Clock: 148.500 MHz
    VIC  32 1920x1080@24Hz 16:9  HorFreq: 27000 Hz Clock: 74.250 MHz
    VIC  33 1920x1080@25Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC  34 1920x1080@30Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
  Audio data block
    Linear PCM, max channels 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 24 20 16
  Speaker allocation data block
    Speaker map:
      FL/FR - Front Left/Right
  Extended tag: Video capability data block
    YCbCr quantization: No Data (0)
    RGB quantization: No Data (0)
    PT scan behaviour: No Data (0)
    IT scan behaviour: Support both over- and underscan (3)
    CE scan behaviour: Support both over- and underscan (3)
  Vendor-specific data block, OUI 000c03 (HDMI)
    Source physical address 1.0.0.0
    Supports_AI
    DC_36bit
    DC_30bit
    DC_Y444
    Maximum TMDS clock: 225MHz
Underscans PC formats by default
Basic audio support
Supports YCbCr 4:4:4
Supports YCbCr 4:2:2
1 native detailed modes
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1920 2448 2492 2640 hborder 0
                540  542  547  562 vborder 0
               +hsync +vsync interlaced 
               VertFreq: 50 Hz, HorFreq: 28125 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1920 2008 2052 2200 hborder 0
                540  542  547  562 vborder 0
               +hsync +vsync interlaced 
               VertFreq: 60 Hz, HorFreq: 33750 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1280 1390 1430 1650 hborder 0
                720  725  730  750 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 45000 Hz
Detailed mode: Clock 27.000 MHz, 160 mm x 90 mm
                720  732  796  864 hborder 0
                576  581  586  625 vborder 0
               -hsync -vsync 
               VertFreq: 50 Hz, HorFreq: 31250 Hz
Detailed mode: Clock 27.000 MHz, 160 mm x 90 mm
                720  736  798  858 hborder 0
                480  489  495  525 vborder 0
               -hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 31468 Hz
Checksum: 0x57 (valid)
```
You should be able to manually add modes via xrandr in the same way as any other Linux machine, eg https://ubuntuforums.org/showthread.php?t=1112186, but it's throwing errors at present.

```hdmi_timings``` currently doesn't play well with KMS. I will look at whether it can be improved, but I'm afraid that displays with dodgy EDIDs are currently a pain.
NB That this will be fixed in the Linux kernel, not in the firmware. Sorry, but I'm not messing with the firmware EDID parser unless I really have to.

Thanks for the raspinfo dump - at least I now know what your system is doing.

indritix, 2019-07-09:
> but I'm afraid that displays with dodgy EDIDs are currently a pain

actually my edidparser shows that my monitor advertises itself correctly, the problem is it's resolution seems not to be supported :(
`HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is BenQ_EW2770QZ
HDMI:EDID failed to find a matching detail format for 2560x1440p hfp:47 hs:32 hbp:81 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
HDMI:EDID calculated refresh rate is 60 Hz
HDMI:EDID guessing the format to be 2560x1440p @60 Hz
HDMI:EDID found unknown detail timing format: 2560x1440p hfp:47 hs:32 hbp:81 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
HDMI:EDID established timing I/II bytes are A5 6B 80`

6by9, 2019-07-09:
In which case that isn't the EDID file that you attached.

Actually I'm noticing from my parser
```
Manufacturer: SAM Model 5b0 Serial Number 0
Made in week 13 of 2009
...
Monitor name: SyncMaster
```
Sounds like a Samsung Syncmaster to me, not a BenQ.

indritix, 2019-07-09:
I just re-dumped it, it should contain the correct infos.
[edid.zip](https://github.com/raspberrypi/firmware/files/3374096/edid.zip)

6by9, 2019-07-09:
```
Monitor name: BenQ EW2770QZ
```
Looks more plausible.

6by9, 2019-07-09:
So that parses reasonably, and xrandr reports a sensible list of resolutions, including 2560x1440@59.95
```
pi@raspberrypi:~ $ DISPLAY=:0 xrandr
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 597mm x 336mm
   2560x1440     59.95 +
   1920x1080     60.00*   60.00    50.00    59.94    30.00    25.00    24.00    29.97    23.98  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00    60.00    50.00    59.94  
   1024x768      75.03    60.00  
   832x624       74.55  
   800x600       75.00    60.32  
   720x576       50.00    50.00    50.00  
   720x480       60.00    60.00    59.94    59.94    59.94  
   640x480       75.00    60.00    59.94    59.94  
   720x400       70.08  
```
My 4k monitor doesn't like 2560x1440 though, so I can't check if the Pi is actually producing a sensible output for that mode. I'll have to try and hunt down an alternate monitor.

6by9, 2019-07-09:
Monitor supporting 2560x1440 @ 59.95 found, and it works fine.

Sorry, there's not much more that we can do here. There are some cables that are marginal and will fail at higher frequencies, but if the same cable is doing 4k to your TV then that sounds unlikely.

indritix, 2019-07-09:
hm... my xrandr looks different though:
`20:37:41 [:~] % DISPLAY=:0 xrandr 
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1920 x 1080, current 1920 x 1080, maximum 1920 x 1080
default connected 1920x1080+0+0 0mm x 0mm
   1920x1080      0.00* 
20:38:15 [:~] % sudo xrandr --newmode "2560x1440_59.95" 241.50  2560 2608 2640 2720 1440 1443 1448 1481 +hsync -vsync
xrandr: Failed to get size of gamma for output default
20:40:01 [:~] % sudo xrandr --addmode HDMI-1 "2560x1440_59.95"
xrandr: Failed to get size of gamma for output default
xrandr: cannot find output "HDMI-1"
20:40:24 [:~] 1 %`

ghollingworth, 2019-07-10:
Can you please download the latest Raspbian from the raspberrypi.org website and install the desktop version. 

Then try again

6by9, 2019-07-10:
```
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1920 x 1080, current 1920 x 1080, maximum 1920 x 1080
default connected 1920x1080+0+0 0mm x 0mm
```
that looks like you aren't using the FKMS driver, although your dmesg appears to say it is loaded.

As Gordon says, please try with a fresh install - something looks very screwy with your current one.

indritix, 2019-07-10:
ok, with a fresh install I'm getting some hope ;)
```
pi@raspberrypi:~ $ DISPLAY=:0 xrandr 
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 597mm x 336mm
   2560x1440     59.95 +
   1920x1080     60.00*   60.00    50.00    59.94    30.00    25.00    24.00    29.97    23.98  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00    60.00    50.00    59.94  
   1024x768      75.03    60.00  
   832x624       74.55  
   800x600       75.00    60.32  
   720x576       50.00    50.00    50.00  
   720x480       60.00    60.00    59.94    59.94    59.94  
   640x480       75.00    60.00    59.94    59.94  
   720x400       70.08  
pi@raspberrypi:~ $
```
but when I select 1440p in "Screen Configuration" (this time it shows up), no changes are made after I click apply (I can see it tries, but unfortunately no luck), and the resolution reverts back to 1080p.
Maybe this time my edid.dat will help.
[edid.zip](https://github.com/raspberrypi/firmware/files/3377564/edid.zip)

indritix, 2019-07-10:
tried with `xrandr --output HDMI-1 --mode 2560x1440 --rate 59.95`, nothing, a loop starts and no screen is detected.
`xrandr --verbose` shows:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
HDMI-1 connected primary 1920x1080+0+0 (0x45) normal (normal left inverted right x axis y axis) 597mm x 336mm
	Identifier: 0x41
	Timestamp:  38842
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       0
	CRTCs:      0
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	EDID: 
		00ffffffffffff0009d1467945540000
		141c0103803c22782e6665a9544c9d26
		105054a56b80d1c0b300a9c081808100
		81c001010101565e00a0a0a029502f20
		350055502100001a000000ff0046354a
		30303431333031510a20000000fd0032
		4c1e5a1b000a202020202020000000fc
		0042656e5120455732373730515a0112
		020323f150901f222120051404131211
		0302010706230907078301000065030c
		001000011d007251d01e206e28550055
		502100001e8c0ad08a20e02d10103e96
		005550210000188c0ad090204031200c
		4055005550210000188c0ad090204031
		200c4055005550210000180000000000
		000000000000000000000000000000e4
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 51 
		supported: 51
	non-desktop: 0 
		range: (0, 1)
  2560x1440 (0x44) 241.500MHz +HSync -VSync +preferred
        h: width  2560 start 2607 end 2639 total 2720 skew    0 clock  88.79KHz
        v: height 1440 start 1443 end 1448 total 1481           clock  59.95Hz
  1920x1080 (0x45) 148.500MHz -HSync -VSync *current
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock  67.50KHz
        v: height 1080 start 1084 end 1089 total 1125           clock  60.00Hz
  1920x1080 (0x46) 148.500MHz +HSync +VSync
        h: w
```

indritix, 2019-07-13:
tried with latest firmware, still impossible to go higher than 1080p.
Why I can select the 1440p resolution in "screen configuration" (or alternatively with xrandr on a terminal) but then after clicking apply the changes don't apply and the resolution reverts to 1080p is still a mystery to me.
if you need logs/traces just let me know..
```
pi@raspberrypi:~ $ vcgencmd version
Jul  9 2019 14:37:58 
Copyright (c) 2012 Broadcom
version d2b1b7fb01475cb3914b2086299e32d724e832f1 (clean) (release) (start)
pi@raspberrypi:~ $ vcgencmd get_config int
arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=250
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=500
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024 
mask_gpu_interrupt1=0x10000
max_framebuffers=2
pause_burst_frames=1
program_serial_random=1
hdmi_drive:0=2
hdmi_force_cec_address:0=65535fingerprint
```
hdmi_force_cec_address:1=65535
hdmi_force_hotplug:0=1
hdmi_group:0=2
hdmi_mode:0=82
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
pi@raspberrypi:~ $

indritix, 2019-07-17:
Ok, I managed to solve the problem with xrandr (thanks to the archlinux wiki), would be nice if there was a way to translate xrandr parameters into config.txt entries though.

closed 2019-07-17 by indritix

sooriravindra, 2019-09-12:
@indritix Could you please post your solution? I am facing a similar issue. xrandr isn't able to set the resolution either.

r0guebyte, 2019-10-28:
@indritix @sooriravindra 

Hello, I've recently tried attaching my Raspi 4b to this same monitor and cannot get it to work with 1440p.  I can select 1080p from the drop down menu in preferences and it works fine, but if I try 1440p it goes to a black screen and is stuck there. 

Can someone share their solution?

Thanks.

Luthor2k, 2019-11-19:
Got it working, I'm using a BenQ pd-2500t, same issues others here face,
I edited config.txt and added:
hdmi_group=2
hdmi_mode=87
hdmi_cvt=2560 1440 60 3 0 0 1
max_framebuffer_width=2560
max_framebuffer_height=1440
hdmi_pixel_freq_limit=400000000

for some reason my wifi adapter will not now grab a real IP but wired works fine so w/e...

illmillrig, 2020-05-10:
> Got it working, I'm using a BenQ pd-2500t, same issues others here face,
> I edited config.txt and added:
> hdmi_group=2
> hdmi_mode=87
> hdmi_cvt=2560 1440 60 3 0 0 1
> max_framebuffer_width=2560 [...]

now do it for dual screen.

it's absolutely a face plant before the finish line with the pi4 not supporting this resolution.  It is one of the most common resolutions in displays right now.  What a tragedy.  This Pi is pretty much worthless to me now because it doesnt support one of the most common resolutions, and yet touts itself as supporting dual 4k.

what a joke.

6by9, 2020-05-10:
> now do it for dual screen.
```
hdmi_group:1=2
hdmi_mode:1=87
hdmi_cvt:1=2560 1440 60 3 0 0 1
hdmi_pixel_freq_limit:1=400000000
```
as in the docs https://www.raspberrypi.org/documentation/configuration/config-txt/video.md section "HDMI mode options / Note for Raspberry Pi4B users".

You're obviously using a Pi4 if you're wanting dual screens, therefore the FKMS driver is active by default. X will select the highest mode automatically as it reparses the EDID and selects a mode of its choosing.
If using the console only, then as an alternative to the above you can add ```disable_fw_kms_setup=1``` to config.txt, and again the kernel will reparse the EDID for choosing a default frame buffer mode.
If you've disabled FKMS, then you can hardly complain that it doesn't work out of the box.

Interesting that we had the same issue on the forum in the last day or so. Is that you too and you felt like a further complain? https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=273548

illmillrig, 2020-05-10:
doesn't work for me, or I wouldn't have "complained".

I'm sorry, what are you accusing me of?

illmillrig, 2020-05-10:
not even one display works at 2560x1440

JamesH65, 2020-05-10:
> it's absolutely a face plant before the finish line with the pi4 not supporting this resolution. It is one of the most common resolutions in displays right now. What a tragedy. This Pi is pretty [...]

Is it? I've never come across a monitor of that resolution. 

It's not in the standard list of 4K resolutions either. Wikipedia has a list https://en.wikipedia.org/wiki/4K_resolution, all of which I believe we support.

illmillrig, 2020-05-10:
> Is it? I've never come across a monitor of that resolution.

most 27"-31" displays are this resolution. If you've bought a monitor in the last 10 years, you've probably seen it ;)

illmillrig, 2020-05-10:
monitor shows up as 1824x984, and I only get one screen.  It's strange because I get the color wheel image on screen1, but everything else is on screen0 during boot and after

6by9, 2020-05-10:
> monitor shows up as 1824x984, and I only get one screen.

You've disabled the line ```dtoverlay=vc4-fks-v3d```, and have overscan enabled.
The overlay is enabled by default, so what is your reasoning for disabling it?
Start again with a clean Raspbian install and X will select 2560x1440 on both monitors.

illmillrig, 2020-05-10:
I have not disabled anything.  Getting tied of your accusations though.

illmillrig, 2020-05-10:
im reflashing now.  I've tried multiple clean installs though.  I also tried ubuntu just in case.

illmillrig, 2020-05-10:
same thing. no luck.

JamesH65, 2020-05-10:
> I have not disabled anything. Getting tired of your accusations though.

OK, and that's all the help you get from me. Being rude to the people who are trying to help you, coming up with nonsense like "the last 10 years" when monitors of that resolution have not been available for anywhere near that long.  I'm out of here.

illmillrig, 2020-05-10:
> OK, and that's all the help you get from me. Being rude to the people who are trying to help you, coming up with nonsense like "the last 10 years" when monitors of that resolution have not been [...]

niether of you have attempted to help at all.  You have accused me of creating duplicate accounts so that I can complain under two names??? is that something you've done?

if you had looked closer, I think it would have been pretty obvious that was not the case, but you just jumped to conclusions it was because the same topic came up twice in two weeks?  seriously?  delusional, much?

you then went on to accuse me of making changes to things that I have not.  you didn't ask me if I made those changes, you accused me. 

you saying that you've never seen a monitor with this resolution is ignorance on your part.  I'm guessing that's why you're so angry, it made you feel stupid, and you're attempting to play the "you're rude" card.

you really weren't helping anyways.  so no real loss there.  all you said was that you never came across a monitor with that resolution, and for that I think you've been living under a rock.  do you actually think that was going to be helpful?  telling me you never saw a monitor with this resolution.  what was I supposed to do with that info?  throw it away? pretend it has a different resolution?  add/remove pixels myself? 

funny way of helping.

illmillrig, 2020-05-10:
btw, 2560x1440 has been around ~10 years.  do a little research first. 2560x1600 is the 16:10 ar.  Apple had been making the 30" cinema display with this res for a long time, much longer than 10 years ago, and other panels have had the res also.  some time around 10 years ago, they started making 16:9 in that res.  guess what the pixel dimensions are... 2560x1440

illmillrig, 2020-05-10:
it is absolutely one of the most common resolutions in displays/  and the pi4 doesn't support it right out of the box.

you can't buy a brand new pi4, and a brand new 27" display, and use them together natively.

illmillrig, 2020-05-10:
https://en.wikipedia.org/wiki/Graphics_display_resolution#2560_%C3%97_1440_(QHD,_WQHD)

illmillrig, 2020-05-10:
https://www.newegg.com/p/pl?N=100160979%20600012694

illmillrig, 2020-05-10:
newegg has more monitors at 2560x1440 than every other resolution except for 1920x1080.