raspberrypi/firmware #1183: script to generate my own firmware

artynet, 2019-07-10:
Hello folks,

maybe I have missed it looking into the repository. Have you provided a set of scripts to reproduce the binaries uploaded in this repository ? If not, would you mind listing all the steps to do it ?

Thank in advance

pelwell, 2019-07-10:
1. The kernel build process it described in https://www.raspberrypi.org/documentation/linux/kernel/building.md
2. Userland libraries can be built using https://github.com/raspberrypi/userland/blob/master/buildme
3. The remainder of the firmware (bootcode.bin, start*.elf, etc.) is built from unpublished sources that are proprietary to Broadcom and Raspberry Pi.

artynet, 2019-07-10:
thank you @pelwell

closed 2019-07-10 by pelwell