raspberrypi/firmware #1188: Raspberry Pi 4 crashes randomly

wrnice, 2019-07-13:
Hello, 

I am not sure if it should go to firmware or kernel, as I don't find anything relevant in logs.

I am experimenting repeated crashes with the raspberry pi 4.

*** Hardware : raspberry pi 4 4GB rev 1.1 , official raspberry 5v 3a power suply , usb mouse, usb keyboard , dvi-d display 1920x1080 @ 59 Hz , micro hdmi to dvi-d cable, new Sandisk Ultra 32 Gb sdcard

*** Software : raspbian Buster downloaded from the raspberry website , either as-is or after apt update / apt upgrade ( July the 13rd 2019 ) 

*** description :  The raspberry hangs without a warning. The power led stays bright lit, keyboard leds stay lit, gigabyte leds stay lit, the screen goes blank but backlight stays on. At the moment the board goes down, on car hear the audio disconnecting if loudspeakers are present. Network goes down ( can't ping or ssh into the pi ).

The only way to "revive" the board is to cycle power off then back on.

fsck says both filesystems are clean ( /boot and /root ).

*** how to reproduce : 
Use the pi for enough time until it crashes again. Sometimes a few minutes, sometimes hours. It also happens when the pi is idling ( ie, no mouse or keyboard input, no active window on the desktop )

It doesn't seem to be related with browser usage, and setting sdram_freq to 2800 in /boot/config.txt doesn't help.

I would love to be of more help, but I am not sure how to give more precise information as the logs are not saying anything that looks relevant.

popcornmix, 2019-07-15:
sdram_freq doesn't support arbitrary frequencies on pi4. The only other frequency supported currently is 1350 which you could try, 2667 will be added with a future spi eeprom update.

It might be worth trying ``over_voltage=1`` (or 2 or 3) in config.txt.

If you have a 3.3V uart then there may be more clues in its output.

Are you sure the display goes off when the Pi 4 crashes? (rather than console timeout from no activity). That is a surprising behaviour.

Setting ``core_freq_min=500`` may be worth also be worth try.

wrnice, 2019-07-15:
Hello ! Thank you for these details, it seems I may not need to try that, as after having upgrade firefox-ESR to 60.8.0esr-1~deb10u1+rpi1, I didn't have any crashes in 48 hours. It seems the previous versions were not compiled against the correct target. 
I may well have let a firefox window open on another desktop the night when it crashed "by itself".

What I don't get, is how firefox-esr could have crashed the pi so badly, that even after a reboot and without starting firefox-esr again, it would make the pi crash. 

I will let a few days pass, and if I don't see crashes anymore I will close this issue. If it happens again I will try the frequency and voltage tweaking that you mentionned.

Thank you !

qqgg231, 2019-07-21:
I did have such crashes after apt upgrade about week or two ago. After reflashing with clean image and not upgrading I don't have crashes anymore

Note it is probably connected to some update which was applied to boot partition, because when I tried use updated boot partition with old image it was still crashing

wrnice, 2019-07-21:
It has been a week without a single crash, and it seems qqgg231 is fine too, so I think this issue is no more an issue !
I will close it now.
Thank you !

closed 2019-07-21 by wrnice

qqgg231, 2019-07-29:
My issue was because of faulty power supply. It is original usb-c power supply from Raspberry Foundation but it is delivering not enough voltage, about 4,5v. Shop agreed to send me new one