raspberrypi/firmware #1189: new firmware and RFIC-MOUSE problem

bacsom, 2019-07-14:
In our school, we use dozens of T3 AirMouses and since the latest versions of Stretch and Buster they don't work correctly. When I click on something the OS starts registering multiple mouse events (scrolling, clicking, etc...) up to the point that the system almost freeze. It must be a firmware related issue, because reverting back to an older firmware (using the command rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b) solves the problem. Unfortunately, this is only a solution under Stretch, because in Buster it brakes Chromium. 

Any help would be greatly appreciated!

P33M, 2019-07-15:
You say a08ece3d48c3c40bf1b501772af9933249c11c5b is broken, but which is the first version of rpi-update firmware that shows the issue?

bacsom, 2019-07-15:
Thank you for your answer!

I used this website to see the available versions:
https://github.com/Hexxeh/rpi-firmware/commits/master?after=f8060224e53e26cdaea5b9f6f010ac8c3692e2ea+34

Nothing works since the Feb 12. 2019 (4.14.98) update. I have tried almost all of them. Do you know if there is anything between 4.14.98 and 4.19.23? Buster seems to work with 4.14 except Chromium.

P33M, 2019-07-15:
In that case it's likely to be an upstream bug, since there were no (or very few) downstream changes other than those required to get 4.19 compiling/booting correctly. In particular there were no downstream changes to the USB host driver.

Can you post the output of `sudo lsusb -v` and `dmesg` with the device plugged in? Also, where can I buy one of these devices?

bacsom, 2019-07-15:
Thank you again for your time, I am really grateful that somebody looks into it.

**working 4.14.98:**
[lsubs 4.14.98](https://pastebin.com/m6w8NkjQ)
[dmesg 4.14.98](https://pastebin.com/ttHYFNPJ)

**not working 4.19.23:**
[lsubs 4.19.23](https://pastebin.com/TgshpndG)
[dmesg 4.19.23](https://pastebin.com/g3HRQ1Zm)

You can buy these under several brands and places (offline or online). For example [Amazon ](https://www.amazon.com/LeSunDa-Wireless-Keyboard-without-Control/dp/B071WGZQ6S/ref=sr_1_3?keywords=t3+air+mouse&qid=1563205329&s=gateway&sr=8-3) or [Aliexpress](https://www.aliexpress.com/item/32608846530.html?spm=a2g0s.9042311.0.0.27424c4dWmr9QU). There are different versions, MX3, for example, has fewer keys, I don't know if it has the issue or not. I can replicate the problem with several T3s and Raspberries (Pi3 and Pi3+) and it only occurs when the gyroscope is active.

lwpz, 2019-07-17:
If it is a firmware issue, then we would need @JamesH65 or @popcornmix to comment. The kernel should be handling devices and not the firmware (if I'm not wrong)...

bacsom, 2019-07-17:
I don't know either, unfortunately I am not a Linux expert :-(

I have checked the output of dmesg and it seems that 4.19.23 recognise more devices than 4.14.98. Is it possible to simply disable the extras until somebody can figure out what is wrong?

4.14.98
[    2.332113] usb 1-1.2: New USB device found, idVendor=25a7, idProduct=2402
[    2.332128] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.332137] usb 1-1.2: Product: RFIC-MOUSE
[    2.332145] usb 1-1.2: Manufacturer: FREEWAY TECHNOLOGY
[    2.345128] input: FREEWAY TECHNOLOGY RFIC-MOUSE as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:25A7:2402.0001/input/input0
[    2.412506] hid-generic 0003:25A7:2402.0001: input,hiddev96,hidraw0: USB HID v1.01 Keyboard [FREEWAY TECHNOLOGY RFIC-MOUSE] on usb-3f980000.usb-1.2/input0
[    2.419064] input: FREEWAY TECHNOLOGY RFIC-MOUSE as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.1/0003:25A7:2402.0002/input/input1
[    2.419390] hid-generic 0003:25A7:2402.0002: input,hidraw1: USB HID v1.01 Mouse [FREEWAY TECHNOLOGY RFIC-MOUSE] on usb-3f980000.usb-1.2/input1

4.19.23
[    2.383867] usb 1-1.2: New USB device found, idVendor=25a7, idProduct=2402, **bcdDevice= 2.00**
[    2.383882] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.383891] usb 1-1.2: Product: RFIC-MOUSE
[    2.383900] usb 1-1.2: Manufacturer: FREEWAY TECHNOLOGY
**[    2.398225] input: FREEWAY TECHNOLOGY RFIC-MOUSE Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:25A7:2402.0001/input/input0
[    2.462932] input: FREEWAY TECHNOLOGY RFIC-MOUSE Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:25A7:2402.0001/input/input1
[    2.463117] input: FREEWAY TECHNOLOGY RFIC-MOUSE System Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:25A7:2402.0001/input/input2**
[    2.463316] input: FREEWAY TECHNOLOGY RFIC-MOUSE as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:25A7:2402.0001/input/input3
[    2.463757] hid-generic 0003:25A7:2402.0001: input,hiddev96,hidraw0: USB HID v1.01 Keyboard [FREEWAY TECHNOLOGY RFIC-MOUSE] on usb-3f980000.usb-1.2/input0
[    2.469265] input: FREEWAY TECHNOLOGY RFIC-MOUSE as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.1/0003:25A7:2402.0002/input/input5
[    2.469657] hid-generic 0003:25A7:2402.0002: input,hidraw1: USB HID v1.01 Mouse [FREEWAY TECHNOLOGY RFIC-MOUSE] on usb-3f980000.usb-1.2/input1

P33M, 2019-07-17:
Annoyingly, lsusb doesn't dump the HID report descriptors. 

Can you run `sudo apt-get install hidrd` and run the command `sudo usbhid-dump -d 25a7:2402 | grep -v : | tr -d ' \n' | xxd -r -p | hidrd-convert -o spec`?

bacsom, 2019-07-17:
Of course. [Here](https://pastebin.com/mrpgJv5z).

rjsachse, 2019-07-18:
Having same issues with freeway mx3 mouse, upgraded stretch had issues with mouse, then installed buster and same issue. On stretch I have SSH and noticed CPU goes very high and runs out of memory and logs out, on buster mouse moves but clicks work.

bacsom, 2019-07-18:
Yes, I also noticed the high CPU usage under Stretch. So MX3 is also affected. :-(

P33M, 2019-07-18:
Can you try adding usbhid.mousepoll=0 to cmdline.txt?

P33M, 2019-07-18:
If mousepoll doesn't change the behaviour then can you
a) remove the cmdline.txt option
b) run `sudo apt-get install evtest` and run it - you will get an option prompt, pick the input device corresponding to the mouse (I would guess input4).
c) post the output from evtest when the mouse has caused lag/misclick events.

bacsom, 2019-07-18:
usbhid.mousepoll=0 had no effect at all :-(

[Here ](https://pastebin.com/f80YaMyY)is the log of evtest. The screen is quickly filled with these lines...

If it helps I am willing to order a T3 for you. In the school, we have about 30 classrooms with Raspberry based multimedia centres and the teachers operate them with the help of these remotes.

P33M, 2019-07-18:
The clue is that the mouse is spamming scrollwheel events with completely nonsensical offsets - if I plug in a mouse and do the same test, I get one or two +1 / -1 events for each click of the wheel. Yet that device is reporting 6 mousewheel events every 16ms.

If you can email info@raspberrypi.org and quote the github issue in the link, we can start a dialogue about getting hold of the problem hardware.

bacsom, 2019-07-18:
Email has been sent :-).

lwpz, 2019-07-19:
I think that you're posting this in the wrong forum. If it's kernel related (which looks so), then it should be posted here. [https://github.com/raspberrypi/linux/issues](url)

HID devices are handled by the kernel and not the firmware (for GPU). Please post in the relevant forum to get a reply.

bacsom, 2019-07-20:
Dear lwpz,

Thank you, I didn't know that. If it is possible I agree to move the whole topic under a different category. Can you help me with that? Because there are several comments here I think it would be a mistake to start it from scratch.

P.S.: The link does not work correctly, it points back to the firmware category.

lwpz, 2019-07-20:
Hi Bacsom,

I’m not able to move. You’ll need to post again and put a link to this post

Try this link. https://github.com/raspberrypi/linux/issues

You can see it being “Linux” and not “Firmware”.

If the older kernel works for you, I don’t see why you can’t just stick to that?

You might want to consider to speak to your manufacturer of the hardware?

Thanks

Regards,
Wern Pin

bacsom, 2019-07-20:
It has crossed my mind to write to the manufacturer directly, and maybe I will do so, but because these remotes are sold under different brands it is not as easy as it looks.

As I mentioned in my first post, older kernel works, but only under Stretch. In Buster, reverting back 4.14.98 brakes Chromium (it doesn't start). Also, maybe it is still easier to hunt this bug down now, then in the future, when Stretch won't be an option at all.

I will consider an other topic under the right category, thank you for the suggestion.

P33M, 2019-07-20:
The issue is with the Linux kernel rather than videocore firmware, but there's no point in cross-posting, as the same people watch both repositories.

P33M, 2019-07-30:
I am now in possession of a device kindly provided by @bacsom.

Initial observations on latest 4.19 (and a Pi 4):
- Pressing any key causes event spam from the mouse reporting endpoint (event4) - which only stops if I hold the mouse still. I guess this is the gyro input. There are sensible x and y event deltas, but lots of scrollwheel spam.
- Pressing number and letter keys report on event0, the keyboard endpoint
- Pressing the special function keys like mail/"internet explorer" cause reports on event1 but not event0
- Pressing the power button causes events on the system control device (event2) - and unilaterally shuts down my Pi without the desktop prompting me 🙃
- I don't get events on event3, no matter what I do to it

bacsom, 2019-07-30:
You are welcome, I hope it will help to get to the bottom of this. :-) It definitely has to do something with the gyroscope because when I turn the mouse function off, it works correctly.

P33M, 2019-07-30:
As an immediate workaround, it should be possible to stop X.org from seeing or interpreting the bogus scrollwheel events. This would probably be better configured in userspace rather than as a kernel quirk.

As a separate line of investigation, figuring out why the kernel is reporting these events in the first place (in an apparent regression since 4.14). This is likely to take a long time.

bacsom, 2019-07-30:
If it is not too complicated, can you tell me how to do that? I am getting better at Linux, but this is beyond my knowledge.

P33M, 2019-07-30:
imwheel looks like a useful program:
https://manpages.ubuntu.com/manpages/bionic/man1/imwheel.1.html
It allows for arbitrary mouse event filtering/remapping in Xwindows.

bacsom, 2019-07-31:
Since yesterday I have tried different methods, unfortunately without much of a success. By remapping button 4 and 5 to button 9 **imwheel** stops scrolling inside of a window. BUT it also disables the OK button and scrolling still happens outside of the window (like when the cursor is above the taskbar). Using **xmodmap** (pointer = 1 2 3 0 0 6 7 8 9 10), scrolling still occurs inside of a window, but OK button works and outside of the window everything is fine.

P33M, 2019-07-31:
So on 4.9/4.14, the same endpoint is being polled for data, and the same data is being returned. It appears that Linux is incorrectly decoding the HID report as seen in evtest:
````

Event: time 1564580973.813432, -------------- SYN_REPORT ------------
Event: time 1564580973.829433, type 2 (EV_REL), code 0 (REL_X), value 67
Event: time 1564580973.829433, type 2 (EV_REL), code 1 (REL_Y), value 12
Event: time 1564580973.829433, type 2 (EV_REL), code 11 (?), value -124
Event: time 1564580973.829433, type 2 (EV_REL), code 12 (?), value 1
Event: time 1564580973.829433, type 2 (EV_REL), code 13 (?), value 45
Event: time 1564580973.829433, type 2 (EV_REL), code 14 (?), value -8
Event: time 1564580973.829433, type 2 (EV_REL), code 15 (?), value -84
```
In 4.19 these are decoded as multiple "EV_WHEEL" events.

Looking at the raw packet data, it looks like the "wheel" events are actually 16-bit values but Linux is fetching 8-bit values out of the responses - hence the huge offsets for some of these reports.

The "wheel" events sort of look like accelerometer data as
a) there are 3 nonzero 16-bit values in the 7 "wheel" reports - 3 axes, perhaps?
b) I can make larger signed 16-bit values appear by shaking the remote. 

For now, if we can't find a way of filtering out the event forwarding to Xwindows I'd recommend using 4.14 on Stretch as upstream bugs can take a while to fix.

P33M, 2019-07-31:
This works for me:
`xmodmap -e "pointer = 1 2 3 0 0 0 0 8 9 10"`

bacsom, 2019-07-31:
Thank you! Unfortunately, CLI and LibreOffice are also affected. Xmodmap definitely better than imwheel, but not a solution. :-(

rjsachse, 2019-07-31:
Also the media keys not working in Kodi either. Most of my life I have used windows and this seems like a driver issue but not 100% sure how Linux drivers works. Oh and I'm still using a rpi 3 and tried with stretch updated Kernal and new buster os, A lot of people seem to use the mx3 mouse with Kodi and surprised not many have complained about this issue yet.

bacsom, 2019-08-01:
Some of them [do](https://forum.libreelec.tv/thread/13994-problem-with-t3-airmouse-on-versions-8-90-006-and-later-including-8-95-001-rpi3/#post122679), but on a different platform, because most KODI users use LibreElec and not Raspbian. Also, not everybody hurries to upgrade their system to the latest available version.

P33M, 2019-08-01:
I've bisected the upstream kernel as it's an easy test to determine broken/not broken.
```
190d7f02ce8ef6774a69d3ec18c288c8a9601a4e is the first bad commit
commit 190d7f02ce8ef6774a69d3ec18c288c8a9601a4e
Author: Benjamin Tissoires <benjamin.tissoires@redhat.com>
Date:   Fri Dec 8 15:28:18 2017 +0100

    HID: input: do not increment usages when a duplicate is found

    This is something that bothered us from a long time. When hid-input
    doesn't know how to map a usage, it uses *_MISC. But there is something
    else which increments the usage if the evdev code is already used.

    This leads to few issues:
    - some devices may have their ABS_X mapped to ABS_Y if they export a bad
      set of usages (see the DragonRise joysticks IIRC -> fixed in a specific
      HID driver)
    - *_MISC + N might (will) conflict with other defined axes (my Logitech
      H800 exports some multitouch axes because of that)
    - this prevents to freely add some new evdev usages, because "hey, my
      headset will now report ABS_COFFEE, and it's not coffee capable".

    So let's try to kill this nonsense, and hope we won't break too many
    devices.

    I my headset case, the ABS_MISC axes are created because of some
    proprietary usages, so we might not break that many devices.

    For backward compatibility, a quirk HID_QUIRK_INCREMENT_USAGE_ON_DUPLICATE
    is created and can be applied to any device that needs this behavior.

    Signed-off-by: Benjamin Tissoires <benjamin.tissoires@redhat.com>
    Acked-by: Peter Hutterer <peter.hutterer@who-t.net>
    Acked-by: Dmitry Torokhov <dmitry.torokhov@gmail.com>
    Signed-off-by: Jiri Kosina <jkosina@suse.cz>
```

In theory, the quirk should fix it...

P33M, 2019-08-01:
@bascom can you try adding `usbhid.quirks=0x25a7:0x2402:0x100000` to /boot/cmdline.txt and rebooting?

bacsom, 2019-08-01:
Yes, doesn't work. :-(

P33M, 2019-08-01:
Hm. Chrome was working for me after I applied it - no bogus scrolling and the event stream reported in evtest was the same as on 4.14.

What's the output of `lsusb` and `cat /proc/cmdline`?

bacsom, 2019-08-01:
**lsusb:**
Bus 001 Device 004: ID 25a7:2402
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**cat /proc/cmdline**
coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight=768 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=fe87906e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

bacsom, 2019-08-01:
It is a regular Pi3.

P33M, 2019-08-01:
The parameter isn't present in the current command line - have you rebooted the Pi since modifying cmdline.txt?

bacsom, 2019-08-01:
Yes, I took the SD card out from the pi, I added the line in notepad (windows) and I put it back and rebooted.

end of config.txt:
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=320
usbhid.quirks=0x25a7:0x2402:0x100000

P33M, 2019-08-01:
It needs to go in /boot/cmdline.txt, not /boot/config.txt.

rjsachse, 2019-08-01:
What does the flag 0x100000 mean? I understand the other numbers are vendor and product id

bacsom, 2019-08-01:
Oooohh, sorry. :-( YESSS it seems to work!

P33M, 2019-08-01:
@rjsachse The flag corresponds to the quirk HID_QUIRK_INCREMENT_USAGE_ON_DUPLICATE which effectively masks the bogus multiple mousewheel events. An offset is applied to each one in turn which means all events other than the first one don't get mapped to a mousewheel event in Linux's input subsystem.

It's a workaround and not a fix - the device has broken HID descriptors for the accelerometer data.

bacsom, 2019-08-01:
So it means, that the remote's problem is in the hardware and there won't be a permanent fix? :-( I have been trying to get in touch with the manufacturer, but the final message I have got from the Chinese seller was: "I have reflected, but no useless.".

rjsachse, 2019-08-01:
Thank you much. I had to change the vendor and product IDs to match but seems to be a lot better. Thank you again for a work around

P33M, 2019-08-01:
I'm going to close the issue as there's no other immediate changes we can make. The commandline quirks parameter should eventually be baked into the big list of devices in hid-quirks.c, but that requires upstream patch submission (which takes months).

closed 2019-08-01 by P33M

P33M, 2019-08-01:
@rjsachse can you share the output of `sudo lsusb -v` for your device please?

bacsom, 2019-08-01:
Dear P33M,
Thank you very much for your work, I really appreciate it!

rjsachse, 2019-08-01:
#### This is the output for the mx3 mouse that looks like this
[image]

#### The lsusb output
```
Bus 001 Device 006: ID 1d57:ad03 Xenta
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1d57 Xenta
  idProduct          0xad03
  bcdDevice            2.00
  iManufacturer           1 FREEWAY TECHNOLOGY
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          159
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              0
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           38
        bInCollection           1
        baInterfaceNr( 0)       1
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 5
        bSourceID               4
        bControlSize            1
        bmaControls( 0)      0x03
          Mute Control
          Volume Control
        bmaControls( 1)      0x00
        iFeature                0
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             2
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bSourceID               5
[... 252 lines cut ...]
    Logical Maximum (1),
    Report Size (1),
    Report Count (16),
    Unit Exponent (0),
    Unit,
    Input (Variable),
End Collection,
Usage Page (FF00h),                 ; FF00h, vendor-defined
Usage (0Eh),
Collection (Application),
    Report ID (186),
    Report Count (31),
    Report Size (8),
    Logical Maximum (255),
    Logical Minimum (0),
    Usage (01h),
    Output (Variable),
    Report ID (186),
    Report Count (31),
    Report Size (8),
    Logical Maximum (255),
    Logical Minimum (0),
    Usage (01h),
    Input (Variable)
```

rjsachse, 2019-08-01:
Also note I had to change /dev/input/event1 group from root to input and then add rw to it as well so Kodi could pick up the media keys.

Seems weird that the group and permissions changed only on event1

P33M, 2019-08-02:
Oddly, the MX3 does specify gyro/accelerometer axes in its report specifiers - but also has the nonsensical 14-dimensional mousewheel report.

rjsachse, 2019-08-02:
I have to take the last comment back, the issue was a old udev rule I had to use to stop Kodi complaining about the remote moving that was on event 1 now it seems to be on event 3 and with the quirk fix everything is working like a charm and better than it has ever been. So this issue has always been there but only Kodi reported it in its logs, so I had to disable the mouse in Kodi so the logs wouldn't be full of errors and large file size. 

Even osmc and libreelec had issues with this remote and created udev rules which might have to be removed now.

Thank you so much again.