raspberrypi/firmware #1190: RTC PCF2129AT support for Raspberry PI 3/4

sparvu, 2019-07-15:
Do we support this RTC chip: PCF2129AT ? I have tried to use i2c-rtc.dtbo but no luck. Not sure if I have enabled it correctly under FreeBSD 12.0 RBPI3B+ board.

pelwell, 2019-07-15:
Almost - it shares a driver with the PCF2127, which we do support, but it needs a different "compatible string" so the driver can tell the difference. I've added the necessary options to the i2c-rtc and i2c-rtc-gpio overlays to the kernel source tree - the update will eventually reach the rpi-update and Raspbian firmware packages, but until then you can download a trial version here: https://drive.google.com/open?id=0B_P-i4u-SLBXN240NEVJUmZXak0

sparvu, 2019-07-15:
Cool. Thanks a lot. Let me try latest version on FreeBSD 12. Will report back the findings.

sparvu, 2019-07-15:
Nice. I have the clock running now on FreeBSD 12. Here the details:

 * I copied latest i2c-rtc.dtbo /boot/msdos/overlays/ 

 * I enabled the driver under config.txt (dtoverlay=i2c-rtc,pcf2129)

 * of course nxprtc driver must be loaded via loader.conf (nxprtc_load="YES")

After reboot I see these:

```
root@k1:~ # dmesg | grep nxp
nxprtc0: <NXP PCF2129 RTC> at addr 0xa2 on iicbus0
nxprtc0: WARNING: RTC battery is low
nxprtc0: registered as a time-of-day clock, resolution 0.500000s
```

and I can read and write from the nxprtc0 using sysctl

```
nxprtc0: read  at 2019-07-15 12:00:40.696296146: 0080-19-01 15:12:00.500000000
nxprtc0: write at 2019-07-15 12:01:12.496294621: 2019-07-15 12:01:12.000000000
```

So now I need to change my battery. A bit strange these are brand new Varta CR1220 batteries, just bought them - but the driver reports them old :) might be.

sparvu, 2019-07-15:
but it does not work. if I shutdown my Raspberry PI 3B+ for 30minutes, after boot the time is not accurately kept and NTP is down due the large time difference for sycronization. 

Something does not work. For example this message shows some problems. Not sure why:

```
nxprtc0: registered as a time-of-day clock, resolution 0.500000s
Warning: bad time from time-of-day clock, system time will not be set accurately
```

pelwell, 2019-07-15:
Have you taken the additional steps listed in this tutorial?: https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time

sparvu, 2019-07-15:
Sure. I have enabled the driver in config.txt but rest is applicable to Linux only. Im testing this on FreeBSD 12 (aarch64) where things are more straight forward. No need to do all those. 

One thing I must still check is the battery on the clock: I always see this message from driver:

```
nxprtc0: WARNING: RTC battery is low
```

I wonder if this might be the problem such my RTC cannot save and keep the clock running. The battery !?

pelwell, 2019-07-15:
If you have a spare SD card and a spare half hour you could try downloading a Raspbian image, patching the overlay and following the guide, just to be sure that it is a hardware issue rather than some subtle software configuration problem.

sparvu, 2019-07-15:
aha. good point. I could try that. thanks for tip.

popcornmix, 2019-07-23:
Latest rpi-update kernel includes this.