raspberrypi/firmware #1191: Get ARM memory returns 948M on a RPI4 4G board

agherzan, 2019-07-16:
I don't see any changes documented in https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-arm-memory as per RPI4 release but it seems like the mailbox interfaces returns 948M even on a 4G device.

timg236, 2019-07-16:
VideoCore can only directly address a maximum 1GB of system memory so this just reflects the VPU view of the world. The actual memory configuration is passed via device-tree to the kernel.

agherzan, 2019-07-16:
I have an issue where the memory configuration ends up being 1G on a 4G rpi4 only if I chain in uboot. So if I boot in uboot, the passed dtb shows the following memory configuration:

```
U-Boot> fdt print /memory 
memory {
        device_type = "memory";
        reg = <0x00000000 0x00000000 0x3b400000 0x00000000 0x40000000 0xbc000000>;
};
```
This is passed on to kernel:
```
root@raspberrypi4-64:~# hexdump  /sys/firmware/devicetree/base/memory/reg 
0000000 0000 0000 0000 0000 403b 0000          
000000c
```

**If I don't use a uboot**, the kernel receives a different memory configuration:
```
root@raspberrypi4-64:~# hexdump  /sys/firmware/devicetree/base/memory/reg 
0000000 0000 0000 0000 0000 403b 0000 0000 0000
0000010 0040 0000 00bc 0000                    
0000018
```

Any idea what is going on here?

pelwell, 2019-07-16:
If U-boot is getting the correct `memory/reg` setting from the firmware then it should try harder not to mangle it - it's out of our control, but I suspect it is either making an assumption about the amount to copy or it is converting to a memory size and back to a `memory/reg` value.

agherzan, 2019-07-16:
Very good point:

```
-               reg = <0x00 0x00 0x3b400000>;
+               reg = <0x00 0x00 0x3b400000 0x00 0x40000000 0xbc000000>;
```

closed 2019-07-16 by agherzan