raspberrypi/firmware #1192: Raspberry Pi4 isn't able to overclock gpu_freq over 600MHz

xlazom00, 2019-07-17:
Raspberry Pi4 isn't able to overclock gpu_freq over 600MHz
is this bug ?
when I change gpu_freq over 600 

I get freq like this
vcgencmd measure_clock arm
vcgencmd measure_clock core
vcgencmd measure_clock h264
vcgencmd measure_clock v3d

pelwell, 2019-07-17:
[ Your results have got lost from your post. ]

Clocking is more complex on BCM2838, and we're still working on this area of the firmware so this limitation may change in future, but we really don't recommend overclocking the Pi 4.

xlazom00, 2019-07-24:
@pelwell 
I also have this wired results from benchmark
When I switch arm_freq=1800
I get worst(half values) results then with arm_freq=1750

bechmark do x264 video encoding on all cores with 400% load

cpu temperature are fine in both cases
Any idea ?
is it possible that I have same problem like with gpu_freq ?

pelwell, 2019-07-24:
That does sound odd - I would have suggested checking the temperatures and measuring the clocks to see what frequency they were really at, but you've done both already.

You could also run `vcgencmd get_throttled` to be certain that you haven't been throttled for some reason during the benchmarks. The bcmstat script is useful for keeping an eye on clocks and temperatures:
```
$ wget https://raw.githubusercontent.com/MilhouseVH/bcmstat/master/bcmstat.sh
$ ./bcmstat.sh
```

timg236, 2019-07-24:
The launch firmware has a fixed VCO for the ARM of 3.5 GHz and uses an integer only divider so the maximum ARM frequency will be 1750 MHz. Selecting more than this will actually cause a lower ARM freq. Try "vcgencmd measure_clock arm" or "vcgencmd measure_clock pllb"
A recent rpi-update image increased the VCO to 4 GHz allowing arm frequencies of up to 2 GHz if you are willing to try the bleeding edge software (minimal test)

You may find that it gets throttled or requires over_voltage, we don't support or test arm_freq > 1500 MHz

xlazom00, 2019-07-25:
@timg236 btw are you able to overclock Pi4 to 2GHz ?
as my board don't boot on that freq.
I am on latest firmware now :)

timg236, 2019-07-25:
I believe other people have but it may depend on power supply / over_voltage etc. Does 1800 MHz work now ?

xlazom00, 2019-07-25:
@timg236 I am able to go to 1,85GHz(over_voltage=8) with latest firmware 
but it will be nice to test how far we can go with this 
https://www.cnx-software.com/2019/07/25/review-raspberry-pi-4-ice-tower-cooling-fan/

btw I am benchmarking with x264 video encoding and raspbian(debian 10) has like 1/2 values over ubuntu 18.04 64bit.
I think it is time to move on to 64bit. More registers are simply better :)

JamesH65, 2019-07-25:
We only test at 1500, so anything more than that is a bonus. The maximum you get overclocking will depend on the silicon itself, and will vary between batches.

xlazom00, 2019-07-25:
@JamesH65 
base on this(arm provided informations)
https://techreport.com/review/28189/inside-arms-cortex-a72-microarchitecture/
it should be possible to go to 2GHz  on 28nm

pelwell, 2019-07-25:
But the actual value depends on lots of different factors, including chip-to-chip variations.

MilhouseVH, 2019-07-25:
I have an RPi4/4GB that is (apparently) stable at `arm_freq=1900`/`over_voltage=3`, (and even 2GHz), under DVFS (`force_turbo=0`), but 1900MHz won't boot _at all_ if `force_turbo=1` is enabled - there's not even anything output to the serial console and I have `BOOT_UART=1` enabled via `sed`, or any LED patterns (solid red only). The highest frequency I can successfully boot with `force_turbo=1` enabled is 1870 (regardless of `over_voltage`).

Other than the obvious (silicon lottery), could there be something about `force_turbo=1` which wedges the startup sequence when using (apparently stable under DVFS) higher frequencies?

Obviously the startup sequence could be stressing a marginal frequency more than at any other time but under DVFS 1900/+3 really does appear to be rock solid, so I'm just wondering if there's the possibility of `force_turbo=1` introducing some other issue during startup for an otherwise stable higher frequency (maybe an unintended rounding issue or clock divisor issue?)

Bleep42, 2019-07-25:
Hi, Just to add my sixpence worth. :-) I've built a Pi4 laptop type computer [https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=244515](url) which I am very lucky to be able to stabley over clock to 1950/+3. I force it to work hard by using VLC to decode a HEVC video, picture:-
[image] 
This is using arm_freq=1950, core_freq=650, gpu_freq=650 (which I realise are not supported) with no Force_Turbo, notice the v3d frequency (750MHz) and the fact that all other frequencies are relatively low. If I set force_turbo=1, I get exactly the same frequencies, except the v3d, which is now 375MHz, picture:-
[image]
If I reduce core and gpu overclock to 600 then, with force_turbo=1 I get, picture:-
[image]
and without turbo, I get, the exact same results, when working hard, which I would sort of expect.
I suppose my query is why I seem to be getting such strange frequencies, if I use anything above 600 for gpu/core, realising that above 600 isn't suported. 
Interestingly, I am using the 650 gpu/core setting with no turbo, as it gives me slightly better results with the decoding of the HEVC in VLC, (presumably because of the v3d going to 750MHz) Yes I do have Kodi installed as well, which just about decodes any video stream I throw at it, using only around 10% CPU, thanks to all the work Popcornmix and others have done. :-)
Regards, Kevin.

xlazom00, 2019-07-25:
@timg236 I am on 2GHz now, without problem(x264 encoder running for 10minutes) with heatsink. Is it possible tune code a little so we can go to more then 2GHz? :)

xlazom00, 2019-07-26:
btw initial_turbo=60(delay force turbo) don't work

QBANIN, 2019-07-29:
When I try to overclock over 1750Mhz (2Ghz in this case), I'm getting this error in dmesg 
```

6.677324] bcm2835-clk fe101000.cprman: pllb: couldn't lock PLL
[    6.677350] ------------[ cut here ]------------
[    6.677383] WARNING: CPU: 1 PID: 64 at drivers/clk/clk.c:888 clk_core_enable+0xb4/0x264
[    6.677407] Enabling unprepared pllb
[    6.677423] Modules linked in:
[    6.677451] CPU: 1 PID: 64 Comm: kworker/1:3 Not tainted 4.19.60-v7l-qba+ #4
[    6.677472] Hardware name: BCM2835
[    6.677496] Workqueue: events deferred_probe_work_func
[    6.677541] [<c0212bf8>] (unwind_backtrace) from [<c020d24c>] (show_stack+0x20/0x24)
[    6.677575] [<c020d24c>] (show_stack) from [<c098aed0>] (dump_stack+0xcc/0x110)
[    6.677609] [<c098aed0>] (dump_stack) from [<c022224c>] (__warn.part.3+0xcc/0xe8)
[    6.677641] [<c022224c>] (__warn.part.3) from [<c02222e0>] (warn_slowpath_fmt+0x78/0x94)
[    6.677673] [<c02222e0>] (warn_slowpath_fmt) from [<c068b2a4>] (clk_core_enable+0xb4/0x264)
[    6.677706] [<c068b2a4>] (clk_core_enable) from [<c068e2f4>] (clk_register+0x578/0x6f4)
[    6.677737] [<c068e2f4>] (clk_register) from [<c068e4e0>] (devm_clk_hw_register+0x4c/0x98)
[    6.677769] [<c068e4e0>] (devm_clk_hw_register) from [<c06938c4>] (bcm2835_register_pll+0xb8/0xec)
[    6.677798] [<c06938c4>] (bcm2835_register_pll) from [<c0693ab0>] (bcm2835_clk_probe+0x1b8/0x280)
[    6.677830] [<c0693ab0>] (bcm2835_clk_probe) from [<c06ef260>] (platform_drv_probe+0x58/0xa8)
[    6.677864] [<c06ef260>] (platform_drv_probe) from [<c06ed420>] (really_probe+0x20c/0x2c8)
[    6.677896] [<c06ed420>] (really_probe) from [<c06ed6b0>] (driver_probe_device+0x70/0x188)
[    6.677927] [<c06ed6b0>] (driver_probe_device) from [<c06ed978>] (__device_attach_driver+0xc0/0xe4)
[    6.677959] [<c06ed978>] (__device_attach_driver) from [<c06eb450>] (bus_for_each_drv+0x90/0xd4)
[    6.677990] [<c06eb450>] (bus_for_each_drv) from [<c06ed18c>] (__device_attach+0xe0/0x148)
[    6.678022] [<c06ed18c>] (__device_attach) from [<c06ed9f8>] (device_initial_probe+0x1c/0x20)
[    6.678053] [<c06ed9f8>] (device_initial_probe) from [<c06ec474>] (bus_probe_device+0x94/0x9c)
[    6.678084] [<c06ec474>] (bus_probe_device) from [<c06ec964>] (deferred_probe_work_func+0x70/0x9c)
[    6.678117] [<c06ec964>] (deferred_probe_work_func) from [<c023d430>] (process_one_work+0x23c/0x518)
[    6.678150] [<c023d430>] (process_one_work) from [<c023e764>] (worker_thread+0x2d0/0x5b8)
[    6.678179] [<c023e764>] (worker_thread) from [<c0243f2c>] (kthread+0x16c/0x174)
[    6.678208] [<c0243f2c>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[    6.678231] Exception stack(0xef293fb0 to 0xef293ff8)
[    6.678252] 3fa0:                                     00000000 00000000 00000000 00000000
[    6.678279] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    6.678304] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    6.678326] ---[ end trace 01737195106bb1d6 ]---
[    6.778987] bcm2835-clk fe101000.cprman: pllb: couldn't lock PLL
```

**What does it mean? I have up-to-date rpi-update installed.**

```
vcgencmd measure_clock pllb
frequency(100)=4000000000
```

ARM freq is oscillating around this value +/- few Hz
```
vcgencmd measure_clock arm
frequency(48)=1927810560
```

Idle CPU temp is ~50 C and around 70C when running 4 cores cpu stress...

pelwell, 2019-07-29:
That looks like a custom build - is it using bcm2711_defconfig or have you tinkered with it?

QBANIN, 2019-07-29:
> That looks like a custom build - is it using bcm2711_defconfig or have you tinkered with it?

Yes. It's a custom kernel build. What should I change in kernel config to make this error go away ?

xlazom00, 2019-07-29:
@QBANIN
I didn't have any problem with with kernel in raspbian (debian 10) + latest firmware from rpi-update
or
64bit kernel from 
https://www.raspberrypi.org/forums/viewtopic.php?p=1506858#p1506858

pelwell, 2019-07-29:
Try building for bcm2711_defconfig, confirm it works (or not), then make changes you need until it breaks.

QBANIN, 2019-07-29:
I did build it for bcm2711_defconfig + few minor changes and uksm patch. It boots and works flawlessly as a headless NAS, just wondering where does this error message come from and what does it mean. It appears only when overclock more than 1750 MHz.

pelwell, 2019-07-29:
Ah timg236's earlier comment has the answer:
> The launch firmware has a fixed VCO for the ARM of 3.5 GHz and uses an integer only divider so the maximum ARM frequency will be 1750 MHz. Selecting more than this will actually cause a lower ARM [...]

QBANIN, 2019-07-29:
> Ah timg236's earlier comment has the answer:

So can I simply ignore it?

JamesH65, 2019-07-29:
If you want to get above 1750, you need to use the firmware from the latest rpi-update.

QBANIN, 2019-07-29:
> If you want to get above 1750, you need to use the firmware from the latest rpi-update.

I did "sudo rpi-update", compiled and reinstalled my custom kernel from up-do-date kernel-git following this instructions https://www.raspberrypi.org/documentation/linux/kernel/building.md 

Are the any more steps to perform?

pelwell, 2019-07-29:
We think not.

xlazom00, 2019-08-09:
@pelwell @timg236 
Are there any change that we will be able to go to more then 2GHz any time soon?

pelwell, 2019-08-09:
No - pushing the envelope even further for the very small number of people who
a) have one of the lucky Pis and
b) think this is important
is well down the list of priorities.

xlazom00, 2019-08-09:
@pelwell And can I change it or it is code in blobs ?

pelwell, 2019-08-09:
It's (necessarily) complex code in closed-source firmware.

l4jos, 2019-09-09:
This is probably made difficult to do because you don't want a burned device after a overclock mistake.

Actually the default RPI case causes high temperatures without even overclocking it. I have speak with te RPI store in Netherlands acout the temperature issue.

I would never recommend overclocking, especially not a RPI(4) which already has temperature issues. This is a bit offtopic, but good news for customers who are dealing with the same thing.

I have speak with the Raspberry store;
Thank you for your email. The Raspberry Pi Foundation (and their suppliers) are working hard to make a resolution for this. It will be delivered in the form of new firmware. There is already a preview for this firmware, but it is still a small portion of the improvements which are planned. Please be patient while looking forward to the new firmware...

About the lifetime: I have discussed this with the designer of the Raspberry Pi 4. The heat of the Raspberry Pi could be even higher, even about 120 degrees without breaking anything. But to be on the safe side, the max temperature before the throttling, is set to 85 degrees. So this is lower than the Raspberry Pi could handle, and the designer himself states the lifetime of the Raspberry Pi will not be affected if it keeps running at 85 degrees for extended periods.

- So, there is no firmware yet which lowers the temperature. -
I am looking forward to this.

xlazom00, 2019-09-09:
@l4jos look at all intel base mini ITX boards with on board intel SOCs. There is huge heatsink that cover SOC. So I would like to test Pi4 soc with huge heat sink. I am using one from copper and I can't get over 40 degree on 2GHz. 
So it will be nice to play with frequency.

l4jos, 2019-09-09:
Nowadays you can easily overclock the Pi with `raspi-conf` (to ondemand 1GHz at maximum, your pi may fail to boot afterwards).

Can you verify output of:
`sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq`
`sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq`
`sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq`

pelwell, 2019-09-09:
@l4jos, @xlazom00 Please don't start a long thread about the pros and cons of overclocking, what is safe and/or possible on a Pi 4, on this thread.

l4jos, 2019-09-09:
Can you try some troubleshooting steps I provided in my second post. I am trying to help you. Thanks.

l4jos, 2019-09-09:
A specific syntax needs some explanation. I am trying to help with a solid answer in my last post.
Everything except `raspi-conf`for changing frequency in is now deprecated.
Read this: [https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md](url)

l4jos, 2019-09-09:
Changing arm_freq_min=900 in config.txt - it increases ondemand idle speed from 600MHz to 900MHz. There is no reason to change this really.

Which kind of goal do you want to achieve?

JamesH65, 2020-04-16:
Closing this issue as questions answered/issue resolved. Use latest firmware.

closed 2020-04-16 by pelwell