raspberrypi/firmware #1194: Unable to handle kernel paging request at virtual address - Whilst using camera

anfractuosity, 2019-07-17:
Hi,

I'm just wondering if anyone might be able to give me any advice regarding the following issue:

It seems to happen when I'm using the Pi Camera.

I'm using the Pi Noir V2 camera, on the Pi Zero W

Any advice would be most appreciated!

```
[  126.816664] Unable to handle kernel paging request at virtual address 00001fba
[  126.821198] pgd = 9088ac36
[  126.826274] [00001fba] *pgd=15571831, *pte=00000000, *ppte=00000000
[  126.831387] Internal error: Oops: 17 [#1] ARM
[  126.836478] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc brcmfmac brcmutil sha256_generic raspberrypi_hwmon hwmon cfg80211 snd_bcm2835(C) snd_pcm rfkill snd_timer snd bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[  126.852022] CPU: 0 PID: 714 Comm: VCHIQ completio Tainted: G         C        4.19.58+ #2
[  126.859619] Hardware name: BCM2835
[  126.866763] PC is at vchiq_bulk_transfer+0x41c/0x548
[  126.873865] LR is at vchiq_bulk_transfer+0x418/0x548
[  126.877773] pc : [<c0555b58>]    lr : [<c0555b54>]    psr: 60000093
[  126.884012] sp : d19d3db0  ip : 00000000  fp : d19d3e24
[  126.890718] r10: 00000006  r9 : 00000072  r8 : c0afd144
[  126.895456] r7 : c0a3aa5c  r6 : 00000000  r5 : d19c5c00  r4 : 00000000
[  126.898754] r3 : 00001fae  r2 : 00000000  r1 : d19c5cdc  r0 : d19c5d88
[  126.901632] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  126.904904] Control: 00c5387d  Table: 157cc008  DAC: 00000055
[  126.911667] Process VCHIQ completio (pid: 714, stack limit = 0xe1598d61)
[  126.918795] Stack: (0xd19d3db0 to 0xd19d4000)
[  126.923954] 3da0:                                     d19d3df0 00000008 00000007 d19d3dc8
[  126.936846] 3dc0: c06a4064 ffffffff d3776818 d3776818 c09d18f8 00000000 c0afd0bc d19c5d88
[  126.943070] 3de0: d19c5cdc d19c5d10 c00cd67c c00ce51c 5751a000 00008090 d3776000 c014c406
[  126.948316] 3e00: c0a3aa68 d3776000 b4f35cc4 d19c5c00 00000001 d7580194 d19d3eec d19d3e28
[  126.953597] 3e20: c055b048 c0555748 b2e076e0 00000000 00000001 00000001 d19d3e64 d19d3e48
[  126.964135] 3e40: c004a258 0000012c b4f35cc4 b4f35d70 c014c407 d7580194 d3776824 d37765f0
[  126.976377] 3e60: c008580c 00000007 ffffffff d19dd864 b2e076e0 00000000 d18f4680 000009e8
[  126.989023] 3e80: 0001300f 012422a0 00008090 b2e076e0 00000000 00f299e8 d19d3f54 d19d3ea8
[  126.997453] 3ea0: c0087484 c00856fc c00cd67c c00ce51c 00000000 ffffffff 60000113 60000113
[  127.004495] 3ec0: c09d18f8 d19ba0a1 d19ba0a0 d19ba0a0 b4f35cc4 b4f35cc4 d689dae0 00000000
[  127.028883] Unable to handle kernel paging request at virtual address da8fc024
[  127.028889] pgd = 9088ac36
[  127.028893] [da8fc024] *pgd=00000000
[  127.028900] Internal error: Oops: 5 [#2] ARM
[  127.028902] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc brcmfmac brcmutil sha256_generic raspberrypi_hwmon hwmon cfg80211 snd_bcm2835(C) snd_pcm rfkill snd_timer snd bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[  127.028963] CPU: 0 PID: 714 Comm: VCHIQ completio Tainted: G         C        4.19.58+ #2
[  127.028966] Hardware name: BCM2835
[  127.028968] PC is at cfb_imageblit+0x454/0x9e4
[  127.028970] LR is at 0xda8fc024
[  127.028973] pc : [<c03e01cc>]    lr : [<da8fc024>]    psr: 20000193
[  127.028976] sp : d19d3788  ip : d7309510  fp : d19d37dc
[  127.028979] r10: 00000018  r9 : 00000000  r8 : 00000020
[  127.028982] r7 : 00000000  r6 : 00000020  r5 : ffaaaaaa  r4 : ff000000
[  127.028985] r3 : da8fc024  r2 : da8fc024  r1 : 00000040  r0 : 00000010
[  127.028989] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  127.028991] Control: 00c5387d  Table: 157cc008  DAC: 00000055
[  127.028995] Process VCHIQ completio (pid: 714, stack limit = 0xe1598d61)
[  127.028997] Stack: (0xd19d3788 to 0xd19d4000)
[  127.029001] 3780:                   00002900 00000a40 00000008 00000040 00000000 d19d3850
[  127.029005] 37a0: da8fc024 da8fc024 d7309510 0000000f 00000a40 d7309518 00000000 d730afbe
[  127.029008] 37c0: 00000001 000000ff 00000000 d702a400 d19d37ec d19d37e0 c03e0bd0 c03dfd84
[  127.029012] 37e0: d19d38b4 d19d37f0 c03de884 c03e0bc4 00000000 00000008 c03e0bd0 d7241d80
[  127.029016] 3800: d19d3890 ffffffff d703e218 ffffffff 00000000 00000000 00000000 00000200
[  127.029020] 3820: d730afae 00000008 d19d387c d19d3838 c03deba4 00000008 d703e070 00000000
[  127.029023] 3840: 00000000 00000000 00000010 00000001 00000138 00000170 00000040 00000010
[  127.029027] 3860: 00000007 00000000 00000001 d7309510 d19d3904 d19d3880 c03de5d0 c03dea30
[  127.029031] 3880: c03de884 c03d8140 d19d38b4 d703e070 d702a400 d730afae 00000008 c03de618
[  127.029034] 38a0: 00000007 00000017 d19d38f4 d19d38b8 c03d8470 c03de624 00000017 00000027
[  127.029038] 38c0: 00000007 00000000 00000051 d730afc0 d730af1c 00000720 d730afae 00000027
[  127.029042] 38e0: 00000700 d730b004 d19d393c d19d38f8 c03da7f4 c03d8384 00000027 d19d3908
[  127.029046] 3900: 00000008 00000002 d702a400 00000017 00000000 d702a400 c09bbb94 00000001
[... 86 lines cut ...]
[  127.154491] pc : [<c0185134>]    lr : [<c06a4064>]    psr: 60000013
[  127.154495] sp : d365bf50  ip : d365bf38  fp : d365bf64
[  127.154497] r10: 00000000  r9 : b6e54e88  r8 : 00000180
[  127.154500] r7 : 00000003  r6 : 000ac848  r5 : b60fe0d4  r4 : d1a050a3
[  127.154504] r3 : 00000002  r2 : 00000000  r1 : ffffffff  r0 : 00000000
[  127.154508] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  127.154511] Control: 00c5387d  Table: 13e48008  DAC: 00000055
[  127.154515] Process rs:main Q:Reg (pid: 276, stack limit = 0xad537cdd)
[  127.154517] Stack: (0xd365bf50 to 0xd365c000)
[  127.154521] bf40:                                     310e5400 00000000 d365bf94 d365bf68
[  127.154525] bf60: c0168014 c01850f0 00000000 00000000 00000000 b60fe0d4 000ac848 00000003
[  127.154529] bf80: c00091a4 d365a000 d365bfa4 d365bf98 c01680d8 c0167ffc 00000000 d365bfa8
[  127.154533] bfa0: c0009000 c01680cc 00000000 b60fe0d4 0000000d b6e54e88 00000180 310e5400
[  127.154537] bfc0: 00000000 b60fe0d4 000ac848 00000003 b60fe238 00000000 b6e5235c b60fe170
[  127.154541] bfe0: 310e5400 b60fe0c8 b6e1aaac b6dd645c 60000010 0000000d 00000000 00000000
[  127.154545] [<c0185134>] (__fdget_pos) from [<c0168014>] (ksys_read+0x24/0xd0)
[  127.154548] [<c0168014>] (ksys_read) from [<c01680d8>] (sys_read+0x18/0x1c)
[  127.154552] [<c01680d8>] (sys_read) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[  127.154554] Exception stack(0xd365bfa8 to 0xd365bff0)
[  127.154558] bfa0:                   00000000 b60fe0d4 0000000d b6e54e88 00000180 310e5400
[  127.154562] bfc0: 00000000 b60fe0d4 000ac848 00000003 b60fe238 00000000 b6e5235c b60fe170
[  127.154566] bfe0: 310e5400 b60fe0c8 b6e1aaac b6dd645c
[  127.154569] Code: e2800028 e3844002 eb147bc5 e1a00004 (e89da818) 
[  127.154572] ---[ end trace f461b6c225fb817b ]---
```

pelwell, 2019-07-17:
Is this a self-built kernel? And if so, what changes have you made to the configuration?

anfractuosity, 2019-07-17:
Yes it is, however I believe I had a similar issue before, with the stock kernel.  The only changes I made were to enable PSTORE to try to store ramoops, I've listed the changes below:

```
CONFIG_PSTORE=y
CONFIG_PSTORE_DEFLATE_COMPRESS=y
CONFIG_PSTORE_COMPRESS=y
CONFIG_PSTORE_DEFLATE_COMPRESS_DEFAULT=y
CONFIG_PSTORE_COMPRESS_DEFAULT="deflate"
CONFIG_PSTORE_CONSOLE=y
CONFIG_PSTORE_RAM=y
```

6by9, 2019-07-17:
```bcm2708_fb_imageblit``` is part of the frame buffer code.

What application are you running to access the camera?

anfractuosity, 2019-07-17:
Interesting, I'm using the picamera library running a Python program I created, which is capturing video and storing to SD card when there's movement.  I can provide the source for that if that's of any help.

"bcm2708_fb_imageblit is part of the frame buffer code." -- does that mean my application is also writing to the framebuffer?

6by9, 2019-07-17:
> "bcm2708_fb_imageblit is part of the frame buffer code." -- does that mean my application is also writing to the framebuffer?

Yes, but closer examination of that stack says that it is the process of putting the kernel panic information on the screen that has hit a second paging failure.
```
[  127.029258] [<c03e01cc>] (cfb_imageblit) from [<c03e0bd0>] (bcm2708_fb_imageblit+0x18/0x1c)
[  127.029261] [<c03e0bd0>] (bcm2708_fb_imageblit) from [<c03de884>] (bit_putcs+0x26c/0x40c)
[  127.029265] [<c03de884>] (bit_putcs) from [<c03d8470>] (fbcon_putcs+0xf8/0x134)
[  127.029268] [<c03d8470>] (fbcon_putcs) from [<c03da7f4>] (fbcon_redraw+0x174/0x1dc)
[  127.029271] [<c03da7f4>] (fbcon_redraw) from [<c03dd11c>] (fbcon_scroll+0x134/0xd64)
[  127.029275] [<c03dd11c>] (fbcon_scroll) from [<c0416308>] (con_scroll+0x1d8/0x1e4)
[  127.029278] [<c0416308>] (con_scroll) from [<c04163a8>] (lf+0x94/0x9c)
[  127.029281] [<c04163a8>] (lf) from [<c0417230>] (vt_console_print+0x210/0x32c)
[  127.029285] [<c0417230>] (vt_console_print) from [<c00609f0>] (console_unlock+0x238/0x560)
[  127.029288] [<c00609f0>] (console_unlock) from [<c006244c>] (vprintk_emit+0x10c/0x3b8)
[  127.029292] [<c006244c>] (vprintk_emit) from [<c00628c0>] (vprintk_default+0x4c/0x7c)
[  127.029295] [<c00628c0>] (vprintk_default) from [<c0063310>] (vprintk_func+0x54/0xbc)
[  127.029299] [<c0063310>] (vprintk_func) from [<c0062c84>] (printk+0x30/0x38)
[  127.029302] [<c0062c84>] (printk) from [<c0014328>] (dump_mem+0x18c/0x1b4)
[  127.029305] [<c0014328>] (dump_mem) from [<c0014728>] (die+0x204/0x2b0)
[  127.029308] [<c0014728>] (die) from [<c001a188>] (__do_kernel_fault.part.0+0x64/0x84)
[  127.029312] [<c001a188>] (__do_kernel_fault.part.0) from [<c06a72a4>] (do_page_fault+0x228/0x3b8)
[  127.029316] [<c06a72a4>] (do_page_fault) from [<c0019fe4>] (do_DataAbort+0x44/0xd0)
[  127.029319] [<c0019fe4>] (do_DataAbort) from [<c0009914>] (__dabt_svc+0x54/0x80)
```

Earlier in the dump is 
```
[  126.866763] PC is at vchiq_bulk_transfer+0x41c/0x548
[  126.873865] LR is at vchiq_bulk_transfer+0x418/0x548
...
[  126.911667] Process VCHIQ completio (pid: 714, stack limit = 0xe1598d61)
```
So that is part of VCHIQ (the IPC mechanism between VideoCore and the ARM.

If you can provide the app then we have a better chance of understanding what is going on.

Others have also observed that the PiZero when driving the camera can end up with the CPU browning out. Adding ```over_voltage=4``` to /boot/config.txt has been reported to help.

anfractuosity, 2019-07-17:
That's interesting you mentioned about over_voltage, I had been using that before.  I turned it off recently as my Pi seemed to be rebooting a lot.

I've attached my program below:

```
#!/usr/bin/python3
import sys
import os
import numpy as np
import io
import random
import picamera
import picamera.array
from picamera import mmal 
import datetime
import ctypes as ct
from PIL import Image

VIDS = "vids"

WAIT = 30 # time to wait before / after motion

prior_image = None

motionv = False

class PiCamera2(picamera.PiCamera):
    AWB_MODES = {
        'off':           mmal.MMAL_PARAM_AWBMODE_OFF,
        'auto':          mmal.MMAL_PARAM_AWBMODE_AUTO,
        'sunlight':      mmal.MMAL_PARAM_AWBMODE_SUNLIGHT,
        'cloudy':        mmal.MMAL_PARAM_AWBMODE_CLOUDY,
        'shade':         mmal.MMAL_PARAM_AWBMODE_SHADE,
        'tungsten':      mmal.MMAL_PARAM_AWBMODE_TUNGSTEN,
        'fluorescent':   mmal.MMAL_PARAM_AWBMODE_FLUORESCENT,
        'incandescent':  mmal.MMAL_PARAM_AWBMODE_INCANDESCENT,
        'flash':         mmal.MMAL_PARAM_AWBMODE_FLASH,
        'horizon':       mmal.MMAL_PARAM_AWBMODE_HORIZON,
        'greyworld':     ct.c_uint32(10)
        }

class MotionAnalyser(picamera.array.PiMotionAnalysis):
    def analyse(self, a):
        global motionv

        # Calculate the motion vector polar lengths
        r = np.sqrt(
            np.square(a['x'].astype(np.float)) +
            np.square(a['y'].astype(np.float))
            ).clip(0, 255).astype(np.uint8)

        if np.count_nonzero(r) > 0 and np.mean(r[np.nonzero(r)]) > 3.0 and (r > 3.0).sum() > 100 :
            motionv = True
        else:
            motionv = False

def detect_motion(camera):
    global motionv
    print(motionv)
    return motionv

with PiCamera2() as camera:

    out = MotionAnalyser(camera) 
    camera.awb_mode = 'greyworld'
    camera.resolution = (1920, 1080)

    stream = picamera.PiCameraCircularIO(camera, seconds=10)
    camera.framerate = 30
    camera.start_recording(stream, splitter_port=1,format='h264')
    camera.start_recording('/dev/null', splitter_port=2, format='h264', motion_output=MotionAnalyser(camera)) 

    idv = 0

    try:
        while True:
            camera.wait_recording(1)

            if detect_motion(camera):

                print('Motion detected!')
                idv = datetime.datetime.now().timestamp()
                camera.split_recording(os.path.join(VIDS,'%d.h264' % idv),splitter_port=1)               
                stream.copy_to(os.path.join(VIDS,'%d.h264' % (idv-WAIT)), seconds=WAIT)
                stream.clear()

                while detect_motion(camera):
                    camera.wait_recording(WAIT,splitter_port=1)

                print('Motion stopped!')
                idv += 1

                camera.split_recording(stream,splitter_port=1)
    finally:
        camera.stop_recording()
```

anfractuosity, 2019-07-17:
I ordered an ESP32 to see if I can also log any kernel messages on reboot too, when overvolt is enabled (the pi is in a waterproof IP68 box, so I can't easily run through other cables through).  Will hopefully be able to install that tomorrow.  (The ramoops /sys/fs/pstore, seems to be empty after a reboot at the moment).

anfractuosity, 2019-07-20:
I think I found why my pi was rebooting with over_voltage, it seems to be overheating, but I would be curious as to why over_voltage is needed on the pi zero, but not on the other pis, is there a difference to how the pi zero is powered?

Esser50K, 2021-05-19:
Can confirm having the same issue with a stock kernel.