raspberrypi/firmware #1195: Problems while booting with enabled GL driver (fake kms)

ChrKuhn, 2019-07-17:
Hello,

I have downloaded the 2019-07-10-raspbian-buster-lite.img and can boot a Raspberry Pi 3 with it. Then I have selected the GL driver (fake kms) with raspi-config. After rebooting, the Raspberry Pi hangs after some seconds during next boot process, the touchdisplay is dark, and it is also not responding to a ping from a remote computer. If I use the image 2018-11-13-raspbian-stretch-lite.img instead, I can select this GL driver with raspi-config and I am able to reboot the Raspberry Pi. But after 
apt update && apt upgrade 
I get the same result, the Raspberry Pi hangs during boot and is not responding to pings from other computers. Is it possible that there are problems with the actual GL driver?
Regards,
Christian

JamesH65, 2019-07-18:
What display are you using?

ChrKuhn, 2019-07-20:
I'm using the 7 inch touch display with a resolution  800 x 480 pixels supported by Raspberry Pi Foundation (I have found this data sheet:  http://www.farnell.com/datasheets/1958036.pdf). It works, if I use an external monitor connected by HDMI cable.
Regards,
Christian

stefi01, 2019-07-22:
i am getting the same problem, hook up to pc monitor and it boots fine, try it with official touch screen and black screen and the green led stays on, used with pi 3+

dreamvictim, 2019-07-23:
I'm also getting the same problem

stefi01, 2019-07-23:
i managed to fix mine :) i had to disconect the touch screen then it booted ok so then i ssh and used the raspi-config command and disabled GL driver, then  touch screen worked ok

ChrKuhn, 2019-07-23:
Actually, I'm falling back to the legacy non-GL driver and can run the KDE desktop on the touch screen. But I think, that the GL driver is the precondition for accelerated applications with OpenCL (https://github.com/doe300/VC4CL) or Tensorflow.js, which is using WebGL.

JamesH65, 2019-07-29:
Just tried this on a 3B+, and it worked fine - will now try a 3B.

JamesH65, 2019-07-29:
OK, so the 3B+ works OK with the VC4 driver enabled, but not the B. Probably a device tree issue. Time to learn about device tree.

JamesH65, 2019-07-29:
OK, this is a duplicate of #1204, please use the information in that thread.

closed 2019-07-29 by JamesH65