raspberrypi/firmware #1196: 35e22a26cbfb1bbc71c15aae0500263122e4222e breaks WiFi networking on RPi3B

DougieLawson, 2019-07-18:
The latest 4.19.58-v7+ kernel and bootcode https://github.com/Hexxeh/rpi-firmware/commit/35e22a26cbfb1bbc71c15aae0500263122e4222e breaks networking on my RPi3B.

On boot-up networking is OK. wpa-supplicant works OK. All is good.

I've got a script that pings 8.8.8.8 every ten minutes to reconnect the WiFi if my router blips or gets rebooted. It starts by killing wpa_supplicant, then runs ifconfig to restart the wlan0 interface, then restarts wpa_supplicant. That's failing with `No such device`:

[WIFIC.sh.txt](https://github.com/raspberrypi/firmware/files/3405188/WIFIC.sh.txt)

```
Jul 18 06:29:07 hitchhiker WiFiCheck.sh[1471]: Could not read interface -Dnl80211,wext flags: No such device
Jul 18 06:29:07 hitchhiker WiFiCheck.sh[1471]: nl80211: Driver does not support authentication/association or connect commands
Jul 18 06:29:07 hitchhiker WiFiCheck.sh[1471]: nl80211: deinit ifname=-Dnl80211,wext disabled_11b_rates=0
Jul 18 06:29:07 hitchhiker WiFiCheck.sh[1471]: Could not read interface -Dnl80211,wext flags: No such device
Jul 18 06:29:07 hitchhiker WiFiCheck.sh[1471]: -Dnl80211,wext: Failed to initialize driver interface
```
The `wlan0` interface can't be found by wpa_supplicant.

kern.log has
```
Jul 16 03:07:08 hitchhiker kernel: [18031.598545] brcmfmac: brcmf_run_escan: error (-52)
Jul 16 03:07:08 hitchhiker kernel: [18031.598559] brcmfmac: brcmf_cfg80211_scan: scan error (-52)
Jul 16 03:12:08 hitchhiker kernel: [18331.602749] brcmfmac: brcmf_run_escan: error (-52)
Jul 16 03:12:08 hitchhiker kernel: [18331.602764] brcmfmac: brcmf_cfg80211_scan: scan error (-52)
```
But, I'm not sure that's related to the `wlan0` device disappearing.

I've reverted back to 4.19.57-v7+.

pelwell, 2019-07-18:
Have you tried winding the firmware back to the last-known-good point but retaining the new kernel, and vice-versa? You can either copy the firmware files manually or use `sudo SKIP_KERNEL=1 rpi-update`.

DougieLawson, 2019-07-29:
I've tested https://github.com/Hexxeh/rpi-firmware/commit/ce2a9f85a6fd88f8c42ef54b7bad99b42e76e403 and that's still breaking WiFi networking. So once again it's going back to https://github.com/Hexxeh/rpi-firmware/commit/f8060224e53e26cdaea5b9f6f010ac8c3692e2ea

Benzomat, 2019-10-04:
You might give the [updated Broadcom firmware](https://github.com/raspberrypi/linux/issues/2453#issuecomment-538493131) firmware a try.

mwoolweaver, 2019-10-05:
> @pelwell that one is back on [Hexxeh/rpi-firmware@f806022](https://github.com/Hexxeh/rpi-firmware/commit/f8060224e53e26cdaea5b9f6f010ac8c3692e2ea) and nothing bad has happened since it was booted @ 07:42. Took both the firmware and the kernel back.

could you explain to me how to do this?

pelwell, 2019-10-05:
Just run `sudo rpi-update f806022`.

rourke, 2019-12-17:
Still having this issue, even with the rpi update.

```
pi@dsmr:~ $ sudo rpi-update f806022
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
```

popcornmix, 2019-12-17:
@rourke what does ``vcgencmd version`` report?

rourke, 2019-12-17:
```
pi@dsmr:~ $ vcgencmd version
Jul  9 2019 14:41:21
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start_cd)
```

popcornmix, 2019-12-17:
You have the firmware associated with that rpi-update.
Although it's not clear if that (very old) firmware/kernel is desirable.

maheshjena1, 2019-12-23:
any update on how to fix this?
I am still getting these errors with latest firmware:

> Dec 21 02:47:52 raspberrypi kernel: [42256.723546] brcmfmac: brcmf_run_escan: error (-52)

Dec 21 02:47:52 raspberrypi kernel: [42256.723561] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

Dec 21 02:48:52 raspberrypi kernel: [42316.722337] brcmfmac: brcmf_run_escan: error (-52)

Dec 21 02:48:52 raspberrypi kernel: [42316.722352] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

Dec 21 02:49:52 raspberrypi kernel: [42376.733601] brcmfmac: brcmf_run_escan: error (-52)

Dec 21 02:49:52 raspberrypi kernel: [42376.733618] brcmfmac: brcmf_cfg80211_scan: scan error (-52)

> 

Also, is there any side effects of this error on bluetooth or UART driver?
In my experiment, I am observing some data loss in UART and in syslog, I could only this error.
Could it be possible this error and data loss over UART are related?

popcornmix, 2019-12-23:
@maheshjena1 can you identify the most recent rpi-update firmware that doesn't have the issue?