raspberrypi/firmware #1197: DPI24 - incorrect display on RPi4

aBUGSworstnightmare-rpi, 2019-07-18:
Using the same uSD card on RPi Zero/RPi3 and RP4 gives incorrect incorrect image on the RP4.

Posted some pictures in this threat https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=244266&p=1502883#p1502829

As Zero/RP3 are displaying the image correctly it seems to me that GPIO timing for RPi4 is different.

JamesH65, 2019-07-30:
Had a look at this, using a Adafruit Kippah and what I think is the same display, up to date firmware, and not seeing any issues that match this. Did get a bit of flickering but reseated the display cable and seems to have gone away. Anything else different about your setup?

JamesH65, 2019-07-30:
Have also tested with playing back video whilst cores running at 100%, still no issues.

kevinfishburne, 2019-10-12:
FYI I'm having a similar issue. I'm using a RPi4 with a HAT which requires dpi24 to output to a CRT TV (similar to the Gert VGA 666). The HAT works fine with the RPi3. It outputs nothing using the same config on the RPi4. The modified dpi24 overlay shared on Google Drive posted earlier in this thread did output video, but it appears to be missing the red channel. I was testing using the nightly Lakka build. I'm now testing using 2019-09-26-raspbian-buster, and it works like it did with Lakka using the modified dpi overlay (still missing the red channel):

[img]http://eightvirtues.com/misc/20191012_000546.jpg[/img]

I'm guessing the modified dpi24 overlay was added to Raspbian since it exhibited the same behavior as my manual replacement. These are the settings I appended to Raspbian's config.txt:

[code]# Generic 24-bit DPI display.
dtoverlay=dpi24
#gpio=0-27=a2,np

# Set default display.
enable_dpi_lcd=1
display_default_lcd=1

# Turn off conflicting peripheral overlays using GPIO pins.
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2s=off
dtparam=audio=off
enable_uart=0

# Enable custom dpi timings.
dpi_group=2
dpi_mode=87

# Set custom dpi timings.
#dpi_timings=2048 1 180 202 300 240 1 3 5 14 0 0 0 60 0 42954545 1 #240p - NTSC perfect clock based on SNES/NES timings
dpi_timings=320 1 23 25 39 240 1 4 3 15 0 0 0 60 0 6400000 1

# Disable overscan.
disable_overscan=1
[/code]

The video mode set by the dpi timings is 320x240. 2400x240 on Lakka exhibited the same behavior. I will update the bug report (https://github.com/raspberrypi/firmware/issues/1197) with this information. I have also contacted the creator of the HAT (RetroTINK), Mike Chi, with this information should he choose to collaborate in resolving it.

closed 2025-02-14 by popcornmix