raspberrypi/firmware #1198: New AWB algorithm breaks custom lens shading

jtc42, 2019-07-19:
This was briefly mentioned in [this comment](https://www.raspberrypi.org/forums/viewtopic.php?p=1477425&sid=533edc03a2ff3344a4052412f65be21e#p1477425), but the new AWB algorithm described [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1477425) breaks custom lens-shading tables in the Pi camera.

**Examples**

[This comment](https://www.raspberrypi.org/forums/viewtopic.php?p=1477425&sid=533edc03a2ff3344a4052412f65be21e#p1477425) describes `raspistill` hanging when built with a custom lens-shading table. 

More relevant to me personally, the [lens-shading implementation in rwb27/picamera](https://github.com/rwb27/picamera/tree/lens-shading) is broken in a similar way.

The module will work fine until you write any data to the lens-shading table. After that, all captures will hang until timeout.

This is immediately seen when, for example, running [this auto-calibration routine](https://gitlab.com/openflexure/openflexure-microscope-server/blob/master/openflexure_microscope/plugins/default/camera_calibration/recalibrate_utils.py#L127) for lens shading, using that picamera fork

**Workaround**

As a workaround, you can switch back to the old AWB version by running `sudo vcdbg set awb_mode 0`. [This comment](https://github.com/raspberrypi/firmware/issues/1167#issuecomment-511558716) describes creating a service to run the command on boot, which may be useful.

**Ideal outcomes**

My understanding is that the new AWB algorithm is part of the closed-source bit of the firmware? Any guidance on how to update custom lens-shading implementations to work with the new algorithm would be ideal.

In the meantime, do you have any information on if `vcdbg set awb_mode` will be sticking around, at least for the foreseeable future? Custom lens-shading is absolutely mission critical to the project I'm working with, so removal of this option would be extremely challenging for us. 

If we know it'll stick around as an option, it'd save us a huge headache.

Many thanks! Please let me know if I can provide any more information, or if any particular logs would be useful.

JamesH65, 2019-07-19:
@naushir For you I think!

6by9, 2019-07-19:
The main thing to say at this stage is the ```sudo vcdbg set awb_mode 0``` is likely to give substandard results on a Pi4. The old AWB algorithm uses parts of the 3D hardware for the stats analysis, and on Pi4 that is not an option (new hardware, and controlled by the ARM not VPU).

For Pi0-3 the option is likely to hang around for a fair while (many months, if not years, as it is effort to remove it), but we will be aiming to resolve the issue of the new AWB and custom lens shading within weeks.
Apologies that we hadn't picked up on the issue report on the forum earlier. Github issues are always a more reliable reporting mechanism if there is an obvious regression.

jtc42, 2019-07-19:
> The main thing to say at this stage is the `sudo vcdbg set awb_mode 0` is likely to give substandard results on a Pi4. The old AWB algorithm uses parts of the 3D hardware for the stats analysis, and [...]

Thanks for the information. Do you have specifics on "substandard results"? Basically everything we do involves taking an AWB once, then immediately turning AWB off, fixing the white balance, and then auto-calibrating lens-shading. 

If the AWB is just slow, we can absolutely deal with that. If it's likely to be WAY out then we may have to just recommend against using the Pi4 for the project until we can come up with a proper fix for our custom lens-shading.

Absolutely no worries about not catching this earlier. I feel it's a fairly niche use-case, and it's been a busy few weeks!

naushir, 2019-07-19:
This should be an easy fix.  Working on it now.

6by9, 2019-07-19:
> Thanks for the information. Do you have specifics on "substandard results"? Basically everything we do involves taking an AWB once, then immediately turning AWB off, fixing the white balance values, and then auto-calibrating lens-shading.

> If the AWB is just slow, we can absolutely deal with that. If it's likely to be WAY out, or totally broken, then we may have to just recommend against using the Pi4 for the project until we can come up with a proper fix for our custom lens-shading.

There are actually 3 different AWB algorithms of varying complexities in the old driver, and if they reported that they couldn't find a sane set of settings they'd drop back.
Only the main one has been tuned on the Pi, and that is the one that can not run on the Pi4. It'll therefore immediately drop back to a partially tuned far simpler algorithm, hence results can't be guaranteed. They'll work in many situations, but there will be some corner cases where they go wrong, and that is why we updated to the newer algorithm.

jtc42, 2019-07-19:
> There are actually 3 different AWB algorithms of varying complexities in the old driver, and if they reported that they couldn't find a sane set of settings they'd drop back.
> Only the main one has [...]

Okay, thanks for the info. For the time being, that should be fine for us.

> This should be an easy fix. Working on it now.

Amazing, thank you! Let me know if there's anything useful we could do. Even just testing or something.

naushir, 2019-07-19:
The changes should be going into the new firmware.  If you run a `rpi-update` early next week, this should hopefully be fixed.

jtc42, 2019-07-19:
Thats fantastic, thanks for the super quick fix.

popcornmix, 2019-07-23:
Latest rpi-update firmware includes this.

jtc42, 2019-07-24:
> Latest rpi-update firmware includes this.

Fantastic, thank you! I'll test this as soon as possible.

JamesH65, 2019-07-26:
@jtc42 Did you have time to test this? If its all OK, please close the issue.

jtc42, 2019-07-26:
> @jtc42 Did you have time to test this? If its all OK, please close the issue.

Hey sorry I'm away until Monday but will test it first thing and close the issue if all good. Apologies for the delay.

jtc42, 2019-07-29:
Updated firmware fixes this brilliantly. Thanks!

closed 2019-07-29 by jtc42