raspberrypi/firmware #1199: EFI map on Raspberry Pi 2 - Move 4KiB of reserved memory upper in RAM

ggardet, 2019-07-22:
According to `lsefimap` command, on RPi2, the 4 KiB of memory form `0x000` to `0xfff` are reserved (from raspberrypi-firmware I guess). 
Would it be possible to move this reserved memory upper in RAM to have 128MB of RAM usable starting from `0x0`? Currently, it causes problems to boot RPi2 with EFI enabled, using u-boot and grub2.

pelwell, 2019-07-22:
The first page of RAM is where the stubs - the first instruction the ARM cores run - are loaded to, and ATAGs if they are used (some OSs expect ATAGs at 0x100). Assuming you don't need ATAGs, the stubs (and the spin tables) are the only issue. There is actually nothing stopping you using U-Boot as a stub, provided it is able to perform some low-level setup such as enabling the caches. Alternatively, if you can cope with a slightly longer load time you could create a large stub with the main code moved after enough padding for your own loader to be loaded into. But the chances of us changing the initial load address of the stub are very small.

ggardet, 2019-07-22:
U-Boot and grub2 are working properly, but the problems arise when kernel is loaded as Linux kernel want to use the 1st 128MB of RAM to uncompress itself.

pelwell, 2019-07-22:
Without an intermediate loader, the kernel manages to avoid trampling on the stub code, so it is possible.

ggardet, 2019-07-22:
> Without an intermediate loader, the kernel manages to avoid trampling on the stub code, so it is possible.

Yes, but without EFI enablement, right?

ardbiesheuvel, 2019-08-14:
> Without an intermediate loader, the kernel manages to avoid trampling on the stub code, so it is possible.

If so, this is entirely by accident.

I would assume that U-Boot's PSCI implementation takes care of keeping the secondaries penned up, and it shouldn't need the ATAGs either, so I wonder if U-boot cannot just release this memory to the OS.

pelwell, 2019-08-14:
> If so, this is entirely by accident.

No - the load address is defined to be 0x8000 for 32-bit and 0x80000 for 64-bit, and the .dtb includes the following to mark that page as no-go:
```
/memreserve/ 0x00000000 0x00001000;
```

ardbiesheuvel, 2019-08-14:
That doesn't matter. The problem is the decompressor, which ignores ATAGs, DTs or EFI memory maps, and decompresses the kernel to the base of the 128 MB region it was loaded in. The fact that it happens to leave 32 KB free at the base of this window is what is saving you here, but if you increase the reserved region beyond that, the memreserve does not save you, and it will be clobbered during decompression.

afaerber, 2019-08-14:
> I would assume that U-Boot's PSCI implementation takes care of keeping the secondaries penned up, and it shouldn't need the ATAGs either, so I wonder if U-boot cannot just release this memory to the OS.

U-Boot does not implement PSCI to my knowledge; by default it does the EFI reservation for the firmware-stub-provided spin-table. For _rpi3_ (and with André's work soon rpi4) it's possible to use Trusted Firmware-A `armstub8.bin` in place of the default stub - this allows to switch from `spin-table` to `psci-1.0` in the DT, working okay, but it does not (yet) suppress U-Boot's then-unneeded memory reservation.

This issue is about 32-bit **rpi2** though - I'm not aware of a TF-A or any other PSCI implementation for it. Is PSCI even possible on Armv7 or Armv6 for lack of EL3?

ardbiesheuvel, 2019-08-15:
Hmm, good point. I was thinking about the arm64 port probably.

In any case, I will check whether we can disregard the first TEXT_OFFSET bytes in the EFI stub loader (which is usually set at 32 KB). This does mean that, if the value of TEXT_OFFSET is ever decreased, we may be back where we started, but given the state of the 32-bit ARM port, that is unlikely to happen.