raspberrypi/firmware #1200: Can't wake composite screen (stretch, buster)

giddyhup, 2019-07-22:
The following applies to the latest stretch and buster distributions. It happens on a Pi 2B and 3B.

A TV screen which is connected via analog port and that shows the expected image on boot does not come back with an image after programmatically putting it to sleep and waking it up again. The screen wakes up but shows no image.

Errors show up in the kernel log:
```
[ 1427.683462] raspberrypi-firmware soc:firmware: Request 0x00048003 returned status 0x80000001
[ 1427.683475] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[ 1427.683479] detected fb_set_par error, error code: -22
```

Setting the framebuffer color depth also results in an error (and the same kernel messages): 
`$ fbset -depth 16`
`ioctl FBIOPUT_VSCREENINFO: Invalid argument`

Command used for sleep:
`tvservice -o`

Wake-up commands (alternatives, all failing):
`tvservice -c "PAL 4:3"`
`tvservice -c "PAL 4:3"; sudo chvt 1; sudo chvt 7`
`tvservice -c "PAL 4:3"; fbset -depth 8; fbset -depth 32; xrefresh -d :0`

/boot/config.txt contains only `enable_tvout=1`, the behavior doesn't change when framebuffer width/height or overscan settings are being used.

`$ uname -a`
`Linux raspberrypi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux`

JamesH65, 2019-07-26:
Just been looking at this with the latest firmware, Pi3B+, single composite display. With vc4-fkms-v3d enabled, fbset doesn't work as before. This, I believe, is expected. The vc4 driver (DRM/KMS) emulates a framebuffer rather than actually being a framebuffer, so fbset doesn't have the same feature set as if it were a real framebuffer. 

If you disable vc4-fkms-v3d, then fbset works as expected, I was able to power off composite, power it back on and create a new FB with fbset. I did need to `fbset -depth 8; fbset -depth 32` as before.

Since the DRM code that emulates the framebuffer is all uptream code I don't think there is much we can do about it. If you don't need the vc4 driver on the older devices (<Pi4), then just turn it off.

giddyhup, 2019-07-27:
Thanks for the remark. I don't think it's enabled, I don't have `dtoverlay=vc4-kms-v3d` in /boot/config.txt.

JamesH65, 2019-07-27:
Odd, since fbset seems to work as expected on my 3B+ system with vc4-fkms-v3d disabled.

giddyhup, 2019-07-29:
Here's my firmware version:
```
Jul  9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
```
How can I demonstrate that vc4-kms-v3d is not enabled? What other information can I provide?

JamesH65, 2019-07-29:
This seems to work fine for me.  I did have some newer changes in firmware that may have affected this area. Not sure what version is currently in rpi-update, but worth trying it, as your version is quite old (comparatively speaking!)

giddyhup, 2019-07-30:
OK. I updated (Jul 25 2019 17:55:26, version a06e5a83b5d8ad586866073a18adaa21c26ef866) and it is working now. Thank you. I guess the issue can be closed once this version is in the regular update.