raspberrypi/firmware #1201: Overscan_Scale=1 not working on composite

dreamvictim, 2019-07-23:
RP3b+ using composite, picture no longer fits on screen. 

Overscan_Scale=1 doesn't appear to be working

popcornmix, 2019-07-23:
What are you running?
Just console, desktop or app that uses dispmanx overlays?

dreamvictim, 2019-07-23:
I'm using Desktop and Open Auto Pro. I've disabled the GL driver as that stops my touchscreen from working.

popcornmix, 2019-07-23:
Why are you using overscan_scale?
That shouldn't be required for console or desktop use.

What does ``vcgencmd version`` report?

dreamvictim, 2019-07-23:
It's the only setting that I've found that fits perfectly on my screen.

Version: ccada45e1764783c9876237218ef886d96b24fc7

popcornmix, 2019-07-23:
Can you post the contents of config.txt.
Does the problem just affect Open Auto Pro or also desktop?

dreamvictim, 2019-07-23:
It affects both. 

Overscan_scale=1 creates black borders when previously it fitted just over and as you can see in the config.txt I used overscan settings to bring the edges in slightly. If I disable overscan the screen doesn't fit at all. 

`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

#overscan fix
overscan_scale=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=10
overscan_right=-10
overscan_top=-20
overscan_bottom=-10

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=3

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=off
disable_splash=1

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=256`

peetereczek, 2019-07-23:
I can confirm issue, having the same case. Overscan is not working at all. What's more also frame buffer setting looks like it's not working.

Pi3B+ updated to Buster and using composite screen as an video output freezes the whole system just after desktop appears. I can move mouse around but nothing is working.

On HDMI output everything fine, no issues at all.

joolswills, 2019-07-23:
Just to note in regards to the use of overscan_scale retropie still ships with overscan_scale set (on Stretch) as it's needed for users that use overscan so Emulation Station (which uses sdl2) follows the overscan settings. To drop it we would need to add some calibration config like Kodi has. There are other apps in retropie that are affected also.

dreamvictim, 2019-07-23:
> I can confirm issue, having the same case. Overscan is not working at all. What's more also frame buffer setting looks like it's not working.
>
> Pi3B+ updated to Buster and using composite screen as [...]

I also had the freeze just after desktop. I have disabled the GL driver in raspi-config which stops it from freezing.

peetereczek, 2019-07-23:
It's not fixing overscan anyway?

dreamvictim, 2019-07-23:
> It's not fixing overscan anyway?

No unfortunately not but it stops it from freezing :)

JamesH65, 2019-07-24:
For the freezing, it might be worth trying an rpi-update, there have been fixes for composite. Usual caveats  - its beta firmware so use at your peril.

dreamvictim, 2019-07-24:
> For the freezing, it might be worth trying an rpi-update, there have been fixes for composite. Usual caveats - its beta firmware so use at your peril.

Thanks but I've already tried the rpi-update.

JamesH65, 2019-07-24:
@dreamvictim When? There has been a very recent update.

dreamvictim, 2019-07-24:
> @dreamvictim When? There has been a very recent update.

Yesterday :)

peetereczek, 2019-07-24:
> > @dreamvictim When? There has been a very recent update.
> 
> Yesterday :)

I did the same, after yesterday's rpi-update still system freezes

JamesH65, 2019-07-24:
Just trying the latest firmware, Buster, on Pi3B+, FMKS/DRM driver and I see no freeze. This was a fresh install/update - were the freezing issues seen on upgraded Stretch installs? Or some other setup? Running terminal, Chromium all seems fine.

As for overscan, there are some fixes going through code review at the moment that should help here. Probably take a few more days to get merged.

JamesH65, 2019-07-24:
Just tried overscan_scale on a current Buster image,Pi3B+, composite, and it definitely scaled down the display to match the overscan settings. What exactly is the problem being seen ie what happened before and what happens now? Note that the docs do not recommend using this options as it does slow down the system due to large amounts of GPU work required for the scaling, and the subsequent reduced image quality.

peetereczek, 2019-07-24:
> Just trying the latest firmware, Buster, on Pi3B+, FMKS/DRM driver and I see no freeze. This was a fresh install/update - were the freezing issues seen on upgraded Stretch installs? Or some other [...]

Which video output you used? We are talking about Raspbian Buster freezes at Pi3B+ with composite / analog output. With HDMI output everything is fine

dreamvictim, 2019-07-24:
I've just done rpi-update again and re-enabled the GL driver. The Open Auto Pro splash screen is now showing and central but it still fails to load so I can't get to the desktop. But good news is the screen looks central so definitely an improvement. I'm going to try again with the GL-driver disabled and see if the desktop is displaying correctly.

JamesH65, 2019-07-24:
All my tests done on a Pi3B+, fully up to date Buster, composite output. My post above has a typo, I did not have KMS enabled.

dreamvictim, 2019-07-24:
right, with the latest firmware from today I can get the screen displaying properly without using "overscan_scale=1" which is something I was never able to do previously. With KMS enabled it hangs on splash screen. With it disabled I can see desktop displayed correctly.

JamesH65, 2019-07-24:
Not sure about the KMS issue, haven't tried that, I will check that when I get in to the office tomorrow. KMS or FKMS? All of the work for Pi4 that was also used on earlier models was in FKMS.

dreamvictim, 2019-07-24:
I've only tried Fake. Thanks :)

JamesH65, 2019-07-25:
OK, there does seem to be an issue in there somewhere with composite and FKMS on <Pi4, currently investigating. Think we know what it is.

popcornmix, 2019-07-25:
rpi-update firmware should have a fix for this issue.

peetereczek, 2019-07-25:
Thanks!

System is now also working on composite output, going to test rest of the overscan parameters

peetereczek, 2019-07-25:
Unfortunately rest of overscan parameters are not working, also there is no reaction for frame buffer parameters. At least systems works, so there is improvement

dreamvictim, 2019-07-25:
Thanks! I can also confirm that the system now boots without freezing.

JamesH65, 2019-07-25:
I would expect overscan parameters to NOT work on FKMS but to work on legacy mode - is that what you are seeing?

peetereczek, 2019-07-25:
Honestly saying, on Stretch setting dtoverlay=vc4-fkms-v3d and overscan worked together, maybe I misunderstood your last question...

So which parameter I should set to have overscan with composite video output working on Buster?

JamesH65, 2019-07-26:
The whole graphics stack has had a major revamp with Buster.  We have moved the a MESA/(f)KMS/DRM stack, with major changes to huge amounts of code, a lot of it coming from upstream, as we move away from the current firmware 3D and rendering to a more standard stack. We are currently backporting from kernel 5.1 to our kernel the overscan facilities for DRM which we are plumbing in. But right now you will need to turn off vc4-fkms-v3d to use overscan.

6by9, 2019-07-26:
Overscan support within DRM/KMS is merged with https://github.com/raspberrypi/linux/pull/3090
Setting it up from the command line is pending as https://github.com/raspberrypi/linux/pull/3103

peetereczek, 2019-07-26:
I managed to test composite output without FKMS, and it seemed fine. I could adjust overscan parameters.
Nevertheless application I'm using on Raspberry is not working without FKMS, so looking forward your updates.

OlegBURm, 2020-11-26:
Hi everyone! Any news about this topic? I have the same issue at Rpi B3+ OS Raspbian with composite video: no_KMS driver gives right picture size + possible overscan adjustment. But FKMS simply hangs - I see no desktop at all, KMS gives oversized pictures,  no overscan adjustment is possible. VLC is freezing while playing video without KMS :( . 
Is there any solution at place already?
Thanks in advance!

dcht00, 2021-12-08:
> But right now you will need to turn off vc4-fkms-v3d to use overscan.

I can confirm commenting this out solved overscan problems for me
`dtoverlay=vc4-fkms-v3d
`

Rpi 3b+ → updated bulllseye → composite out → Sony Trinitron
Overscan & framebuffer size settings seem applied during boot up until a certain point, when display goes rogue.
Plugging in/out HDMI switches gracefully with vc4-fkms-v3d, so that helped to keep trying and eventually find this thread.

Nothing else fixed overscan, including
`"overscan_scale=1".
`

I didn't have any of the freezing problems as described above however.